dresden-elektronik/deconz-rest-plugin #2055: URGENT: my pi3 is not booting up, all house without light control, where is the database

Klagio, 2019-11-10:
Hope I can access the SD card with another computer .... will try to install deCONZ on a proxmox VM (any advice of minimum requirement and which is the best Linux for it?)

But of course I need the files with all the data of my 80+ device (yes I know I should have backed up) .. can you please tell me where they are in the SD card, and how to manually restore them in a new installation, 

thanks!!!

SwoopX, 2019-11-10:
You find the data in the home directory deconz was running under. Assuming it was running as user "pi":

/home/pi/.local/share/dresden-elektronik/deCONZ/zll.db

Klagio, 2019-11-11:
Thanks!  I see that there are 4 files more, nevertheless I saved them all. WIll I need to replace all of them?

Klagio, 2019-11-11:
Now I flashed the SD card with the official deCONZ image for Raspberry, all is working but there is no directory 
/home/pi/.local/share/dresden-elektronik/deCONZ/

where is this directory??

SwoopX, 2019-11-11:
Not sure what you mean. Does you have deconz up and running again? Which user is it running as?

closed 2020-03-17 by stale[bot]