dresden-elektronik/deconz-rest-plugin #2057: Zigbee 3.0 with Conbee 1/Conbee 2

lubbertkramer, 2019-11-11:
With Zigbee 3.0 around the corner and the first devices are coming out i'm wondering how deconz is handling the Zigbee 1.2 vs Zigbee 3.0 and can the Conbee 1 / Conbee 2 be made a Zigbee 3.0 coordinator in my network or is this already by default?

Kane610, 2019-11-11:
Hasn't Hue devices been Zigbee3 for some years already?

lubbertkramer, 2019-11-11:
Please stay on topic, question is about conbee 1/2 and zigbee 3.0 and if one of them can become a zigbee 3.0 coordinator and how they handle zigbee 3.0 support.

More and more zigbee 3.0 are coming now not only Hue for example -> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061

Kane610, 2019-11-11:
What I meant was that if Hue already is 3.0 don't deCONZ already support 3.0?

lubbertkramer, 2019-11-11:
That's still not an answer if the conbee 1/2 can be promoted to a zigbee 3.0 coordinator :)

Kane610, 2019-11-11:
No but it gives nuance :)

lubbertkramer, 2019-11-11:
Stop hijacking this request, hue has nothing to do with having a zigbee coordinator 3.0 with the c onbee 1 or 2. 

Devices that are zigbee 3.0 are not relevant in this topic because because 3.0 is backwards compatible with 1.2 but 3.0 has advantages on 1.2 that's why i would like to have a zigbee 3.0 coordinator where all my zigbee devices connect with -> https://zigbee.org/zigbee-for-developers/zigbee-3-0/

manup, 2019-11-11:
ConBee and RaspBee do support a wide range of Zigbee 3.0 devices. Also the Zigbee Green Power extension, which is part of Zigbee 3.0 and used by devices like Friends of Hue switches. Next year we'll roll out Zigbee 3.0 Install Code support for advanced security.

>but 3.0 has advantages on 1.2

Do you have any specific advantage in mind?

lubbertkramer, 2019-11-11:
Thanks @manup for the reply, well as you mentioned for a security point of view and the mesh/range should be better with zigbee 3.0 then with zigbee 1.2. As mentioned in other issues a lot of us suffer from lost devices (like this issue -> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261)  so i'm looking for a way to improve that, that's why i would like to know if the conbee 1/2 will be a zigbee 3.0 coordinator and if that will improve the quality of the netwerk/mesh

I'm not saying my zigbee network is large but i have a good amount of devices where i somethimes lose connection of the lights and then also of enddevices connected to it -> 

Hardware:
Nuc with deconz running in docker with a conbee 1

My network exists of:
18 zigbee lights (ikea/hue)
12 zigbee plugs (ikea/osram)
3 CC2531 and 1 CC2530 repeater
7 zigbee xiaomi switches
3 hue dimmers
3 zigbee xiaomi magiccubes
3 zigbee xiaomi buttons
4 zigbee xiaomi vibration sensors
17 zigbee xiaomi contact sensors
4 hue motion sensors
18 zigbee xiaomi motion sensors
9 zigbee xiaomi temperature sensors

manup, 2019-11-11:
I'm afraid nothing of this is related to Zigbee 3.0, the mesh functionality is the same as prior versions. The drop offs are related to bugs in firmware or perhaps software. At the moment the most reliable lights are Philips hue ones. 

If not already done I highly recommend to use a USB extension cable.

lubbertkramer, 2019-11-11:
Thanks @manup for the clarification 
I already have indeed the conbee 1 connected to an USB extension cable which was one of the made suggestions

lubbertkramer, 2019-11-12:
@manup would it matter if i exchange the conbee 1 for a conbee 2 or does that really not change anything (still not very clear whats the difference/if one is better then the other)?

pac0san, 2020-01-11:
> ConBee and RaspBee do support a wide range of Zigbee 3.0 devices. Also the Zigbee Green Power extension, which is part of Zigbee 3.0 and used by devices like Friends of Hue switches. Next year we'll [...]

Don't you?

https://courses.csail.mit.edu/6.857/2017/project/17.pdf

sic:

"4.5 Updates on Zigbee 3.0

There is a designed ”moment of insecurity” in the Zigbee HA 1.2 specification that uses a well-known symmetric encryption key known as the Trust Center Link Key to distribute a unique network key when a device first joins the network. This is a tradeoff that the Zigbee Alliance chose to make between security and simplicity - with a mitigated impact given that an attacker would have to be capturing Zigbee network traffic at the same time that a new device is being joined to the network. [5]

This method has been removed from the upcoming Zigbee 3.0 specification and replaced with a process that requires a per-device installation code that is used to generate a unique joining key, which is then used to acquire the Zigbee network key. The install code may be printed on the device, be a 2D barcode that is scanned by a camera, or some other out-of-band method of passing the code from the end-device to the Zigbee Coordinator device (in our case, the SmartThings Hub) such as NFC or Bluetooth Smart. However, our devices did not support the out-of-band key establishment and instead utilized the Trust Center Link Key described above."

Smanar, 2020-01-11:
I have already see certified zigbee 3.0 device but they never use this feature, boring.

ebaauw, 2020-01-11:
Could be (one of) the idea(s) behind the Hue lights now supporting Bluetooth: The Hue app can get the install code from the light over Bluetooth, and pass it to the Hue bridge through the API when pairing the light to the bridge.  Not sure if they do this yet; I don't have any of the new Bluetooth-enabled lights.  Their API has been supporting device IDs in the search for new lights since the start, see https://developers.meethue.com/develop/hue-api/lights-api/, section 1.3.

realwax, 2020-02-05:
Any update on this regarding Conbee 1. It seems like Conbee 2 offers zigbee 3.0 as it is defined to be compatible to zigbee 3.0 in it PDF marketing flyer ;) So what does that means to Conbee 1 - will there be an update, merging / uplifting 1.2 -> 3.0. Can it be chosen? As any CC2531 can operate both it would be a bummer needing to head for extra costs to replace CB1 with CB2 just for that. Any new  roadmap? Thanks 4 information....

Smanar, 2020-02-05:
There is no difference in use between conbee and conbee2.

DoctorZIP, 2020-02-22:
I have a few ET Spirit thermostats and they uses Green Power features (according to 3.0 specs) but there are no appropriate UART docs for ConBee 2 device even existing docs were updated a year ago. ConBee devices are leak of support, waste my time and money.

Pedder007, 2020-02-26:
I got a Xiaomi Mi light sensor (pure light sensor w/o motion detection, shall measure up to 83.000Lux) these days, which works also on basis of Zigbee 3.0. I wasn't able to bind him to my ConBee 2. Tried it several times, but it wasn't discovered by the stick. Does someone have experiences with the sensor?

Smanar, 2020-02-27:
@Pedder007 I think the issue about your device is here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2380

Pedder007, 2020-02-29:
Thanks @Smanar

Hedda, 2020-03-06:
Does or does not the ConBee II as a coordinator firmware fully support the full Zigbee 3.0 stack?

To clarify, does ConBee II as coordinator for with deCONZ provide a fully compliant ZigBee 3.0 solution?

To compare with chips from competition Texas Instruments they have Z-Stack firmware version 1.2 which only supports the Zigbee Light Link (ZLL) profile and Zigbee Home Automation (ZHA) profile but then they also have Z-Stack firmware version 3.0 which supports the full Zigbee 3.0 stack:

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

http://www.ti.com/tool/Z-STACK

That is, can ConBee II coordinator firmware create a Zigbee 3.0 cl mesh network which can contain Zigbee 3.0 devices without running in backwards-compatible mode using Zigbee Light Link (ZLL) profile or Zigbee Home Automation (ZHA) profile?

That is, I do not only want it to create a Zigbee Light Link (ZLL) network or a Zigbee Home Automation (ZHA) network but instead Zigbee 3.0 network mesh.

I believe that just like the original poster, I am not just asking if ConBee supports controlling Zigbee 3.0 devices as that question is really irrelevant here because most Zigbee 3.0 devices are backwards compatible with the Zigbee Light Link (ZLL) profile Zigbee Home Automation (ZHA) profiles.

miguelcaravantes, 2020-07-03:
so what is the official answer? because i spent money in this device, at least I want an it will/it won't answer

indrora, 2020-07-19:
Nudging, because I'd like an answer to this as well.

MattWestb, 2020-07-19:
@miguelcaravantes Im very sorry but i dont think its possible getting any official answer here in the  company's own official issue tracker, but perhaps you never knowing.
I hope we its getting it. But keep the door open for other solutions (no DE ones) if needed to keep your money spent safe (I do).
One side note: Most Ikeas devices its upgraded to Z30 except my old RGBWW and some more old devices.

closed 2020-08-16 by stale[bot]

flonair, 2020-10-26:
Still no answers? How can that be? You can at least communicate if you can't or don't want to implement Zigbee 3.0.

manup, 2020-10-26:
Hi everyone, sorry for the very late response.
The short answer yes.

The longer answer:
I'd like to clarify why the question is difficult to answer with yes or no for real world Zigbee devices, as I guess the thinking might go: Supports Zigbee 3.0 > Yes > must be support all Z3.0 devices. Sadly that's not how the Zigbee standard works.

ConBee and RaspBee (I and II) support a large and growing number of ZHA, ZLL, ZGP and Z3.0 devices and also Zigbee 3.0 features like install codes or Zigbee Green Power (ZGP) devices. For the most part the difficulty with Zigbee device support has nothing to do with the specific standard like ZLL or Z3.0 but with differences in devices and how they need to be implemented in order to support a specific device.

There are various gateways out there which support Zigbee 3.0 but none which supports every device on market. I'd like to think of Zigbee 3.0 as a low level detail with a check mark behind it. ConBee II and RaspBee II operate on a Zigbee 3.0 Stack, but even RaspBee I which is 7 years old and still receives updates supports the same devices and Zigbee 3.0 features. The hardware and firmware doesn't do much more than acting as a modem, the vast part of code and device support is implemented at a higher level in deCONZ and the REST-API plugin. The hardware itself is therefore quite future proof and can also support Thread if it ever becomes relevant.

In the end it might be a matter of taste which software above the hardware is used. ConBee and RaspBee isn't bound to deCONZ and is supported by other software projects like zigbee2mqtt, zigypy-zha, and WebThings gateway as well. In terms of future proof RaspBee I outlived the round Philips Hue bridge 1 and the OSRAM gateway and there no plans to discontinue it.