dresden-elektronik/deconz-rest-plugin #2058: Fading time in scenes with INNR lights not working from off or to off

mickeman, 2019-11-11:
If i create a scene with dim-level 100% and fading time for 10 seconds the light doesn´t turn on until 10 seconds have passed then they turn on instantly to 100%.
The same happens if i make a scene with dim level 0 (off) and fading time 10 seconds the difference is that they turn off instant when i start the scene.
If the light is turned on 1% and i start scene to 100% with 10 sec fade it works like it should, same if i start from 100% and fade to 1%
I have Ikea Trådfri and Philips hue lights also and they work like it should, fade in and out slowly from/to off.
The lights i have is 4 Innr RB185C and 2 Innr RB148T

Setup:
Conbee II (Firmware 26490700)
Latest Hass.io 
deCONZ Add-on 3.7 (deconz 2.05.70)
Scenes created in Phoscon

binford6000, 2019-12-26:
Hi mikeman,
same here with FL 130 C. I use fhem in order to dimUp / dimDown regardless of scenes in phoscon.
But i also want to use phoscon scenes with fading time.

nin9s, 2020-01-21:
do you dim the innr light via the phoscon app? In my case its working as expected via home assistant dimmer configuration

binford6000, 2020-01-22:
> do you dim the innr light via the phoscon app? In my case its working as expected via home assistant dimmer configuration

I try to but it switches hard off/on rather than dimup/dimdown. Therefore i use fhem in order to dimup / dimdown. Apperently the lights support dimming - but actually not in deconz scenes :-(

nin9s, 2020-01-22:
yes that could be true but as said it’s working via home assistant here’s

closed 2020-05-30 by stale[bot]

tonih, 2022-01-07:
i have this issue too, using homeassistant with the conbee 2 and the deconz/phoscon addon