dresden-elektronik/deconz-rest-plugin #2059: Paul Neuhaus Q-REMOTE

cracyfloyd, 2019-11-11:
Can anyone help to integrate this Paul Neuhaus Q-REMOTE in the _Conbee II network. step by step.
Here a link for remote information:
__https://www.paul-neuhaus.de/shop/de/ersatzteil-fernbedienung-sender-100-462-31.html

best regards

cracyfloyd, 2019-11-15:
no one can help? i see this remote in deconz but it it possible that i can see the remote in phoscon and define buttons or anything else?

sheineme, 2019-11-18:
I have the cheap version of a Paul Neuhaus remote - the one that comes included when you buy a Q-lamp (Q-PANEL with adjustable color temp. in my case). I also can't see it in Phoscon. I use deConz headless and I have never checked how it looks like in the GUI.
However, I can use the remote: the buttons 1,2 and 3 refer to the groups with the IDs 1,2 and 3. You cannot specifiy the ID of a group when creating it in Phoscon, so I have modified the underlying sqlite database file to make the groups 1, 2 and 3 what I wanted them to be. Changes needed to be done in the groups table and in the table where the relationship groupID -> lightIDs is set up; if this is of interrest, I can look up in more detail what needs to be done (of course: make a backup the original sqlite database file before hacking around in it ;) )
With that I can control my groups 1, 2 and 3 (Kitchen, Dining-, Livingroom): swithing on/off, dimming and color temperature.
My guess is, this just works because the remote sends standard zigbee group commands over the network

alessandroceleste, 2019-11-20:
I have the same problem with the Paul Neuhaus Q-Remote Controller. It would be great to be able to configure it via Phoscon App.

cracyfloyd, 2020-03-10:
Please....Integrate this Remote...

MrSolo570, 2020-03-17:
How can you actually pair it with deconz / phoscon?

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

thueske, 2020-11-24:
Hello guys, hey @Mimiix :)

I also have a Q-Remote from Neuhaus (https://www.paul-neuhaus.de/shop/de/ersatzteil-fernbedienung-sender-100-462-31.html) to control my Zigbee lamp.

It would be really cool if it would be possible to insert it as a switch in deCONZ. I would be happy to help you with that.

What information is needed for the integration?

Many greetings
Tobias

MrSolo570, 2020-11-24:
I am still interested in solving that too, I can support too if somebody tells me what do :)

WhistleMaster, 2020-12-04:
> How can you actually pair it with deconz / phoscon?
> Edit: ok I managed to do that. But within home assistant it does not show up as entity nor within phoscon. How or where could you possibly set which button controls which function? Thank you

If you subscribe to deconz_event in Home Assistant, and then press some buttons on the remote, do you see any events ?

Mimiix, 2020-12-07:
@thueske Sorry, missed your comment! 

Can you open a new device request and provide screenshots of all clusters? Being read ofcourse.

thueske, 2020-12-07:
@Mimiix yes, will try that later :)

flotux2, 2022-01-19:
Is this feature request still pending? Or has it already been integrated?