dresden-elektronik/deconz-rest-plugin #2060: Xiaomi Mijia Zigbee Power Plug x deCONZ X Hass Randomly Turning Off (Logs included)

sabellwind, 2019-11-11:
Hi all,

I’ve set up my mijia power plug to run my salt lamp. I turn it on via my Philips Hue Dimmer switch (which is also running through deCONZ). My salt lamp is turning off at what seems to be a random interval. I traced it back to the following log events in home assistant:

`
2019-11-08 10:46:48 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"1","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"00:15:8d:00:01:dc:0e:ee-01"}
2019-11-08 10:46:48 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"config":{"on":true,"reachable":true,"temperature":2800},"e":"changed","id":"3","r":"sensors","t":"event","uniqueid":"00:15:8d:00:01:dc:0e:ee-02-000c"}
2019-11-08 10:46:48 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"config":{"on":true,"reachable":true,"temperature":2800},"e":"changed","id":"4","r":"sensors","t":"event","uniqueid":"00:15:8d:00:01:dc:0e:ee-03-000c"}
2019-11-08 10:46:52 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"config":{"on":true,"reachable":true,"temperature":2500},"e":"changed","id":"3","r":"sensors","t":"event","uniqueid":"00:15:8d:00:01:dc:0e:ee-02-000c"}
2019-11-08 10:46:52 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.power_3, old_state=<state sensor.power_3=16; on=True, temperature=28.0, current=None, voltage=None, unit_of_measurement=W, friendly_name=Power 3 @ 2019-11-08T10:35:43.416748+11:00>, new_state=<state sensor.power_3=16; on=True, temperature=25.0, current=None, voltage=None, unit_of_measurement=W, friendly_name=Power 3 @ 2019-11-08T10:35:43.416748+11:00>>
2019-11-08 10:46:52 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"config":{"on":true,"reachable":true,"temperature":2500},"e":"changed","id":"4","r":"sensors","t":"event","uniqueid":"00:15:8d:00:01:dc:0e:ee-03-000c"}
2019-11-08 10:46:52 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.139779717776944] Sending {'id': 2, 'type': 'event', 'event': <Event state_changed[L]: entity_id=sensor.power_3, old_state=<state sensor.power_3=16; on=True, temperature=28.0, current=None, voltage=None, unit_of_measurement=W, friendly_name=Power 3 @ 2019-11-08T10:35:43.416748+11:00>, new_state=<state sensor.power_3=16; on=True, temperature=25.0, current=None, voltage=None, unit_of_measurement=W, friendly_name=Power 3 @ 2019-11-08T10:35:43.416748+11:00>>}
2019-11-08 10:46:52 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.139779717243064] Sending {'id': 2, 'type': 'event', 'event': <Event state_changed[L]: entity_id=sensor.power_3, old_state=<state sensor.power_3=16; on=True, temperature=28.0, current=None, voltage=None, unit_of_measurement=W, friendly_name=Power 3 @ 2019-11-08T10:35:43.416748+11:00>, new_state=<state sensor.power_3=16; on=True, temperature=25.0, current=None, voltage=None, unit_of_measurement=W, friendly_name=Power 3 @ 2019-11-08T10:35:43.416748+11:00>>}
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=sensor.consumption_4, old_state=<state sensor.consumption_4=unknown; on=True, temperature=28.0, power=None, unit_of_measurement=kWh, friendly_name=Consumption 4 @ 2019-11-08T10:30:02.270517+11:00>, new_state=<state sensor.consumption_4=unknown; on=True, temperature=25.0, power=None, unit_of_measurement=kWh, friendly_name=Consumption 4 @ 2019-11-08T10:30:02.270517+11:00>>
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.139779717776944] Sending {'id': 2, 'type': 'event', 'event': <Event state_changed[L]: entity_id=sensor.consumption_4, old_state=<state sensor.consumption_4=unknown; on=True, temperature=28.0, power=None, unit_of_measurement=kWh, friendly_name=Consumption 4 @ 2019-11-08T10:30:02.270517+11:00>, new_state=<state sensor.consumption_4=unknown; on=True, temperature=25.0, power=None, unit_of_measurement=kWh, friendly_name=Consumption 4 @ 2019-11-08T10:30:02.270517+11:00>>}
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.139779717243064] Sending {'id': 2, 'type': 'event', 'event': <Event state_changed[L]: entity_id=sensor.consumption_4, old_state=<state sensor.consumption_4=unknown; on=True, temperature=28.0, power=None, unit_of_measurement=kWh, friendly_name=Consumption 4 @ 2019-11-08T10:30:02.270517+11:00>, new_state=<state sensor.consumption_4=unknown; on=True, temperature=25.0, power=None, unit_of_measurement=kWh, friendly_name=Consumption 4 @ 2019-11-08T10:30:02.270517+11:00>>}
2019-11-08 10:46:53 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"1","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"00:15:8d:00:01:dc:0e:ee-01"}
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=switch.salt_lamp, old_state=<state switch.salt_lamp=on; friendly_name=Salt Lamp @ 2019-11-08T10:35:41.816496+11:00>, new_state=<state switch.salt_lamp=off; friendly_name=Salt Lamp @ 2019-11-08T10:46:53.177321+11:00>>
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=group.all_switches, old_state=<state group.all_switches=on; entity_id=('switch.salt_lamp',), order=2, auto=True, friendly_name=all switches, hidden=True @ 2019-11-08T10:35:41.817547+11:00>, new_state=<state group.all_switches=off; entity_id=('switch.salt_lamp',), order=2, auto=True, friendly_name=all switches, hidden=True @ 2019-11-08T10:46:53.179148+11:00>>
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.139779717776944] Sending {'id': 2, 'type': 'event', 'event': <Event state_changed[L]: entity_id=switch.salt_lamp, old_state=<state switch.salt_lamp=on; friendly_name=Salt Lamp @ 2019-11-08T10:35:41.816496+11:00>, new_state=<state switch.salt_lamp=off; friendly_name=Salt Lamp @ 2019-11-08T10:46:53.177321+11:00>>}
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.139779717243064] Sending {'id': 2, 'type': 'event', 'event': <Event state_changed[L]: entity_id=switch.salt_lamp, old_state=<state switch.salt_lamp=on; friendly_name=Salt Lamp @ 2019-11-08T10:35:41.816496+11:00>, new_state=<state switch.salt_lamp=off; friendly_name=Salt Lamp @ 2019-11-08T10:46:53.177321+11:00>>}
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.139779717776944] Sending {'id': 2, 'type': 'event', 'event': <Event state_changed[L]: entity_id=group.all_switches, old_state=<state group.all_switches=on; entity_id=('switch.salt_lamp',), order=2, auto=True, friendly_name=all switches, hidden=True @ 2019-11-08T10:35:41.817547+11:00>, new_state=<state group.all_switches=off; entity_id=('switch.salt_lamp',), order=2, auto=True, friendly_name=all switches, hidden=True @ 2019-11-08T10:46:53.179148+11:00>>}
2019-11-08 10:46:53 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection.139779717243064] Sending {'id': 2, 'type': 'event', 'event': <Event state_changed[L]: entity_id=group.all_switches, old_state=<state group.all_switches=on; entity_id=('switch.salt_lamp',), order=2, auto=True, friendly_name=all switches, hidden=True @ 2019-11-08T10:35:41.817547+11:00>, new_state=<state group.all_switches=off; entity_id=('switch.salt_lamp',), order=2, auto=True, friendly_name=all switches, hidden=True @ 2019-11-08T10:46:53.179148+11:00>>}
2019-11-08 10:46:53 DEBUG (MainThread) [pydeconz] Unsupported event {'e': 'changed', 'id': '65520', 'r': 'groups', 'state': {'all_on': False, 'any_on': False}, 't': 'event'}
2019-11-08 10:46:53 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
2019-11-08 10:46:56 DEBUG (MainThread) [pydeconz] Unsupported event {'e': 'changed', 'id': '65520', 'r': 'groups', 'state': {'all_on': False, 'any_on': True}, 't': 'event'}
2019-11-08 10:46:56 DEBUG (MainThread) [pydeconz.websocket] Websocket data: {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
`
Any idea what's causing this?

bastshoes, 2019-11-12:
I see this issue not only with Xiaomi plug but also with others.
I have LifeControl Plug and I have same behavior.

sabellwind, 2019-11-13:
> I see this issue not only with Xiaomi plug but also with others.
> I have LifeControl Plug and I have same behavior.

That's really unfortunate. Have you done anything about it?

bastshoes, 2019-11-13:
Not yet. 
There was a topic on community forum with similar issues, but issues looks like has beet solved with FW upgrade. I have latest FW on my stick.
It's not clear where to search the bug.

daddvok, 2020-01-12:
The same with Life Control smart plug, every few minutes off-on spontaneously. Only when plug is turned on.

closed 2020-05-18 by stale[bot]