dresden-elektronik/deconz-rest-plugin #2061: Aqara OPPLE switches support

PremiumUsername, 2019-11-11:
Are these switches supported presently, or on your radar for future support?

[Aqara Opple switches](https://homekitnews.com/2019/10/23/aqara-and-opple-release-new-smart-switches/)

They seem very promising I think.

lubbertkramer, 2019-11-11:
Following, saw them on Ali (https://nl.aliexpress.com/item/4000322297550.html) for a nice price

addellll, 2019-11-11:
+1 on those. I saw them yesterday and price is amazing compered to other wall switches on market, and much more potential compered to the normal Aqara wall switch. I will order and report back.

easybeat, 2019-11-12:
+1 on those please. Happy to see that this got the label "device integration". Crazy price of the switch. I've got already the other aquara switch and it works brilliant!

geurti64, 2019-11-20:
👍
Following as well.
Very interesting scene selectors.

djashjones, 2019-11-25:
Just got my 6 button version. Tried to add the device as a switch and it failed but I can see it in the deconz app. The first button switches on all my "Lights" and the second button switches them off.

What info do you guys need to get this device integrated?

addellll, 2019-11-26:
@djashjones https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

djashjones, 2019-11-26:
[image]
[image]
[image]

Sorry don't know how to listen to events in either HA, nor node-red (which I have both).

addellll, 2019-11-27:
In HA ->  Developer Tools -> Events
You write deconz_event in the lower box and press "Start listening". 
After this you try press the Aqara Opple Switch and see if you get any responds.  If sucessful you will see responds similar to the picture. The "event" number should change depening what button you press and if you press, long press or double press. 
Please try it and report back :)

[image]

PremiumUsername, 2019-11-27:
> Just got my 6 button version. 

How do you experience it, quality wise?

djashjones, 2019-11-27:
Same quality as the 2 battery powered 2 button rocker I have. I do like it a lot. It's like the Ikea remote and the base is magnetic.

I will see what events get fired tonight when I get home from work.

Xylomid, 2019-11-27:
Nothing shows up when I listen to deconz_event in home assistant while pressing buttons. I can see it in deconz like the screenshots above but not in phoscon.

What's weird is that it turns on / turn off my trafri bulbs which are linked in phoscon...

djashjones, 2019-11-27:
Confirmed, Nothing in HA, event wise.

I have Ikea Tradfri bulbs (dimmable) and sockets.

Button 1, top Left, action = Switches off all my lights and sockets.
Button 2, top Right, action = Switches on all my lights and sockers.
Button 3, middle left, action = Decreases brightness on bulbs.
Button 4, middle Right, action = Increases bright on bulbs.

[image]

geurti64, 2019-12-02:
Any news for Homey?

mikekoumas, 2019-12-02:
same issue here as [djashjones](https://github.com/djashjones) said
did you find any solution yet?

mat1990dj, 2019-12-02:
+1, just received the 4-button switches and does as commented by @djashjones 
I'm using pimatic that is using the REST API and it's not showing.
Calling the API on /api/ neither shows the MAC address on the list of all devices
What can we help to integrate the device?

heine79, 2019-12-05:
Off topic: Just Receiver 6 Button version. There was no Instruktion in the Box. Could anyone Tell how top pair it? Reset Button seems to bei useless? Hold it up to 5 Seconds but nothing happens?

fleshinachair, 2019-12-05:
Do you have the new Xiaomi homekit hub?

heine79, 2019-12-06:
> Do you have the new Xiaomi homekit hub?
> […](#)

No. I'm unsing conbee with hassio.

djashjones, 2019-12-06:
I think I paired it by holding the button down (top left, next the led) until it started flashing.

Kroeliemonster, 2019-12-06:
I am also eagerly awaiting the integration of these switches. I went ahead and already bought a 6 button one, if there is anything that can be done to speed up the integration let me know.

tuffelh, 2019-12-08:
I also have the 6- and 4-button Versions waiting dir Support :)
Nice Quality btw...

addellll, 2019-12-09:
Found on zigbee2mqtt git that this switches should be able to support "release after hold". 
If true this switches will support event 1234. 
For you guys that have the normal Aqara single/double rocker you might have noticed that it is not possible to dim lights in a good way. This because its only send one event when hold. The new Aqara Opple switch seams to be able to send both x001 (hold) and x003 (release after hold) event.

djashjones, 2019-12-09:
This will be ideal for my bathroom when get supported. 1 for the on/off, 2 for dimming and the third row for the extractor fan manual on/off

fleshinachair, 2019-12-09:
Or WLED, one to change brightness, one colour and one effects!

eserero, 2019-12-11:
Just got mine :-) , 6 button variance
I concur it controls all the lights: 2 top buttons are on and off, the middle buttons are dim up or down and the 2 lower buttons are more yellow or less yellow color.
doesnt appear on domoticz at all or on phoscon :-(.  is there any  way to control what lights it controls?
is there a way to make it control only specific lights?
[image]
[image]

MarcProe, 2019-12-11:
4-Button-Version:
[image]
[image]
[image]

eserero, 2019-12-11:
So playing around a bit with deCONZ, and noticed there is a way to add xml file to extend the profiles, devices etc.  is this something that can help here? or do we need to get some code support for these switches? 

I tried looking at the user guide "extending the zcldb section, but not sure how to map things. it seems the mapping are correct (on/off , dimmer and color are all identified) but there are no events or configurations like there are with another xiaomi switch I have, and I couldnt yet find something obvious in the file that shows me how to link it to a specific type of device.

ebaauw, 2019-12-11:
> is there a way to make it control only specific lights?

Normally, I’d suggest to create a group, add a light to it and bind the switch’s client clusters to that group.  Not sure if that’ll work for Xiaomi, though.  Might be enough to bind only one cluster, but could also you need to bind all three (_On/Off_, _Level Control_, and _Color Control_).

I don’t understand the use for the 2nd endpoint on the 6-button switch.

eserero, 2019-12-12:
hmmm - I tried, the binding files on timeout :-(

ebaauw, 2019-12-13:
Make sure to press a button to wakeup the switch while issuing the bind.

eserero, 2019-12-13:
Thank you very much - working now after some tries and presses. I had to bind each of the items separately, but now its working on one lamp - which is much better. Next steps is I need this to work with 3 lamps I have in my room. do I need 3 groups? one for on/off, dimming and color? 
I will also try to bind to the second endpoint, see what happens...

ebaauw, 2019-12-13:
Best create a single group, bind all three clusters to it, and add multiple lights to this group.

eserero, 2019-12-13:
OK - managed to bind the second endpoint, but it didnt do anything, tried doing long presses, double presses etc. but nothing...
I"ll try the group next, but really need events now, as I want to link some of the actions to devices that are not zibgee

merdok, 2019-12-13:
I would also be interested in support for this switches. Got myself the 2, 4 and 6 ones. Would be great to get them working with conbee...

eserero, 2019-12-14:
well - its kinda working now, with lots of limitations. you can use it to control on/off, dimming and color as long as your lamps are also controlled by conbee - you can create a group of lamps in phoscon and connect the switches using the bind option in deCONZ.  but thats about it. 
You cant get events to domoticz/HA/Homey etc. (which is the biggest problem), you cant use long press or double press and you cant change the button assignments (for example, you cant do toggle with one of the buttons). 
I actually purchased it so that I can have a central controller in each room that can control the lights and the air conditioner (which is controlled from domoticz) - so eagerly waiting for support.

Another question for the experts out there - is there a way to create a "virtual" light in deCONZ? I mean that I can have a light object in deCONZ that is not really a real physical light bulb.
Because I do get light events to domoticz - and if I can create a virtual light in deCONZ that is controlled by the switch, I can use the events in domoticz to control other hardware.

fleshinachair, 2019-12-14:
Has anyone tried it with the Xiaomi Hub to see how well it works?

eserero, 2019-12-16:
haven't tried it. its only supported in the new hub (the one with the siri support). I guess they will add support for the regular hub in the future, no real reason not to.

djashjones, 2019-12-17:
I'm starting to get the impression that the powers that be have no interest in integrating new devices or is it me?

PremiumUsername, 2019-12-17:
> I'm starting to get the impression that the powers that be have no interest in integrating new devices or is it me?

I don't think history merits that kind of cynicism, most (or all?) previous Xiaomi devices have been integrated as far as I know. I'm more inclined to think that it's this season that is busy for all that makes progress slow. That said, I too would appreciate a word or two letting us know that these devices have not been forgotten, even if no progress have been made as of yet.

djashjones, 2019-12-17:
Don't get me wrong, I love my Conbee II stick it's been rock solid for a year

Just ordered some Samsung Smartthings Zigbee plugs 2019 version and reading back  about the previous version that had interest but was never integrated.

PremiumUsername, 2019-12-18:
I just got mine. Curiously when you connect it to Phoscon (you get no feedback that it has found), it immediately controls all lights.   I don't even have a group for all lights, so I'm not sure how to turn this off. I can't see the Switch in Phoscon or via the REST API. Nothing is sent to Home Assistant.

On a hardware note, I can see from the manual that it has a CR2032 battery but there are no instructions for how to eventually replace it. Has anyone endeavored to find out if it is possible?

PremiumUsername, 2019-12-18:
Found the explanation to the strange behavior and also how to access the battery.   https://community.hubitat.com/t/xiaomi-opple-6-button-remote/27921/9

djashjones, 2019-12-18:
Just about to post that link but you beat me to it!

PremiumUsername, 2019-12-18:
@manup Is there any information we can provide you with that would help you in integrating this device more fully?

fleshinachair, 2019-12-18:
It controls all lights that are on phoscon?

djashjones, 2019-12-18:
yes

eserero, 2019-12-18:
Guys, if you look at my post above - I managed to configure it to work with a specific switch or specific group using the bind option in deCONZ. if you need further help on this let me know and I"ll post more exact explanation.

PremiumUsername, 2019-12-18:
> if you need further help on this let me know

That's a nice offer, but I need the integration with Home Assistant for my purposes.

MarcProe, 2019-12-19:
> I"ll post more exact explanation.

Please do.

djashjones, 2019-12-19:
> 
> 
> > if you need further help on this let me know
> 
> That's a nice offer, but I need the integration with Home Assistant for my purposes.

me too. its has more options than the IKEA 5 button I'm currently using.

eserero, 2019-12-19:
> > I"ll post more exact explanation.
> 
> Please do.
agree with everyone we need to get events to home automation systems, otherwise its a crippled solution. using Domoticz myself.

anyway - here is what I did to get it working to my living room lights only.

1. create a group in phoscon (say lights, or whatever switches)
2. if you run phoscon on chrome, right click on the page and select "inspect".
3. go to the network tab - you should see something like the below (I dont know how to get the group number from deconz...)

[image]

4. click on one of the lines with "group " in them - on the right side you can see the group numbers. if you expand them you can see the group names you defined in phoscon. find the group that you want and remember the number 
5. go to deconz, and select the "bind panel" from the panel dropdown
6.  
[image]

7. open the cluster menu from the xiaomi switch and drag the "on/off" line to the "source" button in the bind box. it should be 006. on the destination part select "group" and type in the group number with the preceeding 0x00 as you can see in the screenshot.
8. now press on the "bind" button. - note that it took me a few tries and repeatedly clicking on one of the buttons of the switch in order to wake him up.
9. repeat the operation for "level control" 0008 and "color control" 0300.

AndreasEge, 2019-12-21:
@essero
I have followed your steps above.
When I get to step 7 i do not see any cluster info. What am I missing? 
[image]

AndreasEge, 2019-12-21:
Can the method provided by eserero in any way be used to provide controll of none zigbee devices in Home Assistant? E.g. is it possible to create a light group in Phoscon with no lights and somehow integrate that group in Home Asssitant and use the state of that group to control automations?

ostgardh, 2019-12-21:
deCONZ version 2.05.71
I have try as well.
Have done same as @eserero (Add switch to grop id) but is not working.
And have check deconz event listener when I push the button..
But nothing, quiet..

ebaauw, 2019-12-21:
To see button events, you need proper support in the REST API plugin.  The binding is only for controlling lights directly (and prevent the switch from controlling all lights in your network).

AndreasEge, 2019-12-21:
@ebaauw is there an easy way to get proper support in the REST API plugin while waiting for an official integration? Obviously there is communication from the Opple switch to deconz.

PremiumUsername, 2019-12-21:
> Can the method provided by eserero in any way be used to provide controll of none zigbee devices in Home Assistant? E.g. is it possible to create a light group in Phoscon with no lights and somehow integrate that group in Home Asssitant and use the state of that group to control automations?

That's a very roundabout way of doing it, but if it is possible to create an empty light groups you could use that in HA to control an automation (given that it's possible to turn off and on such an group in phoscon). Unless you are in a huge rush, it's better to wait for proper button events with true deconz integration, instead of using that stopgap-solution in my opinion.

ostgardh, 2019-12-21:
I have dc_eventlog running.
I see other sensor and buttonevent but not from Aqara OPPLE...

AndreasEge, 2019-12-21:
I will try to be patient :-)

ebaauw, 2019-12-21:
> Obviously there is communication from the Opple switch to deconz.

Technically, no, the switch does _not_ communicate with the gateway; it sends broadcast (groupcast) messages, which are also picked up by the gateway.  The gateway eavesdrops on the messages from the switch to the group.

Note that these OPPLE switches work very differently from the other Xiaomi switches, which indeed send reports straight to the coordinator, and cannot control lights directly.

> is there an easy way to get proper support in the REST API plugin while waiting for an official integration?

Wouldn't call this easy.  You need to:
- Whitelist the switch in https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/de_web_plugin.cpp#L103
- Create a button map, probably very similar to https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/sensor.cpp#L498
- Link the switch to the button map in https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/sensor.cpp#L1130
- Set `checkReporting`, `checkClientCluster`, and call `checkSensorGroup()` for the switch in https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/de_web_plugin.cpp#L3036
- Whitelist the switch in https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/bindings.cpp#L2138
- Whitelist the switch in https://github.com/dresden-elektronik/deconz-rest-plugin/blob/bb47cdbc19c257e3000fdb8475aa456830e6df94/bindings.cpp#L2392
- And probably whitelist the switch in one or two other places ("there's always one more place to whitelist a device").

It's not too difficult, if you take a similarly behaving switch that's already supported as example and search the code for references whitelisting that switch.  But it's a bit of trial and error.  That's why I no longer create PRs for switches I don't have access to.

[450 more replies]

closed 2020-07-02 by Mimiix