dresden-elektronik/deconz-rest-plugin #2063: lights unresponsive after power outage

andreibanaru, 2019-11-11:
Had a power outage and now all 10 lights from the living-room, hallway light and 3 lights from bathroom no longer respond. Kitchen and 3 more from bedroom seem to work just fine.

Don't want to do manual reset on 14 lights. What else can be done?

[image]

manup, 2019-11-11:
Have you tried to power-cycle the affected lights?

A reset shouldn't be needed, since devices store all network parameters.

andreibanaru, 2019-11-11:
Yes, no changes. I find it weird since this happened per room basis.

andreibanaru, 2019-11-12:
> Have you tried to power-cycle the affected lights?
> 
> A reset shouldn't be needed, since devices store all network parameters.

Could bringing the Tradfri switch close to the light bulb push the settings again?

closed 2020-03-18 by stale[bot]