dresden-elektronik/deconz-rest-plugin #2064: Deleted Devices still visible in Deconz

zigistar, 2019-11-11:
Hi,

some of my devices are still visible in Deconz despite they are removed 5 month ago and has been flagged as deleted in zll.db database (see screenshots). I also removed them multiple times in Deconz manually (Edit-Menu -> Delete selected Nodes). This removes the nodes only temporarily, they come back again when I reboot the Raspi.

What shall I do to make them disappear permanently? Shall I modify the zll.db with sqlite3?
Any recommendations?

Thanks a lot for your help.

I'm running 2.05.69 and firmware 26330500 on a Raspi 3+

[image]

[image]

Pirol62, 2019-11-28:
Same issue but mine is connected to an osram plug 01. 
Not visible in Phoscon.

ryancasler, 2019-12-24:
I have been having the same problem.  With 1 Tradfri Remote, 1 EcoLink Remote and 1 Lutron Connected Bulb remote.  No matter how many times i delete them they come right back.  Where else can I delete them from?  They have to be in a database table somewhere.

realwax, 2020-02-24:
Same issue here... 
[image] As there is no active bulb or sensor numbering it can't be deleted via deconz rest api. Please fix!

lensherm, 2020-03-12:
Has this been or is planning to be addressed?

Buzztiger, 2020-04-05:
Same problem. They keep coming back after a restart. Anyone figured out how to remove them for good?

ebaauw, 2020-04-05:
So one of my innr SP 120 plugs kept a Hue button end-device in its neighbour table, even though the button had left the network weeks ago.  Even after removing all references to the button from the database, deCONZ would re-create a node for the button, every time the plug's neighbour table was read.  Once you've ended up in a situation like this, there's nothing deCONZ can do.  I had to reset and re-pair the plug to remove the neighbour table entry.  Only after that, the button stopped re-appearing.

I want deCONZ to reset devices when they're removed (through the API or from the GUI), see #2377.  Note however, that won't prevent this situation from occurring ever again.  If the device's _Leave_ message won't reach one of its neighbours (e.g. because the neighbour is offline), it'll keep the device in the neighbour table entry.  Also, it might take up to an hour for the _Leave Request_ to reach a "deep sleeper" battery-powered end device.  If you reset or power down the device before that, it won't send the _Leave_ message.  You might want to wake battery-powered devices when removing them.

Until I'll have made the changes discussed in #2377, best remove devices through the API, supplying a `{"reset": true}` body to the DELETE request.

meremortals70, 2020-05-30:
Another request here.  Need the ability to clean up devices when deleted.

meremortals70, 2020-06-08:
The reason I opened a new issue was it would seem this was dead, is anyone working on this please?

meremortals70, 2020-07-05:
Yep Deconz is no longer in my network

ebaauw, 2020-09-06:
@Mimiix, please assign the right label to shut up the bloody bot.

labaland, 2020-09-15:
bumb

sebres, 2021-01-17:
> Shall I modify the zll.db with sqlite3?

Just for the case someone will try to do such manual deletion from database (zll.db)...
After delete per SQL and restart of deCONZ, the device is still present, so I assume it must be still "unpaired" from conbee somehow ("Delete Node" obviously doesn't do that at the moment), so the gateway can forget all that devices.

I was unable to find in [User Manual deCONZ](https://www.dresden-elektronik.de/funk/software/deconz.html?file=files/dresden-elektronik/content/downloads/dokumente/funktechnik/deCONZ-BHB-en.pdf) how such removal can be done.
Unfortunately, I did not see this (and other) issue, as I chosen the gateway to purchase... 
Very questionable practice to ignore such serious issue (over 1 year)... Someone who gets just tested several devices is unable to "unregister" them, so it's opening a certain security vulnerability as well as probably can be affected by the limit - e. g. impossible to test, develop, whatever else with more than 200 devices once gets paired with self conbee stick.

Back to SQL, here is a statement that show all rows to be deleted:
```sql
.headers on
select * from sensors where sid in (select id from nodes where state = 'deleted');
select * from device_gui where device_id in (select id from nodes where state = 'deleted');
select * from devices where id in (select id from nodes where state = 'deleted');
select * from nodes where state = 'deleted';
```
Make a backup and execute the statement, after replacing of `select *` with `delete`... 

Just as already said, it won't help at all.

ebaauw, 2021-01-17:
> After delete per SQL and restart of deCONZ, the device is still present, so I assume it must be still "unpaired" from conbee somehow ("Delete Node" obviously doesn't do that at the moment), so the gateway can forget all that devices.

That has nothing to do with pairing.  The deleted device would still be in the neighbour tables of its former neighbouring routers.  When the GUI queries these to draw the map, the node is "re-discovered".  Some devices seem to cleanup that table only when power cycled.  Best to delete devices only from the API, with `{"reset": true}`, as mentioned above.  Before resetting, the device sends a _Leave_ message, which causes its neighbouring routers to remove the device entry from their tables.

sebres, 2021-01-17:
> That has nothing to do with pairing.

To me a delete imply everything (a complete removal so the gateway won't see that anymore), no matter how one would call that.

> Best to delete devices only from the API

For who it is better (why a default GUI or web-app cannot remove/unpair/whatever a device)?
Additionally doesn't API need to know what the device is (sensor, switch or light) to delete it with `{"reset": true}`? So what is about device which is a "leak" (neither in lights, nor sensors or switches). So then no chance without to reset gateway completely?

Frank-Bemelman, 2021-12-24:
Hmm. Arrived in this thread today, to find there is not a solution yet. 

I opened the file /config/.storage/core.device_registry and removed the entries by hand. Saved file. Restarted HA and they were gone. 

Cheers,
Frank

manup, 2022-10-07:
Deleted devices don't show up anymore since a while and are hidden / not loaded on startup.

closed 2022-10-07 by manup