dresden-elektronik/deconz-rest-plugin #2065: can't get rid of group0

WladyX, 2019-11-11:
Hello,
When I installed conbee2 for the first time i skipped the group creating phase and so I got Group 0, to which i added some lights.
I reset to factory defaults the conbee2, but those xiaomi lights are subborn and do not want to move out of group 0.
I've used the rest API to remove the groups from the lights, it kinda worked for a while, but when I got a power outage the switches remembered group 0 and so they are back in group 0.
Please help me get rid of this group. The other switches that I did not add manually to group 0 are not visible there, but the ones I added initially are there whatever steps i take.
Thank you for the great work you are doing.

WladyX, 2019-11-12:
[image]
[image]

Exhortae, 2019-12-15:
Hi,

I have the same problem, I tried adding a new aqara switch (same as poster) with neutral wire. And I got this Group 0 that I can't delete. Also the left gang has been added as a light in group 0 and I also cannot remove it.

BrianPuccio, 2020-01-06:
I ran into the same problem when I added an Aqara plug to my collection of Ikea Tradfri plugs. No Group 0 until the Aqara was added, cannot get rid of it now.

xxLeoxx93, 2020-03-01:
Same issue here!

smesguich, 2020-04-12:
Same issue Here #2679 but affecting all my devices :

> I've just check something strange. I'm mooving from my Legrand GW to Deconz and with an Aqara/LUMI Power Outlet, a "Group 0" is created but it's affecting all Legrand devices. As far as I can see in the IHM (Phoscon and old one), the Group 0 contain only the Aqara Device.

Gerben321, 2020-05-31:
Same here after added some Ikea E27 lights.

Smanar, 2020-05-31:
But you have problem with this group ?
You can't just ignore it ?

Gerben321, 2020-05-31:
I get what you're saying, no I don't have a problem, yes I can ignore it. But really? That's no solution, it's not even a work-around. This is clearly some kind of bug that should be addressed and looked at. I understand it's no high priority issue but still. :)

Smanar, 2020-05-31:
Yeah make something to make it invisble in phoscon.
But it can be more problematic on some users.

skitzow88, 2020-05-31:
Well i worked arround this issue as follows using the rest api https://dresden-elektronik.github.io/deconz-rest-doc. 

- GET https://dresden-light.appspot.com/discover
- POST ip:port/api -> raw : { "devicetype": "my application" } -> get the username, this is your apikey you will use in the next step.
- ip:port/api/apikey/groups/0
- DELETE ip:port/api/apikey/groups/0

Smanar, 2020-05-31:
But I have checked the code, it s not possible to delete the group.

>    if (!group || (group->state() == Group::StateDeleted) || (group->address() == gwGroup0))
    {
        rsp.httpStatus = HttpStatusNotFound;
        rsp.list.append(errorToMap(ERR_RESOURCE_NOT_AVAILABLE, QString("/groups/%1").arg(id), QString("resource, /groups/%1, not available").arg(id)));
        return REQ_READY_SEND;
    }

This procedure change something for you ?

xxLeoxx93, 2020-07-03:
Not solved for me -> Should be still open!

ebaauw, 2020-07-03:
`/groups/0` is a standard group containing all `/lights` resources.  Same as Hue API. You shouldn’t be able to delete it, nor add or remove lights to it.  Probably only bug is that when you try anyway, the API returns unclear error messages, or none at all.

xxLeoxx93, 2020-07-25:
But why then is the group 0 not hidden then in phoscon?

ebaauw, 2020-07-25:
That's a question for Phoson, not for the REST API Plugin.

closed 2020-07-25 by ebaauw