dresden-elektronik/deconz-rest-plugin #2066: FoH Niko Switch only emits events on Zigbee channel 15

BrianKlDK, 2019-11-12:
Hi.
I've messing around with Deconz for a while, trying out different devices, etc.
I have most stuff running fine, but while trying out a change from Raspberry PI (Hassio) to regular Linux PC and changing from Raspbee to ConBee II I came across what seems weird.
Amongst my devices, I have a Niko FoH switch, and while I could get to add just fine regardless of channel for my ConBee, it ONLY emits events to Home Assistant when the GW is running on channel 15.
11, 20 and 25 sees no events.
Incidentially the switch was added/paired while the GW was on channel 11, so I was a bit puzzled at first that it actually only worked on 15.

Could there be some weird hardcoding issue or something like that?

Best regards
Brian Klausen

manup, 2019-11-12:
To bring the switch to a specific channel you need to complete the setup procedure as described in
https://phoscon.de/en/support#pairing-friends-of-hue-switch

Step 5. is the important one which pins the switch to the new selected channel.

BrianKlDK, 2019-11-14:
Ah, that explains it.
I have been Googling for the pairing procedure, and haven't been able to come across something concrete.
For the longest time I did the long press followed by 2 buttons, but wasn't aware that the long press had to be a particular button for a particular channel - it just incidentally worked :).
Then I came across some other info about different buttons for different channels, but that you could skip the 2 buttons afterwards. Turns out that has a significance too :).

Thanks a lot for clarifying - and another link added to the collection.

Best regards

closed 2019-11-14 by BrianKlDK

manup, 2019-11-14:
:) yes it's tricky but also a feature to prevent changing the channel by accident like it is possible with the older Hue tap. Future versions of the Phoson App will have a proper wizard which guides through the steps with ease.