dresden-elektronik/deconz-rest-plugin #2068: Not all values get changed for IKEA tradfri lights

marcschumacher, 2019-11-13:
Hi,

I own an IKEA of type `TRÅDFRI Lampe E14 WS opal 400lm` (according to phoscon interface). I issue the following calls (in that order):
To switch the light off:
```
curl -X "PUT" "http://192.168.200.45/api/<userid>/lights/28/state" \
     -H 'Content-Type: application/json; charset=utf-8' \
     -d $'{ "on": false }'
```
then to switch it on with full brightness and certain color:
```
curl -X "PUT" "http://192.168.200.45/api/<user>/lights/28/state" \
     -H 'Content-Type: application/json; charset=utf-8' \
     -d $'{
  "bri": 255,
  "colormode": "ct",
  "ct": 454,
  "on": true
}'
```
and then I want to change the brightness and color temperature:
```
## Set light state
curl -X "PUT" "http://192.168.200.45/api/1C19E33C5E/lights/28/state" \
     -H 'Content-Type: application/json; charset=utf-8' \
     -d $'{
  "bri": 128,
  "colormode": "ct",
  "ct": 800,
  "on": true
}'
```

The last call does not work at all, nothing changes. Sometimes I am only not able to change brightness. If instead of supplying keys `bri`, `colormode`, `ct`and `on` I only supply `on`and `bri` it works well to change the brightness:
```
curl -X "PUT" "http://192.168.200.45/api/<user>/lights/28/state" \
     -H 'Content-Type: application/json; charset=utf-8' \
     -d $'{
  "bri": 255,
  "on": true
}'
```

This issue causes the whole home assistant installation for me not to work as defined. Can you tell me what I am doing wrong or provide a bugfix if this is a bug?

Best
Marc

ebaauw, 2019-11-13:
The REST API plugin needs to translate the request bodies with both `bri` and `ct`  into multiple ZigBee commands (one to the _Level Control_ cluster and one to the _Color Control_ cluster).  It adds the default transitiontime of 4 (= 0.4 seconds) to each command.  Trådfri lights have a firmware bug where they ignore the second command, when the first command is still in transition.

Either send `bri` and `ct` as two API requests (which likely causes enough delay between the requests for the first transition to be finished), or add a `"transitiontime": 0` to the body (which will result in a lousy user experience as the light changes unnaturally quickly).

A couple of other issues:
- You cannot set `colormode` from the API; it's readonly.  Error checking and reporting by the REST API plugin is far from perfect, so it's probably ignored silently;
- A `ct` value of 800 is probably well out of range.  The REST API plugin replaces that with the `ctmax` value;
- A `bri` value of 255 is not valid.  The REST API plugin replaces that with 254.

marcschumacher, 2019-11-23:
Is there a firmware update available that might fix this issue?

realwax, 2020-02-28:
May I ask which firmware version you are using? I experience this issue since 1.x but it got worse with 2.x. I wonder how IKEA operates that on their own bridge as it is a usability issue.

Mimiix, 2020-06-05:
As this issue is a duplicate on #1261 , i will close it for now.

closed 2020-06-05 by Mimiix