dresden-elektronik/deconz-rest-plugin #2069: Cluster marked as unknow

Smanar, 2019-11-13:
Hello, we are trying to include new device and whe have something strange.
https://user-images.githubusercontent.com/25209236/68803156-7ef70880-065f-11ea-8dcb-57b1dcdf2939.png

Clusters are good, but marked as unknow !
Someone know a procedure to solve that ?

ebaauw, 2019-11-13:
Looks like a missing or invalid `general.xml`.

Smanar, 2019-11-13:
Arf, I thought about that but the manufacturer part is only for the cluster FC01, and is marked as "unknow" too, I thought the rest was standard.

It can't be an inclusion bug during the procedure  ?

ebaauw, 2019-11-13:
Yep.  The cluster names and _Custer Info_ panel are populated from `general.xml`.  A single XML error causes the entire file to be ignored.

Smanar, 2019-11-13:
Lol, right, it's exactly that.

closed 2019-11-13 by Smanar