dresden-elektronik/deconz-rest-plugin #2074: can not delete my old sensor

salopette, 2019-11-15:
Hello, I can not delete my old sensor.
I have not had this sensor for a long time, when I delete it, it comes back after the restart.

[image]

Smanar, 2019-11-15:
It can be provoked by neightboard table.
If you are lucky just wait for somes days and retry to delete it.
But some users have still this kind of ghost device after 5 months.

salopette, 2019-11-15:
hi, it's been with me for months! I delete it again and again, but it comes back after the restart :-(

Smanar, 2019-11-15:
You still have this sensor somewhere ?
Try to include it again and remove it a second time, for me 1 month is too much for a neightboard table problem.

ACrazyConcept, 2019-11-15:
I have 3 ghost nodes as well. The latest one is the TRÅDFRI signal repeater.
I tried deleting all rows i could find referring to one of them in all the tables. Didn't work. Probably made it worse I guess.

zigistar, 2019-11-15:
+1 

same issue here. 3 Sensor and 1 light (all Philips Hue) sold 5 month ago and deleted multiple times.
They come back every time I reboot the Raspi. 

Opened a ticket about this issue some days ago too -> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2064

closed 2020-03-21 by stale[bot]