dresden-elektronik/deconz-rest-plugin #2076: Several nodes not working since docker update v2.05.70

yvessa, 2019-11-16:
Hello,

Since last update v2.05.70, a lot of nodes are not reporting motion presence. I tried to downgrade but same issue. I'm using a docker image.

Thank you in advance for your help.

Here is the beginning of the log showing several errors

[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.70
[marthoc/deconz] Web UI port: 8080
[marthoc/deconz] Websockets port: 8443
[marthoc/deconz] VNC port: 6900
/usr/bin/xauth:  file /root/.Xauthority does not exist

New 'ryzen7:1000 (root)' desktop at :1000 on machine ryzen7

Starting applications specified in /etc/X11/Xvnc-session
Log file is /root/.vnc/ryzen7:1000.log

Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd ryzen7:1000 to connect to the VNC server.

libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libpng warning: iCCP: known incorrect sRGB profile
18:31:22:928 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/
18:31:22:934 CTRL. 3.16.218:31:22:953 COM: /dev/ttyACM1 / serialno: 
18:31:22:953 COM: --dev: /dev/ttyACM1 (ConBee II)
18:31:22:953 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
18:31:22:977 parent process /bin/sh
18:31:22:977 gw run mode: docker
18:31:22:977 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
18:31:22:977 DB sqlite version 3.16.2
18:31:22:977 DB PRAGMA page_count: 48
18:31:22:977 DB PRAGMA page_size: 4096
18:31:22:977 DB PRAGMA freelist_count: 3
18:31:22:977 DB file size 196608 bytes, free pages 3
18:31:22:977 DB PRAGMA user_version: 6
18:31:22:977 DB cleanup
18:31:22:977 DB create temporary views
18:31:22:986 don't close database yet, keep open for 900 seconds
18:31:22:987 started websocket server at port 8443
18:31:22:988 found node plugin: libde_rest_plugin.so - REST API Plugin
18:31:22:989 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
18:31:22:991 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
18:31:22:999 COM: /dev/ttyACM1 / serialno: 
18:31:22:999 COM: --dev: /dev/ttyACM1 (ConBee II)
18:31:23:055 Device firmware version 0x26490700
18:31:23:059 unlocked max nodes: 200
18:31:23:131 Device protocol version: 0x010B
18:31:23:138 new node - ext: 0x00212effff042b0e, nwk: 0x0000
18:31:23:200 don't close database yet, keep open for 900 seconds
18:31:23:201 LightNode 1: IkeaSwitch Sous-Sol Calo added
18:31:23:212 don't close database yet, keep open for 900 seconds
18:31:23:212 LightNode 2: IkeaSwitch Corridor added
18:31:23:217 don't close database yet, keep open for 900 seconds
18:31:23:217 LightNode 3: IkeaSwitch Salle Manger added
18:31:23:223 don't close database yet, keep open for 900 seconds
18:31:23:223 LightNode 4: IkeaSwitch Chambre Maitres added
18:31:23:228 don't close database yet, keep open for 900 seconds
18:31:23:228 LightNode 5: IkeaSwitch Petite Chambre Haut added
18:31:23:233 don't close database yet, keep open for 900 seconds
18:31:23:234 LightNode 6: IkeaSwitch Atelier added
18:31:23:239 don't close database yet, keep open for 900 seconds
18:31:23:239 LightNode 7: IkeaSwitch Petite Chambre Bas added
18:31:23:245 don't close database yet, keep open for 900 seconds
18:31:23:245 LightNode 8: IkeaSwitch Corridor Ext added
18:31:23:250 don't close database yet, keep open for 900 seconds
18:31:23:250 LightNode 9: IkeaSwitch Garage Etabli added
18:31:23:256 don't close database yet, keep open for 900 seconds
[... 231 lines cut ...]
18:34:29:397 ZCL attribute report 0x00158D00025EFA1D for cluster 0x0403, ep 0x01
18:34:35:883 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
18:34:35:883 ZCL attribute report 0x00158D0002583A0F for cluster 0x0400, ep 0x01
18:34:37:691 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
18:34:37:691 ZCL attribute report 0x00158D00025D9109 for cluster 0x0402, ep 0x01
18:34:37:700 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
18:34:37:700 ZCL attribute report 0x00158D00025D9109 for cluster 0x0405, ep 0x01
18:34:37:704 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
18:34:37:704 ZCL attribute report 0x00158D00025D9109 for cluster 0x0403, ep 0x01
18:34:39:339 APS-DATA.indication from unknown node 0xB359
18:34:39:339 ZCL attribute report 0x0000000000000000 for cluster 0x0402, ep 0x01
18:34:41:056 0xD0CF5EFFFE7B85A4 error APSDE-DATA.confirm: 0xD0 on task
18:34:47:927 no button map for: lumi.sensor_wleak.aq1 ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
18:34:47:927 ZCL attribute report 0x00158D00023347A0 for cluster 0x0000, ep 0x01
18:34:47:927 0x00158D00023347A0 extract Xiaomi special attribute 0xFF01
18:34:47:927 	01 battery 3045 (0x0BE5)
18:34:47:927 	03 temperature 23 °C
18:34:47:927 	04 unknown 17320 (0x43A8)
18:34:47:927 	05 RSSI dB (?) 42 (0x002A)
18:34:47:927 	06 LQI (?) 4294967296 (0x0100000000)
18:34:47:927 	08 unknown 516 (0x0204)
18:34:47:927 	0a unknown 0 (0x0000)
18:34:47:927 	64 on/off 1
18:34:50:605 max transmit errors for node 0xD0CF5EFFFEDCD9DB, last seen by neighbors 122 s
18:34:59:615 Bind response success for 0xd0cf5efffe7b8bb1 cluster 0x0000

yvessa, 2019-11-16:
Not sure it's a v2.05.70 issue. Anyhow after reading there is some issues with it, I stay with v2.05.69. 
 
I have noticed some routers (ikea switches) were in gray, I unplugged and plugged back these. Then, few sensors are now working. Let's wait few hours if network rebuild by itself.

Is there a troubleshooting page about these kind of issues? I do not want to start from scratch, I have like 70 nodes... Some hints might be helpful avoiding clogging forum. I might not have found the good location yet.

Thanks

yvessa, 2019-11-16:
Forgot to ask, is there a way to know which parent (ikea switch) is allocated to a child (sensors)? I remember reading in the past than xiaomi sensors are locked to a parent. That might explain some issues I have. It would be nice to know parent/child association to diagnose.

ebaauw, 2019-11-16:
If you're lucky, the deCONZ GUI might show a line connecting the end device to its parent.  Note that these lines are maintained by querying the neighbour tables on the devices and might not reflect the actual situation.

Otherwise: sniff the ZigBee traffic and see at MAC level to which router the device sends its messages.

yvessa, 2019-11-19:
Thanks ebaauw, most were not showing a line. ZigBee sniffing would definitely help. It would be nice if we could reallocate a node to a specific parent.

In order to get rid of these issues, I shutdown deconz completely. Then unplug and plug back router nodes (ikea switches). When I started deconz everything was back to normal. Also, updated to v2.05.70. At least I did not have to re-add all my sensors.

I tried unplugging/plugging routers w/o shutting down deconz and it did not work. 

I can't explain what happened exactly.

peer69, 2019-11-21:
I tried that as well. Even unplugged the Conbee Stick as I didnt want to shut down the whole ESXi Host after the deconz vm was shut down. I will probably now set up a test machine with some of the devices as I have a second adapter but I obviously wont be able to replicate my productive environment so I dont know if this might prevent similar scenarios with further updates at all. 
It really felt like this update caused the coordinator to drop all routers and sensors connected through routers from the network so this might be neighbor table related. I moved sensors formerly connected through routers near the gateway but this did nothing. Only sensors connected directly to the gateway before the update were reachable afterwards.

closed 2020-03-27 by stale[bot]