dresden-elektronik/deconz-rest-plugin #2078: deCONZ soft crash?

ebaauw, 2019-11-16:
Last night, deCONZ appears to have "soft crashed" on me:
- The GUI was re-initialised (battery symbols no longer show; removed devices re-appeared);
- the REST API plugin was re-initialised (state of all CLIPGenericFlag and CLIPGenericStatus sensors was reset);
- WebSocket connections were closed;
- The process survived (`systemctl status deconz-gui` shows it's been running for well over a week).

deCONZ v2.05.70, stock REST API plugin, RaspBee v0x26330500, Raspberry Pi 3B+, Raspbian Jesse:
```
Linux pi2 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

Relevant part (I hope) of the deCONZ log:
```
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:39:101 APS-DATA.indication request id: 37 -> finished
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:39:101 APS-DATA.request id: 37 erase from queue
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:39:490 APS-DATA.request id: 43, addrmode: 0x03, addr: 0x000d6f0010656ff1, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:39:782 APS-DATA.request id: 45, addrmode: 0x03, addr: 0x001788011052adc9, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 1 len: 5 tx.options 0x00
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:40:981 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:40:981 device state timeout ignored in state 3
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:41:984 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:41:984 device state timeout ignored in state 3
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:42:982 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:42:982 device state timeout ignored in state 3
Nov 16 00:44:43 pi2 deCONZ[21997]: PROTO: CRC error
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:089 aps request id: 43 prf: 0x0000 cl: 0x0031 timeout (confirmed: 0) to 0x000D6F0010656FF1 (0xFA1D)
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:089 aps request id: 43 prf: 0x0000 cl: 0x0031 timeout NOT confirmed to 0x000D6F0010656FF1 (0xFA1D)
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:090 aps request id: 45 prf: 0x0104 cl: 0x0006 timeout (confirmed: 0) to 0x001788011052ADC9 (0xFA01)
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:090 aps request id: 45 prf: 0x0104 cl: 0x0006 timeout NOT confirmed to 0x001788011052ADC9 (0xFA01)
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:102 APS-DATA.request id: 43, status: 0x06 giveup
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:102 APS-DATA.request id: 43, set state: 0x05
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:102 emit artificial APSDE-DATA.confirm id: 43
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:103 APS-DATA.request id: 43, failed-status: APP_ENONET (0x06)
Nov 16 00:44:43 pi2 deCONZ[21997]: 00:44:43:147 Current channel 0
Nov 16 00:44:50 pi2 deCONZ[21997]: 00:44:43:158 CTRL got nwklibEGL warning: DRI2: failed to authenticate
Nov 16 00:44:51 pi2 deCONZ[21997]: libpng warning: iCCP: known incorrect sRGB profile
Nov 16 00:44:56 pi2 deCONZ[21997]:  update id 13
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:169 Device TTL 3600 s flags: 0x7
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:169 aps request id: 43 failed, erase from queue
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:180 Device protocol version: 0x010B
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:251 aps request id: 45 failed, erase from queue
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:251 emit artificial APSDE-DATA.confirm id: 45
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:251 0x001788011052ADC9 error APSDE-DATA.confirm: 0xA7 on task
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:313 Current channel 0
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:329 aps request id: 45 failed, erase from queue
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:343 CTRL got nwk update id 13
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:353 CTRL ANT_CTRL 0x03
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:560 start reconnect to network
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:577 Device protocol version: 0x010B
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:709 Current channel 25
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:739 CTRL got nwk update id 13
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:749 CTRL ANT_CTRL 0x03
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:43:982 APS-DATA.request id: 72, addrmode: 0x03, addr: 0x001788011041edd0, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:46:598 APS-DATA.indication srcAddr: 0xfe04, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B04, lqi: 239, rssi: -61
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:46:598 #011asdu: 18b10a0b05292c00
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:46:599 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:46:599 ZCL attribute report 0x00158D000217AD31 for cluster 0x0B04, ep 0x01
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:47:598 APS-DATA.indication srcAddr: 0xfe04, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B04, lqi: 239, rssi: -61
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:47:598 #011asdu: 18a30a0b05292c00
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:47:599 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:47:599 ZCL attribute report 0x00158D000217AD31 for cluster 0x0B04, ep 0x01
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:48:593 APS-DATA.indication srcAddr: 0xfe04, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0B04, lqi: 239, rssi: -61
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:48:593 #011asdu: 188e0a0b05292c00
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:48:594 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 00B
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:48:594 ZCL attribute report 0x00158D000217AD31 for cluster 0x0B04, ep 0x01
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:48:811 reconnect network done
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:48:947 saved node state in 23 ms
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:48:985 sync() in 37 ms
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:48:986 REST API plugin shutting down
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:49:196 sync() in 12 ms
Nov 16 00:44:56 pi2 deCONZ[21997]: 00:44:51:303 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
[... 57 lines cut ...]
Nov 16 00:44:57 pi2 deCONZ[21997]: 00:44:56:911 LightNode 322: Bedroom 2 added
Nov 16 00:44:57 pi2 deCONZ[21997]: 00:44:56:930 don't close database yet, keep open for 900 seconds
Nov 16 00:44:57 pi2 deCONZ[21997]: 00:44:56:930 LightNode 233: Dining Room 3 added
Nov 16 00:44:57 pi2 deCONZ[21997]: 00:44:56:944 don't close database yet, keep open for 900 seconds
Nov 16 00:44:57 pi2 deCONZ[21997]: 00:44:56:944 LightNode 111: Entrance 1 added
```
Here's the homebridge log: showing the websocket connection being closed, a failed API request (as the web server restarted) and the CLIP sensors being reset:
```
Nov 16 00:44:45 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mpi2: gateway request 376796: ok#033[39m
Nov 16 00:44:46 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mLiving Room Plug: state changed event: {"lastupdated":"2019-11-15T23:44:46","power":44}#033[39m
Nov 16 00:44:46 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mLiving Room Plug: sensor power changed from 40 to 44#033[39m
Nov 16 00:44:46 pi2 homebridge[28899]: #033[36m[Hue]#033[39m Living Room Plug: set homekit current consumption from 40 W to 44 W
Nov 16 00:44:47 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mLiving Room Plug: state changed event: {"lastupdated":"2019-11-15T23:44:47"}#033[39m
Nov 16 00:44:48 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mLiving Room Plug: state changed event: {"lastupdated":"2019-11-15T23:44:48"}#033[39m
Nov 16 00:44:49 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mpi2: websocket connection closed - retrying in 30 seconds#033[39m
Nov 16 00:44:50 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mpi2: gateway request 376797: get /sensors#033[39m
Nov 16 00:44:50 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[1m#033[31mpi2: gateway request 376797: get /sensors#033[39m#033[22m
Nov 16 00:44:50 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[1m#033[31mpi2: gateway communication error ECONNREFUSED on 127.0.0.1#033[39m#033[22m
[... cut ...]
Nov 16 00:44:59 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mpi2: gateway request 376798: ok#033[39m
Nov 16 00:44:59 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mHome Status: sensor status changed from 1 to 0#033[39m
Nov 16 00:44:59 pi2 homebridge[28899]: #033[36m[Hue]#033[39m Home Status: set homekit status from 1 to 0
Nov 16 00:44:59 pi2 homebridge[28899]: #033[36m[Hue]#033[39m #033[90mErik: sensor flag changed from true to false#033[39m
Nov 16 00:44:59 pi2 homebridge[28899]: #033[36m[Hue]#033[39m Erik: set homekit on from 1 to 0
```

manup, 2019-11-16:
Spooky, is this the tagged 2.05.70 release or latest commits?
Might be related to https://github.com/dresden-elektronik/deconz-rest-plugin/commit/048180ee819d050160b6d2ecb2ab61bd1ea56635

Which will be part of 2.05.71.

ebaauw, 2019-11-16:
Installed thru `apt update`/`apt upgrade` using `http://phoscon.de/apt/deconz stretch-beta InRelease`.

ericvandamme, 2019-11-20:
Where can the log be found? I just had a similar issue last night and it looks like a similar setup.

ebaauw, 2019-11-20:
That depends on how you run deCONZ.  When using systemctl, the output is logged to syslog.

ericvandamme, 2019-11-20:
Hmmm. Appears to be a different issue. I just lost deCONZ control completely despite still "running". Basically it wrote this to the log for 8 hours until I woke up and realized nothing worked.

```
Nov 20 01:31:08 deconz deCONZ[1196]: libEGL warning: DRI2: failed to authenticate
Nov 20 01:31:08 deconz deCONZ[1196]: libpng warning: iCCP: known incorrect sRGB profile
Nov 20 01:31:35 deconz deCONZ[1196]: libEGL warning: DRI2: failed to authenticate
Nov 20 01:31:35 deconz deCONZ[1196]: libpng warning: iCCP: known incorrect sRGB profile
Nov 20 01:32:01 deconz deCONZ[1196]: libEGL warning: DRI2: failed to authenticate
Nov 20 01:32:01 deconz deCONZ[1196]: libpng warning: iCCP: known incorrect sRGB profile
Nov 20 01:32:28 deconz deCONZ[1196]: libEGL warning: DRI2: failed to authenticate
Nov 20 01:32:28 deconz deCONZ[1196]: libpng warning: iCCP: known incorrect sRGB profile
Nov 20 01:32:54 deconz deCONZ[1196]: libEGL warning: DRI2: failed to authenticate
Nov 20 01:32:54 deconz deCONZ[1196]: libpng warning: iCCP: known incorrect sRGB profile
Nov 20 01:33:21 deconz deCONZ[1196]: libEGL warning: DRI2: failed to authenticate
Nov 20 01:33:21 deconz deCONZ[1196]: libpng warning: iCCP: known incorrect sRGB profile
Nov 20 01:33:47 deconz deCONZ[1196]: libEGL warning: DRI2: failed to authenticate
Nov 20 01:33:47 deconz deCONZ[1196]: libpng warning: iCCP: known incorrect sRGB profile
Nov 20 01:33:48 deconz deCONZ[1196]: QNetworkReplyImplPrivate::error: Internal problem, this method must only be called once.
Nov 20 01:34:14 deconz deCONZ[1196]: libEGL warning: DRI2: failed to authenticate
Nov 20 01:34:14 deconz deCONZ[1196]: libpng warning: iCCP: known incorrect sRGB profile
```

manup, 2019-11-20:
The recently released 2.05.71 version should fix the issue aka soft restart of deCONZ.

ebaauw, 2019-11-20:
Cannot find it on the `apt` beta repository.  Did find .71 `deconz-dev` though.  Both Stretch and Buster.

manup, 2019-11-20:
Just looked at the repository, the files are all there. Also at http://deconz.dresden-elektronik.de/raspbian/beta/

I'll need to do some tests under on my RPi ..

ebaauw, 2019-11-20:
```
$ sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Hit:3 http://phoscon.de/apt/deconz stretch-beta InRelease                            
Hit:4 https://deb.nodesource.com/node_12.x stretch InRelease
Fetched 25.4 kB in 1s (14.6 kB/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
$ sudo apt list deconz
Listing... Done
deconz/now 2.05.70-raspbian-stretch-beta armhf [installed,local]
$ sudo apt install deconz
Reading package lists... Done
Building dependency tree       
Reading state information... Done
deconz is already the newest version (2.05.70-raspbian-stretch-beta).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
$
```

ebaauw, 2019-11-21:
So the stable repository has 2.05.71, but the beta repository doesn't.  That's a first...

manup, 2019-11-21:
Took a while to figure it out, we're using reprepo to create the APT repository. For some reason the stretch-beta package (from an earlier test build) wasn't removed before creating the repository. A new build is running now, with a cleaned up stage. Should be up an available in 1–2 hours.

ebaauw, 2020-01-31:
Haven't seen this anymore on .71 and .72.

closed 2020-01-31 by ebaauw