dresden-elektronik/deconz-rest-plugin #2079: Not working with brand new Hue Ambilight E27

sysadmama, 2019-11-16:
Hi,

yesterday I got my brand new Hue Ambilight E27 bulb. I paired it with RaspBee as I do with every device and it immediately showed up as a `Color temperature light`.
```
Vendor  Philips
Model  LTA001
Version 1.53.3_r27175
```

It is not possible to use the bulb via deconz web interface; UI even does not recognize when the bulb is completely `off`; and when I change setting in UI, the bulb does not change.

All other devices connected to RaspBee are working fine; it's not the only Hue in this setup, but the only with a 1.53.* firmware. Might that be a problem? I'm using `2.05.70-raspbian-stretch-beta`.

Cheers,
  Marianne

ebaauw, 2019-11-16:
I haven't seen this model before.  Is this one of the new lights also supporting Bluetooth?  Unfortunately, Signify seem no longer to maintain the list of [supported lights](https://developers.meethue.com/develop/hue-api/supported-devices).  I haven't seen that firmware version either and it's not listed on the [changelog](https://www2.meethue.com/nl-nl/ondersteuning/versie-informatie/lamps-luminaires).  Could very well be the initial shipping version of the new model.  You would need to pair it with a Hue bridge to check that no update is available.

Looks like the gateway cannot (doesn't know how to) reach the light.  Note that Hue lights don't support notifications, so the gateway needs to poll the light for its actual state.  Does it still react to group commands (the REST API plugin should have added it to `/groups/0`)?

You might try and power cycle the light - typically this fixes routing problems.  Otherwise reset and re-pair the light.  You'd need to Hue dimmer switch to do a touchlink-reset (press and hold the _On_ and _Off_ buttons while holding the switch next to the light).  Be sure to power cycle the light before that, or it won't accept touchlink commands.

sysadmama, 2019-11-16:
Hi @ebaauw and thank you very much for your answer,

just for completeness I tell you how the story ended: your'e absolutely right, it is one of the combined Zwave/Bluetooth devices. But - I remembered last week I added a Hue Filament bulb, and it works like a charm with RaspBee. Had a look at it:
````
Vendor Philips
Model LWV001
Version 1.55.8_r28815
````

So I installed this Hue Bluetooth App thing on my iPhone, paired both bulbs with it – and got a firmware upgrade for the ambient bulb (not for the Filament as it already had 1.55.8_r28815).

Now both bulbs are running on `1.55.8_r28815`, and everything is great again: RaspBee can communicate with both without any problems.

BTW: power cycle and re-adding did not fix the issue with the old firmware. Seems there is no other solution than to upgrade the bulb. Happily it's bluetooth, so no Hue Bridge needed (as I only have the very old one).

So I'll close this issue as it is not deconz related.

Thank you very much,
  Marianne

ostgardh, 2019-11-26:
I have two Philips Model LWV001
Version 1.55.8_r28815
and i can add bulb to deconz / Phoscon GUI.
But it possible to controll and after some min I lost contact.
Is possible to add bulb to Hue Bluetooth App and controll the light.
I have check firmeware i Hue Bluetooth App.. But is ok , no new...

I have try to reset and try.. Same problem.. I can´t controll from  deconz...

I don´t understand what sysadmama  have fix this..

//P

sysadmama, 2019-11-26:
ostgardh,

you're absolutely right, I'm sorry. It fixed it for me for the moment (it was more than some minutes), and then I forgot about it. But since then I can confirm your observations.

I paired this specific bulb with the old Hue Bridge now, and then it works. I still have paired my Filament bulb with RaspBee, and it's available AND I can control it.

That's why I reopen this issue.

Cheers,
  Marianne

ostgardh, 2019-11-26:
Ok.. 
I have one
VendorPhilips
ModelLWO001
Version 1.55.8_r28815

as well.. And that is working 100% ok

ostgardh, 2019-11-27:
After I delete the light in deconz(again) and add new after power reset is working now..
I don´t understand why.. But is working after some trouble..

aerobless, 2020-01-01:
I'm having the same problem with another hue filament bulb:

Vendor Philips
Model LWA004
Version 1.55.8_r28815

I can discover it alright with deconz, but once it's added it can't be controlled. I've tried resetting the bulb and restarting deconz several times but the result is the same :/. Connecting to it via bluetooth didn't help either, and there is no update available. Hopefully a fix can be found for this.

closed 2020-05-07 by stale[bot]