dresden-elektronik/deconz-rest-plugin #2082: Osram Smart+ Switch Mini support

henkkakay, 2019-11-17:
On the back of the switch, there are some details (model number?):
AC0251 100NJ | x 0218

The switch has 3 buttons, up (close to LED), centre/circle, down. Below is lists the long/short presses for each button. Screencaps show the info panels that I could find with possibly useful info.

===============

Down - short press:
02:34:45:661 APS-DATA.indication srcAddr: 0x613d, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 191, rssi: -67

Down - long press:
02:40:02:070 APS-DATA.indication srcAddr: 0x613d, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 191, rssi: -67
02:40:02:158 APS-DATA.indication srcAddr: 0xa5a1, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:02:159 ZCL attribute report 0x000B3CFFFEFAD82E for cluster 0x0008, ep 0x01
02:40:02:159 APS-DATA.request id: 225, addrmode: 0x02, addr: 0xa5a1, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
02:40:02:182 APS-DATA.indication srcAddr: 0x61c4, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:02:183 ZCL attribute report 0x000D6FFFFEAF9D14 for cluster 0x0008, ep 0x01
02:40:02:188 APS-DATA.request id: 228, addrmode: 0x02, addr: 0x61c4, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
02:40:02:212 APS-DATA.indication srcAddr: 0xfd84, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:02:214 ZCL attribute report 0x086BD7FFFE06BB89 for cluster 0x0008, ep 0x01
02:40:02:214 APS-DATA.request id: 232, addrmode: 0x02, addr: 0xfd84, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 3 len: 5 tx.options 0x00
02:40:02:251 APS-DATA.indication srcAddr: 0x0919, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:02:252 ZCL attribute report 0x086BD7FFFE065A7E for cluster 0x0008, ep 0x01
02:40:02:252 APS-DATA.request id: 235, addrmode: 0x02, addr: 0x0919, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 4 len: 5 tx.options 0x00
02:40:02:272 APS-DATA.indication srcAddr: 0x55bc, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:02:273 ZCL attribute report 0x000B3CFFFEF931EB for cluster 0x0008, ep 0x01
02:40:02:273 APS-DATA.request id: 238, addrmode: 0x02, addr: 0x55bc, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 5 len: 5 tx.options 0x00
02:40:02:321 APS-DATA.indication srcAddr: 0x5378, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:02:322 0x14B457FFFE3208F1 level 254 --> 251
02:40:02:323 ZCL attribute report 0x14B457FFFE3208F1 for cluster 0x0008, ep 0x01
02:40:02:324 APS-DATA.request id: 241, addrmode: 0x02, addr: 0x5378, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 6 len: 5 tx.options 0x00
02:40:02:410 APS-DATA.confirm id: 225, status: 0x00 SUCCESS
02:40:02:410 APS-DATA.confirm request id: 225 -> erase from queue
02:40:02:427 APS-DATA.confirm id: 228, status: 0x00 SUCCESS
02:40:02:427 APS-DATA.confirm request id: 228 -> erase from queue
02:40:02:489 aps request id: 225 finished, erase from queue
02:40:02:519 APS-DATA.confirm id: 232, status: 0x00 SUCCESS
02:40:02:519 APS-DATA.confirm request id: 232 -> erase from queue
02:40:02:533 APS-DATA.confirm id: 235, status: 0x00 SUCCESS
02:40:02:533 APS-DATA.confirm request id: 235 -> erase from queue
02:40:02:569 aps request id: 228 finished, erase from queue
02:40:02:610 APS-DATA.confirm id: 238, status: 0x00 SUCCESS
02:40:02:610 APS-DATA.confirm request id: 238 -> erase from queue
02:40:02:624 APS-DATA.confirm id: 241, status: 0x00 SUCCESS
02:40:02:624 APS-DATA.confirm request id: 241 -> erase from queue
02:40:02:679 aps request id: 232 finished, erase from queue
02:40:02:731 aps request id: 235 finished, erase from queue
02:40:02:809 aps request id: 238 finished, erase from queue
02:40:02:889 aps request id: 241 finished, erase from queue
02:40:03:234 APS-DATA.indication srcAddr: 0x5378, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:03:235 0x14B457FFFE3208F1 level 251 --> 213
02:40:03:236 ZCL attribute report 0x14B457FFFE3208F1 for cluster 0x0008, ep 0x01
02:40:03:236 APS-DATA.request id: 250, addrmode: 0x02, addr: 0x5378, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
02:40:03:302 APS-DATA.confirm id: 250, status: 0x00 SUCCESS
02:40:03:302 APS-DATA.confirm request id: 250 -> erase from queue
02:40:03:369 aps request id: 250 finished, erase from queue
02:40:04:238 APS-DATA.indication srcAddr: 0x5378, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:04:239 0x14B457FFFE3208F1 level 213 --> 174
02:40:04:240 ZCL attribute report 0x14B457FFFE3208F1 for cluster 0x0008, ep 0x01
02:40:04:240 APS-DATA.request id: 3, addrmode: 0x02, addr: 0x5378, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
02:40:04:293 APS-DATA.confirm id: 3, status: 0x00 SUCCESS
02:40:04:293 APS-DATA.confirm request id: 3 -> erase from queue
02:40:04:329 aps request id: 3 finished, erase from queue
02:40:05:005 Current channel 15
02:40:05:028 Device TTL 1087 s flags: 0x7
02:40:05:240 APS-DATA.indication srcAddr: 0x5378, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
02:40:05:241 0x14B457FFFE3208F1 level 174 --> 136
02:40:05:241 ZCL attribute report 0x14B457FFFE3208F1 for cluster 0x0008, ep 0x01
02:40:05:241 APS-DATA.request id: 13, addrmode: 0x02, addr: 0x5378, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
02:40:05:302 APS-DATA.confirm id: 13, status: 0x00 SUCCESS
[... 107 lines cut ...]
02:39:06:234 APS-DATA.indication srcAddr: 0x484c, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -58
02:39:06:235 ZCL got data for node=0x484C, cl=0x0006, at=0x0000, status=0x00, type=0x10
02:39:14:049 APS-DATA.request id: 117, addrmode: 0x03, addr: 0x7cb03eaa0a002d42, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x03 queue: 1 len: 5 tx.options 0x00
02:39:14:101 APS-DATA.confirm id: 117, status: 0x00 SUCCESS
02:39:14:102 APS-DATA.confirm request id: 117 -> erase from queue
02:39:14:121 aps request id: 117 finished, erase from queue

Centre/Circle - short press:
2:36:56:594 APS-DATA.indication srcAddr: 0x613d, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 175, rssi: -69
02:36:57:591 APS-DATA.indication srcAddr: 0x613d, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 167, rssi: -70

Centre/Circle - long press:
02:41:38:209 APS-DATA.indication srcAddr: 0x613d, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 170, rssi: -47
02:41:39:203 APS-DATA.indication srcAddr: 0x613d, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -47
02:41:41:200 APS-DATA.indication srcAddr: 0x613d, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -50

===================================

[image]
[image]
[image]
[image]
[image]
[image]
[image]

henkkakay, 2019-11-17:
Appears to be similar to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1893

danzbox, 2019-11-17:
And also https://github.com/dresden-elektronik/deconz-rest-plugin/issues/294
and I don't get the lack of support. This is not some unpopular noname sensor but from Osram Smart+ Series. 
@manup in #294 there is already a link to the solution by kkoenig. What is needed from the Community to add support of this switch.

alloro72, 2019-12-15:
How to use the switch mini without zigbee2mqtt?

henkkakay, 2020-02-09:
whats the etiquette for "up" in github? Anyway - this is still relevant

hensoko, 2020-02-11:
👍

alloro72, 2020-02-11:
Please add support to Osram Switch mini on Deconz
Thank you

capellavacancies, 2020-02-25:
I can see this switch on deconz but can't see it in phoscon app

Larx, 2020-03-09:
Please support this device. It is a low-price solution from a major manufacturer with the selling point of being kind-of semi-outdoor usable.

jlangebek, 2020-03-12:
Please add support to Osram Switch mini on Deconz
Thank you

jeanclode76, 2020-03-16:
Hello,
yes please add support to Osram+ Switch mini on Deconz !
Thanks

Florian-Schmidt, 2020-04-14:
Hello, 

already voted for Osram 4 button switch but I would appreciate Osram Smart+ Switch also! please add these switches.

olicooper, 2020-04-29:
I've waited 2 years for support, is there a guide to follow to add support ourselves?
I don't quite understand where the lack of support lies - is it in the rest plugin, deconz firmware or the core application? There isn't enough documentation to be able to understand and fix this ourselves imo.
Can we suppress the button controlling all lights via the GUI?

Larx, 2020-04-30:
I'd like to support, too, but as nobody "official" seems interested in this issue, I'm lacking the knowledge where to start.

Florian-Schmidt, 2020-04-30:
It's strange that nothing happens from deconz side on the issues. I'm watching two issues linked other issues, seems that there are enought persons who want to use Osram components with deconz. 

The other issue is #374 

@olicooper These repositories are interesting as they provide python scripts to handle Osram mini switch and 4 button switches. I hadn't found them yet.
The 4 button switch repository (chalford-st) seems to work with home assistants zigpy integration (have seen this integration during last HA beta).
Until today I added all my devices via deconz (Phoscon WebGUI) and I didn't try home assistant zigpy yet. Perhaps this could be an option and/or using these repositories as custom_components.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open! 

Continue in #2763

closed 2020-06-06 by Mimiix