dresden-elektronik/deconz-rest-plugin #2084: TRÅDFRI bulb GU10 WS 400lm and TRÅDFRI wireless dimmer doesn't work together

Tobiaslindberg, 2019-11-17:
Hello!

I'm running conbee 1 on windows with about 40 nodes and just bought 3 new `TRÅDFRI bulb GU10 WS 400lm` and added them to the network. These new lights works great from within the Phoscon app and via Home Assistant but when I'm trying to add them to a `TRÅDFRI wireless dimmer` they don't respond to movement of the dimmer. If I add any other lights to the dimmer, those lights responds to dimming up/down and turning on/off but the new bulbs doesn't. I have tried to re add them multiple times and tried different `TRÅDFRI wireless dimmer` but I can't get the new bulbs to work.

Phoscon `V2_05_70`
Firmware `0x26330500`
<details>
 <summary>Images</summary>
(Color temperature light 26 is one of the new lights)

[image]

[image]

[image]

[image]

[image]
</details>

<details>
 <summary>List of devices</summary>
.

> 2x TRADFRI bulb E27 WS clear 950lm - v 1.2.217
>3x TRÅDFRI control outlet - v 2.0.019
>1x TRADFRI transformer 30W - v 1.2.245
>12x TRADFRI bulb GU10 W 400lm - v 1.2.214
>1x TRÅDFRI bulb E27 CWS opal 600lm - v 1.3.009
>2x Philips LTC001 - v 1.16.1_r19181
>3x TRÅDFRI bulb GU10 WS 400lm - v 2.0.022
>11x TRÅDFRI wireless dimmer - v 1.2.248
>4x TRÅDFRI on/off switch - v 2.0.020
>1x Philips Dimmer Switch - v 5.45.1.17846
>1x TRÅDFRI Motion sensor (old) - v 1.2.214
>2x TRÅDFRI Motion sensor - v 2.0.022
>3x Xiaomi Window / Door Sensor - v 20161128
</details>

Has anybody seen this before or know how to get it working?

Kane610, 2019-11-17:
Unfortunately the wireless dimmer sucks and Noone should use it

Tobiaslindberg, 2019-11-17:
@Kane610 Thanks for you non helpful answer.

After further testing I can see that if I add the new bulbs (`TRÅDFRI bulb GU10 WS 400lm`) to a `Philips Dimmer Switch` I can turn them on/off and dim them upp/down. So the issue seems to be either with the dimmer itself OR with some kind of comparability issue in deCONZ.

closed 2020-03-23 by stale[bot]