dresden-elektronik/deconz-rest-plugin #2085: OSRAM

muchasuerte, 2019-11-17:
Hello, 

I'm traing to update OSRAM Plug01 configuration :

it seems not working by GUI and I got that this configuration should be made by REST API how I can discover how to do it?

curl -H 'Content-Type: application/json' -X PUT -d '{"???": '???'}' -i 'http://<phoscon>/api/<api-key>/lights/12'

I would like configure it setting the PowerOn -> Previous

>   "manufacturername": "OSRAM",
  "modelid": "Plug 01",
  "name": "Light 12",
  "state": {
    "alert": "none",
    "on": true,
    "reachable": true
  },
  "swversion": "V1.05.09",
  "type": "On/Off plug-in unit",

I got these info but I was not able to figure out how push the configuration
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1014#issuecomment-445452409

is it possible? or is ti a device limitation?

Smanar, 2019-11-17:
I have tried on deconz and I have "unsupported attribute" for reading and "failed" for writing, so I think this device don't support this feature.

ebaauw, 2019-11-17:
OSRAM use a manufacturer-specific cluster to set the poweron defaults, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1833.

muchasuerte, 2019-11-17:
I did a test by following the direction, I have replaced the general.xml but it doesn't work I'm retrieving the error below

[image]

I'm using deCONZ Version -  2.05.67, should I have to update it?

ebaauw, 2019-11-17:
No, updating deCONZ won't help here.  You might want to check if the plug firmware can be upgraded.  Maybe the plug simply doesn't support poweron settings.

Come to think of it, poweron settings on a plug seem kinda odd.  I suppose the only valid use case if you want them to come on automatically after a power failure.

closed 2020-03-23 by stale[bot]