dresden-elektronik/deconz-rest-plugin #2088: Add support for Elko Thermostat

corvy, 2019-11-18:
I have one of these thermostat installed and successfully paired with Deconz. But I cannot control it in any way (I understand at least). So I can give you the data I have on this devices. 

[http://proff.elko.no/getfile.php/13124148/produkter/doc/EKO05680/Datasheet4523430.pdf](url)

[image]

[image]

[image]

[image]

[image]

ebaauw, 2019-11-18:
You should be able to set the target temperature by writing the _Occupied Heating Setpoint_ attribute 0x0012.  The _Local Temperature_ attribute 0x0000, reports the current temperature.  It doesn't look like the thermostat provides much more functionality.

You might want to check if you can create a binding from the _Thermostat_ cluster to the RaspBee/ConBee and setup reporting for the above attributes.

As thermostats are expose as `/sensors` resources, the device needs to be whitelisted for support by the REST API plugin.  For that, we'd need a screenshot of the _Node Info_ panel with the _Manufacturer Code_.

corvy, 2019-11-19:
Grand, then I could create a simple REST API integration from Home Asssistant to control it. I think I will be able to do that. The screenshot of Node Info: 

[image]

corvy, 2019-11-19:
[image]

[image]

Seems it has some more functionality :)

ebaauw, 2019-11-19:
I’ve got no clue what that does.

corvy, 2019-11-19:
Ok, well I can try it out on the API and see what I get from it. You can now whitelist it? Any way I could patch my install or does it require compilation?

corvy, 2019-11-20:
I think this has been added here: 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1291

closed 2019-11-20 by corvy