dresden-elektronik/deconz-rest-plugin #2089: add automatic polling aqara curtain motor

daddvok, 2019-11-18:
If you manually pull the curtains, the motor does not automatically give a new state, but if you count the cluster window covering, then the state is updated to the correct. Please add automatic polling of this device

pitikay, 2020-02-10:
I would love to see something like this implemented, my IKEA Fyrtur automatically updates its position every second or so, while the aqara motor needs around 5 minutes to send me an updated position (open/closed) sometimes! Can I somehow make a script that simply polls the current position every second?

Mimiix, 2020-06-05:
It seems this issue is resolved. If it is not, please re-open or create another issue.

closed 2020-06-05 by Mimiix