dresden-elektronik/deconz-rest-plugin #2090: HassIO Node-Red can not setup Conbee server

schumi2004, 2019-11-19:
So i have HassIO 2.12 running fully updated and setup with a Conbee2 stick and i can see all my devices.

Config deCONZ:
Note: for some reason it doesn't work with: "device": "/dev/ttyACM0" (no serial found)
```
{
  "device": "/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_blablablabla",
  "vnc_password": "vnc_pass_here"
}
```
Ports:
```
5900/tcp **disabled** deCONZ graphical desktop via VNC
80/tcp **40850** deCONZ API backend
8080/tcp **40860** deCONZ WebSocket Server
```

I also have Node-Red installed and added palette: **node-red-contrib-deconz**

And i followed this youtube video and it's pretty straight forward: https://www.youtube.com/watch?v=i3TiZiuNofM

But every attempt to connect to Conbee fails, localhost, HassIO IP, etc etc, correct ports  but no dice.
Any ideas why you can't setup Conbee server in Node-red from within HassIO?

Grevlingkroken, 2019-11-30:
Exactly the same setup here. I enable "Authenticate App" in Phoscon, and I only see "link button not pressed" when I try to configure node-red-contrib-deconz.

KekieM, 2019-12-12:
Same issue here. connection is made to the server and pressing the refresh device button shows the folowing in the device logging. 

12 Dec 21:13:51 - [info] [deconz-server:Phoscon-GW] discoverDevices: Requesting: http://192.168.178.100:40850/api/(xxxxxxxxxxxxxxx)

but no devices apear in the device list.

ajdezigns, 2019-12-23:
Same issue here Mine was working fine until a few days ago when I upgraded home assistant and decons plugin now it will not connect

drreality41, 2019-12-25:
Same problem here since last December update,
Node red authentication broken!

frenck, 2019-12-26:
Not sure why you created an issue here since this is an issue with the Hass.io add-on. Please create issues in the right place.

In this case, I would love to point you guys to the deCONZ add-on manual, since I've written a section about connecting Node-RED with the add-on with the newer releases.

@ dresden: Feel free to close this. I'm sorry our issues end up on your end.

schumi2004, 2019-12-26:
@frenck I closed the issue.
At the time i thought this was the best place to report it, my apologies.

p.s.: Could you give me a link to the section your talking about maybe since i can't find it.

closed 2019-12-26 by schumi2004

frenck, 2019-12-26:
Here you go: <https://github.com/home-assistant/hassio-addons/blob/master/deconz/README.md#using-the-deconzphoscon-api-with-another-add-on>

It also is shown when opening the add-on on your Home Assistant frontend

schumi2004, 2019-12-26:
@frenck thanks for the link but the instructions mention not to use the settings mentioned to set it up within homeassisrant which this ticket was all about. Can you clarify if it should work or not within homeassisrant?

ajdezigns, 2019-12-29:
@frenck I’ve done all that and it still will not reconnect I’ve been using deconz and node red for almost a year but not sure why it won’t work as before

omygolly, 2020-01-03:
I'm having exactly the same issue. Hassio, deconz, node-red. It's been working fine for 6 months but I'm now getting **WebSocket error: Error: connect ECONNREFUSED** errors. Can't access API either. Any ideas?

omygolly, 2020-01-03:
You can fix this by resetting the config and network settings in the Deconz-plugin in Hassio and start over with a fresh one.

schumi2004, 2020-01-03:
@omygolly 
I think i did that already and i have given up on this.
The only thing i see is : 11:19:03:864 gateway locked
And no matter what i configure it just won't connect, currently not even an error.

EddyErkel, 2020-04-16:
This worked for me (many thanks to Bart van Soest): 
[https://community.home-assistant.io/t/home-assistant-community-add-on-node-red/55023/764](url)

Open SSH terminal (via the SSH & Web Terminal add-on under Hassio)
Open the Phoscon App (open web-ui in the deCONZ add-on under Hassio)
In the Phoscon App goto: Hamburger menu -> Gateway -> Advanced -> Click ‘Authenticate app’ (This will open the gateway for 60 seconds)
In the SSH terminal type:
curl -d ‘{“devicetype”: “[give the application name, for instance Node-RED]”}’ -H “Content-Type: application/json” -X POST http://[your local ip address for Hassio / deCONZ]:40850/api
This should result in:
[{“success”:{“username”:"[this is the api-key]"}}]#
Goto Node-RED and install the ‘node-red-contrib-deconz’ pallette
Drag the ‘deconz-event’ node in a flow
Fill in the server details:
Name: Conbee2
IP Adress: [your local ip address for Hassio / deCONZ]
Port: 40850
API key: [the api-key from above]
WebSocket Port: 8081
SSL: not selected

jrnhrmn, 2020-05-15:
> I could not get this to work:
> https://github.com/home-assistant/hassio-addons/blob/master/deconz/README.md#using-the-deconzphoscon-api-with-another-add-on
> 
> * To allow these add-ons to connect to deCONZ, use the following settings:
>   Host: core-deconz    (should core-deconz be typed literally? That and local IP did not work)
>   (API) Port: 40850
>   WebSocket Port: 8081
> 
> This however worked for me (many thanks to Bart van Soest):
> https://community.home-assistant.io/t/home-assistant-community-add-on-node-red/55023/764
> 
> * Open SSH terminal (via the SSH & Web Terminal add-on under Hassio)
> * Open the Phoscon App (open web-ui in the deCONZ add-on under Hassio)
> * In the Phoscon App goto: Hamburger menu -> Gateway -> Advanced -> Click ‘Authenticate app’ (This will open the gateway for 60 seconds)
> * In the SSH terminal type:
>   curl -d ‘{“devicetype”: “[give the application name, for instance Node-RED]”}’ -H “Content-Type: application/json” -X POST http://[your local ip address for Hassio / deCONZ]:40850/api
> * This should result in:
>   [{“success”:{“username”:"[this is the api-key]"}}]#
> * Goto Node-RED and install the ‘node-red-contrib-deconz’ pallette
> * Drag the ‘deconz-event’ node in a flow
> * Fill in the server details:
>   Name: Conbee2
>   IP Adress: [your local ip address for Hassio / deCONZ]
>   Port: 40850
>   API key: [the api-key from above]
>   WebSocket Port: 8081
>   SSL: not selected

Tried this but it kind of looks like it will not accept an A class network of 10.74.5.0/24

EddyErkel, 2020-05-16:
For me it worked. Is the IP address you used for '[your local ip address for Hassio / deCONZ]' the same as the one you use to access your Hassio installation (e.g. like http://hassio-ip:8123/)?
It should not be the IP address of the deCONZ itself, well not when you have installed deCONZ as an add-on to Home Assistant.
Are you able to access  your deCONZ using the deCONZ port (e.q. like http://hassio:40850/)? 
If not, maybe you configured a different port for you deCONZ?
Hope this helps.

schumi2004, 2020-05-16:
Worked perfect here and almost using the same subnet without issues.

jrnhrmn, 2020-05-16:
I finally got it working. Got the API key and added the server but apparently needed to restart Node Red. Now it works fine. Thanks.

thelefty, 2020-05-17:
> For me it worked. Is the IP address you used for '[your local ip address for Hassio / deCONZ]' the same as the one you use to access your Hassio installation (e.g. like http://hassio-ip:8123/)?
> It should not be the IP address of the deCONZ itself, well not when you have installed deCONZ as an add-on to Home Assistant.
> Are you able to access your deCONZ using the deCONZ port (e.q. like http://hassio-ip:40850/)?
> If not, maybe you configured a different port for your deCONZ?
> Hope this helps.

I cannot access deconz using the http://MYIPADDRESS:8123/. I'm trying to do what you said with the ssh but this -----------> curl -d ‘{“devicetype”: “[give the application name, for instance Node-RED]”}’ -H “Content-Type: application/json” -X POST http://[your local ip address for Hassio / deCONZ]:40850/api ------ isn't working for me as well. Any help would be appreciated.