dresden-elektronik/deconz-rest-plugin #2091: Can deconz upgrade firmware of Osram bulbs?

NeighborGeek, 2019-11-19:
I bought a raspbee and have just successfully paired one of my ikea bulbs and updated the firmware.  I have a bunch of Sylvania Osram bulbs that I'd like to be able to update as well.  Can deconz do that?  If so, where do I download the firmware files from?

robidog, 2019-11-19:
Incidentally this was posted on Twitter a few minutes ago:
[https://phoscon.de/en/support#ota-update-osram-devices](https://phoscon.de/en/support#ota-update-osram-devices)

Coincidence?

NeighborGeek, 2019-11-19:
Sweet!  I'll look forward to trying it this evening.

NeighborGeek, 2019-11-20:
Unfortunately, no dice so far.  Following that process, I downloaded the firmware matching my bulb's image type (0x000F) and saved it to the OTAU folder.  I selected the bulb in deconz, selected the file, and at that point the bulb's progress field changed to PAUSED.  Then I clicked the update button, and the progress changed to IDLE, and stayed that way.  

I tried restarting deconz, in case it checks the OTAU files at startup, and I power cycled the bulb as well.  I do see one thing that could be an issue - Looking at the node info for this bulb, the manufacturer id is listed as 0xbbaa, while the image file lists a manufacturer ID of 0x1189.  I'm guessing that deconz recognizes that the bulb and firmware have different manufacturer IDs and won't apply the mismatched firmware.  

Does this mismatch mean that I actually do have different bulbs from what the firmware is intended for, or is there something else going on here? 

For anyone else with osram bulbs, what is their manufacturer id?

 
[image]
[image]

NeighborGeek, 2019-11-20:
Paying closer attention, I realized that the filename of image type 0x000F indicates that it's for a BR30 bulb, while the actual bulbs I have are A19's, so it's probably not the right firmware anyway.  The only types listed for A19 on the ledvance site are 0x000C and 0x000D though, so perhaps they just don't offer the one for my bulbs.  

FWIW, the bulb in question is model # 73824

mysteryan, 2019-11-20:
A bit sidetrack, but since the firmware is available, would it be possible to flash EU firmware on the US bulbs.  In other words, change the primary protocol used from ZHA to ZLL that EU uses?

robidog, 2019-11-22:
@NeighborGeek I just noticed this in the FAQ: "This guide requires the deCONZ GUI with at least version 2.5.71". Have you updated to that version yet?

NeighborGeek, 2019-11-22:
Yes, I am using 2.05.71. Thanks for asking though, I hadn’t seen that part.

mysteryan, 2019-11-29:
> Incidentally this was posted on Twitter a few minutes ago:
> https://phoscon.de/en/support#ota-update-osram-devices
> 
> Coincidence?

Works great.  I upgraded all my "LEDVANCE" manufactured bulbs.  Download page doesn't provide firmware for older "OSRAM" manufactured bulbs though.

hellcry37, 2019-12-08:
for me update of osram bulbs works but version show remains the same, doesn't update in phoscon app

manup, 2019-12-08:
If the version number isn't updatet you can you please try to read it manually.

* In deCONZ open the *Cluster Info* panel
* Expand the cluster list of the node (right side little circle)
* Click on the *0000 Basic* cluster in the list
* Press the **read** button in the Cluster Info panel

nassost79, 2019-12-14:
Hi all, I'm running HassOS on a Raspberry pi4 with Conbee 2. 
I want to update the firmware of my Osram bulbs. I 've found the file necessary from the Ledvance site (0x0013) and downloaded it in my PC.  How do I transfer it to the STD OTAU plugin?

Ltek, 2019-12-14:
@manup 
Just started playing with the FW update option. I noticed none of my 10 devices update the info in the STD OTAU screens. Would be great if you can have it cache the info... doing an automatic query every time deConz launches?

@nassost79 
the guide is here... https://phoscon.de/en/support#ota-update-osram-devices

nassost79, 2019-12-14:
@Ltek I know about this guide, it's the one I'm trying to follow.

My question is about this particular step ("b") where I must point the plugin to the firmware file location. I have downloaded the firmware file to my windows PC but I have no way to copy it where the plugin will be able to find it.

[image]

Ltek, 2019-12-15:
@nassost79  ... you installed the deCONZ app itself, thus you got the setup file there somehow.

nassost79, 2019-12-15:
@Ltek I installed via the Hass.io add-on store. I am running HassOS, if you didn't notice in my initial post. Maybe this is the problem, I don't know.

[image]

hellcry37, 2019-12-15:
@nassost79 i had this issue also.

You are running an addon, and that addon is basically a docker container. so to be able to do this update you have to download file from web and copy it on linux host (host machine, rasberry pi or whatever you use) then from there you have to copy it in the docker container exactly in the correct location /data/otau/

My advice don't do it, deconz update suks badly it wont show you in interface your new firmware version and all kind of shits like this.

dedepene, 2020-01-12:
> My advice don't do it, deconz update suks badly it wont show you in interface your new firmware version and all kind of shits like this

Ermmm....no. Just updated 3 Osram bulbs without a hiccup, accurate version showed up in the phoscon app.

Ltek, 2020-01-18:
I updated 3 Smart+ strips controllers... worked fine for me. 

@manup 
Would love to see a better UI (wizard) for doing it... as the UI is tedious and not intuitive. Best if it can be done with Phoscon.

closed 2020-05-25 by stale[bot]