dresden-elektronik/deconz-rest-plugin #2092: Add Support for Bilisty Rotary Dimmer Switch

mdolnik, 2019-11-19:
[image]

I received this dimmer switch via Alibaba.

It doesn't seem to be on it's own website www.bilisty.com but the link to the Alibaba page is here: https://www.alibaba.com/product-detail/zigbee-switch-module-solution-light-dimmer_62191423426.html

I am able to see the device in deCONZ and Phoscon, but it doesn't physically respond to any commands and it shows as a simple on/off switch instead of a dimmer switch.

[image]

[image]

[image]

[image]

Anytime I send any commands in deCONZ (eg Going to On/Off cluster, and pressing `exec` on any commands) it will say `executing` then it will say `success`, but the lights do not change.

Any suggestions?

ebaauw, 2019-11-19:
> but it doesn't physically respond to any commands 

You did connect dumb lights by wire, didn’t you?  Do they react when you press the button?  Rotate the dial?  After pressing the button, has the value of the _OnOff_ attribute changed?  You might need to press _Read_ to update the attribute.

After sending a command, does the value of the _OnOff_ attribute change?

> it shows as a simple on/off switch instead of a dimmer switch.

That’s courtesy of the dimmer firmware, nothing deCONZ can do about that.  The (for Alibaba usually vague) product description could be read To indicate that the level can only be set mechanically.  Your lights might not turn on when the dial is in the lowest, 0W setting.

[image]

They also suggest Bluetooth low energy support and, per Q4 2019 ZWave support.  The dimmer does have an _OTAU_ cluster, so, at least in theory, there could be (future) firmware updates.

mdolnik, 2019-11-19:
> You did connect dumb lights by wire, didn’t you?
I don't know what this means, but I have the 120v dimmer output connected to a triac dimmer then to 12V LED lights.

> Do they react when you press the button? Rotate the dial?
Yes, right now it works as a normal dimmer switch.

> After pressing the button, has the value of the OnOff attribute changed? You might need to press Read to update the attribute.
No, when I press the physical button the `OnOff` attribute in the `On/Off` cluster does not update, even after pressing the `read` button.

> After sending a command, does the value of the OnOff attribute change?
Yes, the `OnOff` value will update accordingly when sending On/Off/Toggle commands from deCONZ, I do not have to press the read button to have it update. But the actual lights do not change.

> > it shows as a simple on/off switch instead of a dimmer switch.
> That’s courtesy of the dimmer firmware, nothing deCONZ can do about that. The (for Alibaba usually vague) product description could be read To indicate that the level can only be set mechanically.
That's unfortunate, but not a deal breaker for me, I would be ok with only being able to control on/off remotely, at least for now.

> Your lights might not turn on when the dial is in the lowest, 0W setting.
I've noticed this with my LED setup, but the dimmer remembers the last brightness it was on, and it also has the ability to set the minimum brightness so this is not an issue. Plus the rotary doesn't have a physical "lowest setting" as the knob freely turns infinitely.

Is there anything else I can try to get this to work?

I may see if I can get the company itself to have a look at this thread, maybe they can chime in or offer solutions.

ebaauw, 2019-11-19:
> Plus the rotary doesn't have a physical "lowest setting" as the knob freely turns infinitely.

That’s hopeful: it does remember the level electronically, rather than mechanically.

Did you try connecting to it over bluetooth?  Does it come with a bluetooth app?  Maybe that’ll support firmware updates (like the latest Hue bulbs).

It looks like the ZigBee function is “detached” from the dimmer function.  Some Xiaomi switches have a manufacturer-specific attribute to set that (although by default they are attached).  Could be the case for this dimmer, but you’d need to sniff the ZigBee traffic with its native gateway to find out.  Maybe an extensive Internet search could find some integration with another home automation system that can tell us if any such setup would be needed?

mdolnik, 2019-11-22:
> Did you try connecting to it over bluetooth? Does it come with a bluetooth app? Maybe that’ll support firmware updates

After contacting the manufacturer, they mentioned that the bluetooth version is a completely different product so that doesn't seem to be an option for me. Doesn't deConz have the ability to do firmware updates? This switch does have an OTAU cluster.

> It looks like the ZigBee function is “detached” from the dimmer function. Some Xiaomi switches have a manufacturer-specific attribute to set that (although by default they are attached). Could be [...]

If I were to get the gateway that they're selling, how would I go about this? Would I use ZShark? I have never done anything like this and wouldn't know where to start.

I have asked the manufacturer to get their tech department to have a look at this thread, but I don't know they likeliness they will weigh in or not. Fingers' crossed.

closed 2020-03-28 by stale[bot]