dresden-elektronik/deconz-rest-plugin #2093: Xiaomi Bewegunsmelder mit fehlerhaftem Batteriestatus

Uwe1984, 2019-11-19:
Der Batteriestatus verändert sich nicht und ist unplausibel. 
Ich habe einiger dieser Sensoren und der Batteriestatus wäre für eine Überwachung wichtig. 

[image]

Smanar, 2019-11-20:
It's something normal, the device send a battery voltage, so Deconz make a convertion to a percentage.
It's the normal working mode for a battery. 
You have just buy the battery : 13.5V
One year later : 12V
6 mont later : 10 V
1 month later 4V.

A battery discharge is not something regular. You can have 100% for 1 year, and the device go to 95% to 50% in 1 month.

Uwe1984, 2019-11-25:
Thank you for your answer. I can understand that and will observe it for a while. I had other values with the CC2531 Zigbee Stick. That's why I thought it was a issue

closed 2020-03-31 by stale[bot]