dresden-elektronik/deconz-rest-plugin #2094: Add Support for Xiaomi Smart Plug (TW Country model)

stccmaya, 2019-11-20:
I received this Xiaomi Smart Plug.

But this model: lumi.plug.mitw01 (TW Country version)

Add Support this lumi.plug.mitw01 model?

[image]

[image]

[image]

stccmaya, 2019-12-09:
no any respone : (

stccmaya, 2020-01-14:
anyone help me

mhin76, 2020-01-16:
I would also be very interested in this add on. I am using a LUMI.plug with power configuration visible in the deCONZ app (version 2.05.71), but when I log in to phoscon app or to my Node Red, I do not see any power consumption value. I see in Node Red a ZHA:Consumption device/sensor, but that never returns any value.

SwoopX, 2020-01-27:
@stccmaya Please provide a screenshot from Endpoint 02 (Home automation - Mains powered outlet) and 03 (Home automation - 0053) for each of the Analog input cluster with the attributes being read before taking it.

stccmaya, 2020-01-29:
@SwoopX 

Home automation - Mains powered outlet (Analog input)

[image]

Home automation - 0053 (Analog input)
[image]

SwoopX, 2020-01-29:
Ok, thanks. As I don't know, how the device was used: Is it reasonable to conclude (based on the first screenshot) that the applied power is 0,5W? The second screenshot is the candidate for consumption. Would it make sense that 15.11 kW have been consumed since the device has been put into service? I need to figure out the unit of measurement and for that confirm reasonable consumption values...

stccmaya, 2020-01-30:
Yes, I think there is nothing wrong with the unit, because this device is a microwave oven.

The first screenshot machine is in standby state, so the power consumption is very low power consumption.

I will use it in the Homeseer system to make it support power consumption

mhin76, 2020-01-30:
Hi Stccmaya. I am using the lumi.plug with date code 10-11-2016. Under "Mains Power Outlet" I read the current power consumption (I thought it is Watt). The "0053" shows constant 0.83 (not sure about the unit). I connected this zigbee previously to my raspberry pi via a cc dongle using MQTT. I know that it was reading also the temperature (there is a temp sensor in that plug).
So not sure why it does not work under the ConBee II gateway. Temperature is not visible.
Does it mean with a new deCONZ rest plugin the power consumption is also sent to Node Red via the node-red-contrib-deconz module or would that developer first have to add it?

SwoopX, 2020-01-30:
@stccmaya thanks!

@mhin76 let's not start mixing things up and get the topic here sorted out first.

> Does it mean with a new deCONZ rest plugin the power consumption is also sent to Node Red via the node-red-contrib-deconz module or would that developer first have to add it?

From what I see in the code, everything's already there for the consumption sensor. However, I believe there's a small mistake:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/10810836905fbcd1e7c9fafcaf386702c9fed41f/de_web_plugin.cpp#L6607-L6638

I'd have expected line 6625 to be starting with "if", not "else if", and therefore simply jumping over the consumption code...

@ebaauw do you want to share your thoughts on this?

ebaauw, 2020-01-30:
I think this code runs inside a loop over all sensor resources, so `else if` is good: a single resource only has one type.

SwoopX, 2020-01-31:
Hm, you're right. Forgot to have a look at it as a whole. Could the data type play a role by the way? I guess consumption is everywhere else defined as quint64 instead of qint64.

closed 2020-06-07 by stale[bot]