dresden-elektronik/deconz-rest-plugin #2095: All devices unreachable after update to .71

peer69, 2019-11-20:
Yesterday I have updated to .71 which caused a connection loss to all my zigbee devices except some sensors directly connected to the gateway. No routers did connect nor any sensors previously connected through a router. 
As a precaution I take snapshots before any update but unfortunately restoring the snapshot did nothing. The connections did not restore. I also tried to switch back to my old Conbee I, restoring from a backup and restoring the network configuration. did nothing. 

Powercycling devices did nothing, no visible reaction in deconz gui. 
Ultimatley I spent the night resetting all of my 70 devices. I did not remove them from deconz I just resetted them (which was still a pain especially for hidden sensors and osram bulbs/lightstrips) and they joined back the network without any issues except for some sensors which lost their custom name and changed back to the default. 

Unfortunately I did this back at the old snapshot which is running .69. 
Did anyone experience this issue with .71? I read some issues here with similar symptoms but no solution or any clue what might cause this. In the past I was regularly updating without any issues but for now i might be better off staying with a now working release until the cause of this issue is known. 

OS: Ubuntu 18.04.3 running on ESXi 7.5
Deconz: 2.05.69
Gateway: Conbee II / 264A0700

osos, 2019-11-20:
same for me once i tried .70, switched back to .69 and re-paired and have not dared to update since.

yvessa, 2019-11-20:
here is what I did to solve the issue 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2076#issuecomment-555267213

peer69, 2019-11-21:
I tried that as well. Even unplugged the Conbee Stick as I didnt want to shut down the whole ESXi Host after the deconz vm was shut down. I will probably now set up a test machine with some of the devices as I have a second adapter but I obviously wont be able to replicate my productive environment so I dont know if this might prevent similar scenarios with further updates at all.
It really felt like this update caused the coordinator to drop all routers and sensors connected through routers from the network so this might be neighbor table related. I moved sensors formerly connected through routers near the gateway but this did nothing. Only sensors connected directly to the gateway before the update were reachable afterwards.

yvessa, 2019-11-22:
Yep almost sure it was related to updating from v2.05.69 to v2.05.70. The evening before the issue it ran perfectly. During the night it took the update automatically as I run it thru docker container with watchtower. Then, when I woke up almost no sensors were working.  

The only thing I havent mentioned before running back conbee II, I edited my config to pull v2.05.70.  So shutdown deconz, unplug/plug back routers, edit config to grab update v2.05.70 and start deconz.

I do backup everyday of my deconz folder config files, even putting back old config did not work. So issue was probably coming from conbee stick.

Too bad you had to redo your setup completely.  I have many outdoor sensors and it's winter here so that would have been my absolute last solution!

It took update v2.05.71 nicely yesterday.

muchasuerte, 2019-11-24:
I have the 2.05.67 and I have the same problem after a night I'm no more able to see any nodes until deconz restart, on my deploy I'm using the deb package 2.05.67-raspbian-stretch-stable

hpd96, 2019-11-24:
Hello,
same happens here! was updating many months since e.g.  deconz-2.05.31-qt5.deb  

updating using rasbian DBK from .69 to .70 --> no nodes connected visible in QT deconz GUI -- 33 nodes boxes visible but no rubberband from cluster 0000 to any node!!

downgrading to .69 from .70 --> again no nodes useable

nothing working in pwa Phoscon - same in openHAB 

deleting switch/plug device and re-installing did not worked :-(

installing .71 and again no nodes a/v

deleted
 .local/share/dresden-elektronik
and restart -> no nodes a/v

clicking on 
<a id="resetGatewayBtn" data-translateid="text_resetGateway">Gateway zurücksetzen
pairing working again - had to relearn 33 nodes 👎 

this is a A1 show stopper !

please check the releases .70 / .71 - have no further logs.

BR/
Peter

peer69, 2019-11-24:
Today I felt lucky and updated to 71. everything went fine. Still, having to repair all of the 60 nodes, some in places that die not make that easy was a real pain. Especially because I set up the network from scratch just some weeks ago when I was unable to migrate to a Conbee II. 
I know there is a beta logo on this software. Still this feels like it could use a change in the and test  cycle. Unfortunately it’s not really possible to build a production near test environment for home users.

yvessa, 2019-11-24:
I have been using deconz for about a year, first time I encountered a software update not going smoothly.  I guess I got lucky to get it up again w/o repairing... Repairing does not make sense when you have more than 10 sensors.  

Any official answer from developers of what might have caused these issues?

ademobile, 2019-11-26:
Since upgrading to .71 every two days or so I am now left with an unresponsive network (i.e. nothing responds) and I have to pull the power. I get the feeling that something is crashing. Where do I find any logs (that's if it has been caught!)?

peer69, 2019-11-26:
I guess @manup should be able to help with where to look and what to make of the log entries.

roaringbits, 2019-11-27:
Since upgrading to .71 my ikea outlets have started failing. First one, took it out of the network. Now the rest is going one by one (resetting while searching helps sometimes)

osos, 2019-12-01:
@manup and @ChrisHae are you looking into this?

yvessa, 2019-12-07:
Well network starting to do weird behaviour with .71.  

Yesterday 1 Ikea Smart Switch was going on/off really quick, was hearing the mechanical relay clicking really fast every minute or so. I unplugged/plugged back this switch, solved the issue. However I lost 2 others Ikea Switch from the network. I did reboot deconz, it worked but this morning these 2 ikea switches were not available anymore. I soft reset these 2 switches within deconz, they are back... Let see if they hold.  

I will need to remove my dehumidifier from ikea switch, switching on/off rapidly a dehumidifier compressor could destroy it.

yvessa, 2019-12-10:
I needed to shutdown deconz and unplugged/plugged back all ikea switches. Then restart deconz and I got back the nodes.

If I only unplugged/plugged back missing ikea switches, other ikea switches were falling. Same as soft resetting switches from deconz gui.

muchasuerte, 2019-12-15:
look like the issue is caused by the ikea switches, I have no problem with my zigbee network usually, just sometime when I unplug the ikea outlet then I have to restart the deconz service and wait some time to get back all the nodes again.

I did the firmware upgrade some week ago now I'm using the TRADFRI-control-outlet-2.0.022

closed 2020-04-20 by peer69