dresden-elektronik/deconz-rest-plugin #2096: Add Support for Livolo TI0001 Switch

Olepir, 2019-11-20:
Is it possible to add support for Livolo TI0001 Switch?
It uses some kind of custom Zigbee protocol

https://github.com/Koenkk/zigbee2mqtt/issues/592

Cyrille63, 2020-03-21:
Is it possible to add support for Livolo switches ?

MOE-JOE2, 2020-03-22:
+1 here

Smanar, 2020-03-22:
There is someone that can compile the code for some tests ?

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2036

Cyrille63, 2020-03-27:
I can do that, is there a procedure to compile ?

Cyrille63, 2020-04-02:
I have install a linux machine - Lubuntu 18.04.04 - deCONZ is working - I have installed hte development package, compilation is OK

I can do test

Smanar, 2020-04-02:
Lol, choose an issue, you don't need to post on both I receive double notification.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix