dresden-elektronik/deconz-rest-plugin #2097: Phoscon App als localhost keine Gateway Anzeige

JBSullivan, 2019-11-20:
Ich habe das Problem, wenn ich die Phoscon App an dem Rechner aufrufe, wo der ConBee2 Stick drin steckt ( localhost oder 127.0.01.), ich kein Menü für das Gateway angezeigt bekomme. Auch wenn ich die Netzwerk IP Adresse dieses Rechners am selben PC aufrufe, ist kein Menü für das Gateway zu sehen.

Rufe ich aber die Netzwerk IP des Rechner wo der Stick drin steck über einen anderen PC aus dem gleichen Netzwerk auf, bekomme ich das Gateway Menü angezeigt.

Zum Einsatz kommt die letzte deConz Version (2.05.71)

[image]

manup, 2019-11-20:
Hmm, this should work with localhost and 127.0.0.1.
Which browser is that?

Does the address http://127.0.0.1/pwa yield any Javascript errors in the browser developer console (F12)?

JBSullivan, 2019-11-20:
Browser is Edge in Version
[image]

No erros in the console, but hier is a screenshot of some warnings if i had followed the Gateway Link

[image]

manup, 2019-11-20:
Looks like the have some tricky HTML brackets here which this version of Edge doesn't parse. I'll put that on the todo list to check.

Can you please try with another browser or a recent Edge version (this here seems to be from 2018).

JBSullivan, 2019-11-20:
That`s (maybe) the next Problem - if i try annother Browser on the same PC (Chrome, Firefox) i MUST set a new Password. If i use the same Password i used at the Edge Browser there is no login possible.

b2un0, 2019-12-03:
relates to #2106

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix