dresden-elektronik/deconz-rest-plugin #2099: Conbee 2 stick stuck in "Joining..." state

victor-mikhalev, 2019-11-20:
OS Windows 10
```
[10228] 23:05:17:000 HTTP Server listen on address 0.0.0.0, port: 9090, root: C:/Users/Victor/AppData/Local/deCONZ/bin/plugins/de_web/
[10228] 23:05:17:015 CTRL. 3.24.0
[10228] 23:05:17:045 ZCLDB init file C:/Users/Victor/AppData/Local/dresden-elektronik/deCONZ/zcldb.txt
[10228] 23:05:17:055 gw run mode: normal
[10228] 23:05:17:055 GW sd-card image version file does not exist: C:/Users/Victor/AppData/Local/dresden-elektronik/deCONZ/gw-version
[10228] 23:05:17:055 DB sqlite version 3.24.0
[10228] 23:05:17:056 DB PRAGMA page_count: 30
[10228] 23:05:17:056 DB PRAGMA page_size: 4096
[10228] 23:05:17:056 DB PRAGMA freelist_count: 0
[10228] 23:05:17:056 DB file size 122880 bytes, free pages 0
[10228] 23:05:17:056 DB PRAGMA user_version: 6
[10228] 23:05:17:056 DB cleanup
[10228] 23:05:17:056 DB create temporary views
[10228] 23:05:17:057 don't close database yet, keep open for 900 seconds
[10228] 23:05:17:058 started websocket server at port 20877
[10228] 23:05:17:064 found node plugin: de_rest_plugin.dll - REST API Plugin
[10228] 23:05:17:066 found node plugin: de_signal_plugin.dll - Signal Monitor Plugin
[10228] 23:05:17:066 Otau image path does not exist: C:/Users/Victor/otau
[10228] 23:05:17:071 found node plugin: std_otau_plugin.dll - STD OTAU Plugin
[10228] 23:05:17:072 watch: C:/Users/Victor/AppData/Local/deCONZ/bin/plugins/de_web
[10228] 23:05:17:072 watch: C:/Users/Victor/AppData/Local/deCONZ/bin/plugins/de_web-src
[10228] 23:05:17:077 found node plugin: templ_plugin.dll - Templ Plugin
[10228] 23:05:17:146 Device firmware version 0x264A0700
[10228] 23:05:17:155 unlocked max nodes: 200
[10228] 23:05:17:232 Device protocol version: 0x010B
[10228] 23:05:17:240 new node - ext: 0x00212effff04db27, nwk: 0x0000
[10228] 23:05:17:242 CTRL [00:21:2e:ff:ff:04:db:27] not viable in zll.db, skip load cache (101)
[10228] 23:05:17:290 Current channel 11
[10228] 23:05:17:301 CTRL got nwk update id 1
[10228] 23:05:17:305 CTRL ANT_CTRL 0x03
[10228] 23:05:17:333 Device protocol version: 0x010B
[10228] 23:05:17:389 Current channel 11
[10228] 23:05:17:404 CTRL got nwk update id 1
[10228] 23:05:17:409 CTRL ANT_CTRL 0x03
[10228] 23:05:18:054 UPNP socket not bound, state: 0
[10228] 23:05:22:053 GW update firmware found: C:/Users/Victor/AppData/Local/dresden-elektronik/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
[10228] 23:05:22:053 GW firmware version: 0x264a0700
[10228] 23:05:22:053 GW firmware version is up to date: 0x264a0700
[10228] 23:05:22:262 don't close database yet, keep open for 900 seconds
[10228] 23:05:23:086 Announced to internet http://dresden-light.appspot.com/discover
[10228] 23:05:37:053 start reconnect to network
[10228] 23:09:22:055 GW firmware version: 0x264a0700
[10228] 23:09:22:055 GW firmware version is up to date: 0x264a0700
```

manup, 2019-11-20:
Is this already the latest version 2.05.71?
Do you use an USB extension cable?

If on 2.05.71 and the joining still persists we need to check the parameters.

ebaauw, 2019-11-20:
> [10228] 23:05:17:077 found node plugin: templ_plugin.dll - Templ Plugin

What's this plugin?

> Is this already the latest version 2.05.71?

Has that version been published to the `apt` repository?  I can find .71 of `deconz-dev`, but no .71 `deconz` (on Stretch nor on Buster).

manup, 2019-11-20:
>Has that version been published to the apt repository? I can find .71 of deconz-dev, but no .71 deconz (on Stretch nor on Buster).

Strange, yes it should be in the APT repository, I'll check it out.

>[10228] 23:05:17:077 found node plugin: templ_plugin.dll - Templ Plugin
What's this plugin?

That's a very basic template plugin to build the Phoscon App static html files, shouldn't be in the 2.05.71 release anymore.

victor-mikhalev, 2019-11-21:
> Is this already the latest version 2.05.71?

Seems to be the latest:
[image]

> Do you use an USB extension cable?

No.

> If on 2.05.71 and the joining still persists we need to check the parameters.

[image]
[image]

manup, 2019-11-21:
>>Do you use an USB extension cable?

>No.

I can only assume that the Zigbee radio can't access the medium (air) due interference.
This is quite common when no USB extension cable is used (see also: https://phoscon.de/en/support#conbee2-connection-issues)

In order to start the network the coordinator first checks if it can access the medium and looks for a rather silent channel from the channel mask.  If that fails it will continue so (since 2.05.70, before it was just given up after the first attempt).

TL;TR: Please try a USB extension cable or another USB slot.

victor-mikhalev, 2019-11-21:
> Please try a USB extension cable or another USB slot.

Connected it via extension cable, tried different USB ports. Nothing changed. Still "Joining...".

victor-mikhalev, 2019-11-22:
Tried connect to another PC. Same issue. Seems stick is brocken.

pandiloko, 2019-11-25:
I have the same problem. Also version 2.05.71 and config seems OK: 

[image]
[image]

Also tried the USB extension and other ports. I even tried with Debian Buster and Ubuntu (different computers). Device arrived today.

pandiloko, 2019-11-26:
New day, new luck. I tried yet another computer (Debian Buster) and everything worked out on first try. Well I did need to restart for the dialout group change to be applied (I guess X restart would be enough but whatever) but otherwise everything went smooth including detecting the first device. Tonight I'm testing again in the initially intended server.

pandiloko, 2019-11-26:
So back on the initial server I have the same problem. This is the output of dmesg:

```
[78496.755104] usb 1-4.4: new full-speed USB device number 34 using xhci_hcd
[78496.857188] usb 1-4.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[78496.859205] usb 1-4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[78496.861179] usb 1-4.4: Product: ConBee II
[78496.863184] usb 1-4.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[78496.865261] usb 1-4.4: SerialNumber: DE2129373
[78496.875104] cdc_acm 1-4.4:1.0: ttyACM0: USB ACM device
[78500.716593] usb 1-4.4: USB disconnect, device number 34
[78500.720783] cdc_acm 1-4.4:1.0: failed to set dtr/rts
[78500.960948] usb 1-4.4: new full-speed USB device number 35 using xhci_hcd
[78501.063312] usb 1-4.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[78501.065474] usb 1-4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[78501.067589] usb 1-4.4: Product: ConBee II
[78501.069684] usb 1-4.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[78501.071797] usb 1-4.4: SerialNumber: DE2129373
[78501.082072] cdc_acm 1-4.4:1.0: ttyACM1: USB ACM device
```
So it disconnects and connects again and the only error message is ` failed to set dtr/rts` . I saw some references in internet to som udev rules. Does anyone know this error?

pandiloko, 2019-11-27:
I got it to work finally by disconnecting other usb devices. I’m currently in the process of finding out any combination of ports that allows me to have more devices but at the moment it must be the ConBee alone to work. Pretty annoying. Hope it helps someone. Are there any driver configurations to address this issue?

closed 2020-04-02 by stale[bot]