dresden-elektronik/deconz-rest-plugin #2100: Request Support for Develco Heat Alarm (HESZB-120)

tyr84, 2019-11-20:
Develco smoke detector (SMSZB-120) is allready supported. Looks like the heat alarm is more or less the same device.

It can be added as of now, however it seems like only the "switch" is recognised by phoscon. There should also be a binary sensor (smoke/heat detected) and a temperature sensor.

[image]
[image]
[image]

[Technical manual](https://www.develcoproducts.com/media/3372/heszb-120-technical-manual-heat-alarm.pdf)

SwoopX, 2019-11-21:
Jep, from the cluster perspective, it looks identical. I can take care of integration, but it will probably take a week or two...

> It can be added as of now, however it seems like only the "switch" is recognised by phoscon. There should also be a binary sensor (smoke/heat detected) and a temperature sensor.

This one is on Phoscon. It needs to be implemented seperately. I'd also love to see the sensors popping up.

OT: Damn, I don't get why for all other sensors than the smoke the manufacturer code is read. I'd assume pairing went smoothly without any tricks? Would love to understand the differences then.

tyr84, 2019-11-30:
yep, went smooth. I got a feeling it might be some firmware issue on the smoke detectors themselves

SwoopX, 2019-11-30:
Thanks, thought so. Pretty sure I finally got the reason why. If I'm right, it's literally on incorrectly set bit causing all the trouble (for the smoke sensors)

SwoopX, 2019-11-30:
Just updated the open pull request for this device ;)

closed 2020-01-19 by tyr84

SwoopX, 2020-01-27:
@tyr84 is the device working ok?

SwoopX, 2020-11-29:
#3775