dresden-elektronik/deconz-rest-plugin #2104: Support request Sinope TH1123ZB & TH1124ZB

the-speedmax, 2019-11-21:
Hello,
I already own a Conbee II. I'm looking to add some thermostat to my setup.
Do you know if this model can work with deconz? Sinope TH1123ZB & TH1124ZB
These are Zigbee.
https://www.sinopetech.com/en/product/smart-thermostat-for-electric-heating-3000-w-zigbee/?setLang=en
Tks for your help!

Ackshiel, 2020-01-26:
Adding supporting information.
General
[image]
Thermostat
[image]
Rest of thermostat info
[image]
Temp measurement
[image]
simple metering
[image]
electrical measurement
[image]

TheHerb007, 2020-04-06:
Looking to get one of these, can I then assume this model is supported? Thanks!

Ackshiel, 2020-04-07:
> Looking to get one of these, can I then assume this model is supported? Thanks!

I was too impatient to wait for this to be supported, so I ended up going to zigbee2mqtt and it is working very well for me.

the-speedmax, 2020-04-07:
I did the same. Zigbee2mqtt.
I have 5 of them. Working fine.

julnobugs, 2020-04-07:
In the end, I did choose ZHA as I didn't knew if the TH1123ZB was properly supported or not by conbee/deconz. I wish it was.

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

MRobi1, 2020-09-24:
I currently have 15 of these with zigbee2mqtt. Thinking of moving to Deconz. Wondering if outside temperature display is supported here?

SwoopX, 2020-09-24:
How do you mean that? Sure, it's possible to further improve support if someone is willing to compile and test updated versions :)

MRobi1, 2020-09-25:
> How do you mean that? Sure, it's possible to further improve support if someone is willing to compile and test updated versions :)

The Sinope TH1123ZB & TH1124ZB support displaying the outside temperature on the 2nd line of the display. 

On zigbee2mqtt you publish the outside temp value to zigbee2mqtt/<FRIENDLY_NAME>/set/thermostat_outdoor_temperature so I've got an automation setup whenever the outside temp changes to publish it to the thermostats.
There's also a PR for Home Assistant to add a send_outside_temperature service for these devices in ZHA.
So it'd be great to get support added to Deconz as well. How exactly that would be accomplished is way beyond me though :)

SwoopX, 2020-09-25:
I took a look over at z2m and gee, the device has more capabilities than you can deduce from the screenshots. I guess most, if not all can be integrated, but that involves quite some extensions on the API. Apparently, there were also some clusters skipped in the screenshots.

If you're using HA, then that's the most uncomfortable combination. You cannot directly test my changes once ready but have to wait for us to merge the code, promote it to stable and then hope for HA folks to update as well (which is not in our hands).

amritk, 2020-10-21:
> How do you mean that? Sure, it's possible to further improve support if someone is willing to compile and test updated versions :)

I will order a few right now, I have the conbee 2. Will let you know

MRobi1, 2020-12-01:
@SwoopX Just thought I'd check in on this. I see some changes have been merged in Enhancements and fixes, but not exactly sure what those are. Has the ability to set outdoor temp on these devices been added? If so, how would one send the temperature reading to the device? 
Thanks!

amritk, 2020-12-01:
So I have been testing them out, for setting the temp and everything they work great. Would anyone know how to change the time on them?

TheHerb007, 2020-12-01:
I have a bunch of these, currently using ZHA. As far as setting the time
mine pulled the system and set to that, it was automatic when I paired
them. I have not been able to change from 24h time -> 12h time not set
outside temperature as (from what I have read at least) these values are
inside the zigbee manufacturer cluster and I cannot write to these with the
current ZHA in HA as I believe it requires a manufacturer PIN. The above
are relatively minor for me though and the overall operation of the
thermostats in ZHA work great.

- Eric

SwoopX, 2020-12-01:
Time snyc currently requires creation of a corresponding time resource for a device and explicit whitelisting. Wouldn't be much of a problem to do that.

The Sinope specific cluster attributes have not been added, as I haven't found the time yet. However, once added, the extra functionality should be configurable via deconz GUI quite easily. Also, there's been put quite some work in the thermostats lately, so the Sinope's could probably participate in that as well.

gyzod, 2021-02-21:
I use TH1123ZB with deCONZ ConBee II and Openhab3 deCONZ binding.  Everything works fine.  But, is it possible to show to outdoor temperature on the device ?

Thanks

TheHerb007, 2022-01-24:
> Would love it too, I just ordered a TH1124ZB before reading this thread. I
> thought it was already the case since it was marked as supported by deconz
> on I thought it was already the case since it was marked as supported by
> deconz on https://zigbee.blakadder.com/Sinope_TH1124ZB.html but didn't
> see that outdoor temp isn't mentionned ;-(

I’ve been using the ZHA integration in HA and have been able to see the
outdoor temp via manual cluster changes. Seems definitely possible from
what i can see.

- Eric

BabaIsYou, 2022-01-24:
> I’ve been using the ZHA integration in HA

I'm using deCONZ and Phoscon gateway Plugin in Domoticz, don't know if it can be done a similar way.

BabaIsYou, 2022-02-07:
> for setting the temp and everything they work great.

Hi @amritk,
I also use a ConBee II (with Domoticz_Deconz plugin, based on Deconz-rest-plugin (as far as I understood it). How did you get outdoor temp working ?

BabaIsYou, 2022-02-08:
> I’ve been using the ZHA integration in HA and have been able to see the outdoor temp via manual cluster changes. Seems definitely possible from what i can see. - Eric
Hi @TheHerb007 ,
What do you mean by "manual cluster change" ?

TheHerb007, 2022-02-08:
In HA if you go into the specific device it will let you view the Zigbee
clusters and within each cluster the individual values at each address. You
can get/set various values on the device in that way. If you google zigbee
clusters and zha it will probably have instructions as well.

BabaIsYou, 2022-03-21:
@TheHerb007 Many thanks, but the only way to get accces to Sinope specific cluster content in my config is using deConz-GUI (ConBee II with Domoticz_Deconz plugin, based on Deconz-rest-plugin). Then I,m looking for a way to eventually bind this cluster using DDF files but I'm a newbee regarding this !

Mimiix, 2022-03-22:
@BabaIsYou you need to create a ddf for it. It's not a bug afaik, it's something that's missing. @Smanar can you help here?

Smanar, 2022-03-23:
Exactly, no more to say.
Except the "write feature" is on the way for tuya stuff, so I think it will be availble for other stuff soon https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5868

BabaIsYou, 2022-03-23:
> Exactly, no more to say. Except the "write feature" is on the way for tuya stuff, so I think it will be availble for other stuff soon #5868

Ok, just have to wait ... Would be interesting to know which of Sinope or Stelpro that have this feature, will be availaible first ... I've other thermostats to buy ;-)

BabaIsYou, 2022-03-23:
> So about [#5897 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5897#issuecomment-1076580949)
> 
> I think it's a problem when deconz have created the DDF

I created the DDF because I tought that it could be the way to get access to Sinope Specific Cluster to write information in it. Now that I understood that it's not the way to do this I deleted the DDF file and restarted deCONZ. But the power attribute is still there.
`{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "0b75871216584adf3539b260fac2f41b",
  "lastannounced": null,
  "lastseen": "2022-03-23T19:05Z",
  "manufacturername": "Sinope Technologies",
  "modelid": "TH1124ZB",
  "name": "Consommation Salon",
  "state": {
    "consumption": 51182,
    "lastupdated": "2022-03-23T19:08:45.613",
    **"power": 0**
  },
  "swversion": "2007",
  "type": "ZHAConsumption",
  "uniqueid": "50:0b:91:40:00:02:ca:1c-01-0702",
  "id": "18",
  "deviceclass": "sensors"
}
`
That's not a DDF issue as I undestand.

Smanar, 2022-03-24:
By defaut, on the lagacy code, it s an issue ^^ (other problem)

```
                clusterId = clusterId ? clusterId : METERING_CLUSTER_ID;
                if ((sensor.modelId() != QLatin1String("ZB-ONOFFPlug-D0005")) &&
                    (sensor.modelId() != QLatin1String("TS0121")) &&
                    (!sensor.modelId().startsWith(QLatin1String("BQZ10-AU"))) &&
                    (!sensor.modelId().startsWith(QLatin1String("ROB_200"))) &&
                    (!sensor.modelId().startsWith(QLatin1String("lumi.plug.ma"))) &&
                    (sensor.modelId() != QLatin1String("Plug-230V-ZB3.0")) &&
                    (sensor.modelId() != QLatin1String("lumi.switch.b1naus01")) &&
                    (sensor.modelId() != QLatin1String("lumi.switch.n0agl1")) &&
                    (!sensor.modelId().startsWith(QLatin1String("SPW35Z"))))
                {
                    item = sensor.addItem(DataTypeInt16, RStatePower);
                    item->setValue(0);
                }
```

The code use a blacklist, and as your device is not inside, you have the useless field.

But you was right, make a DDF is a better idea.
- Soon the the code will be removed
- DDF is the future
- Even you can't use already the "write feature" you can already remove the useless field (the lagacy code will be not patched)

If you make a DDF it will be a base for later.

BabaIsYou, 2022-03-25:
Hi,

I understand that I can delete it, but another avenue could be to have it from useless to usable. Then in my frontend (actually Domoticz) I can choose which one I want to keep (if not both) and which one I want to hide. 

But in this particular case I think that you got it when you wrote `So you have some configuration for both, but in a perfect world you will need to have.

    state/power in ZHAConsumption is in cluster 0x0702 attribut 0x0400
    state/power in ZHAPower is in cluster 0x0B04 attribut 0x050B
`
In fact the attribute 0x400 n cluster 0x0702 (called Instantaneous report) does not exists in this thermostat. Making a read on the cluster greyed this attribute ! 

The other one (state/power in ZHAPower is in cluster 0x0B04 attribut 0x050B )is active and reports.

That's why, I guess, in ZHAConsumption, and in ZHAPower, they have been associated to the same place : cluster 0x0B04 attribut 0x050B.

Then, I still not undertstand, as they point to the same attribute, one is reporting and the other does not.

Smanar, 2022-03-25:
>That's why, I guess, in ZHAConsumption, and in ZHAPower, they have been associated to the same place : cluster 0x0B04 attribut 0x050B.

Deconz make sensor according to cluster 
>"uniqueid": "50:0b:91:40:00:02:ca:1c-01-0702",

Mean all values from this sensor are from the cluster 0x0702

So if your device don't have 0x0400 in cluster 0x0702, it mean you don't need the state/power on the ZHAconsumption (Its the sensor for 0x0702)

It's the device firmware that choose, not the user. There is no utility to have 2 time the same value on 2 differents sensors.

Smanar, 2022-03-26:
>once in ZHATemperature (50:0b:91:40:00:02:ca:1c-01-0402) and once in ZHAThermostat (50:0b:91:40:00:02:ca:1c-01-0201). And both relates to Cluster 0x0402, Attribute 0x000

I can be wrong, you can check deconz to be sure, but you can have both ^^
- Cluster 0x0201 attribute 0x0000
- Cluster 0x0402 attribute 0x0000

Most of thermostat have only the cluster 0x0201, but your device is more advanced.

BabaIsYou, 2022-04-14:
@Smanar. Regarding to screenshots in issue #5897 that is now closed, Do you have any suggestion on modification do I have to introduce in this DDF to have Power correctly reporting from correct cluster/attribute ? Many thanks.
`{

  "schema": "devcap1.schema.json",

  "manufacturername": "Sinope",

  "modelid": "TH1124ZB",

  "product": "TH1124ZB",

  "sleeper": false,

  "status": "Gold",

  "path": "/devices/th1124zb.json",

  "subdevices": [

    {

      "type": "$TYPE_THERMOSTAT",

      "restapi": "/sensors",

      "uuid": [

        "$address.ext",

        "0x01",

        "0x0201"

      ],

      "fingerprint": {

        "profile": "0x0104",

        "device": "0x0301",

        "endpoint": "0x01",

        "in": [

          "0x0000",

          "0x0201",

          "0x0204",

          "0x0B05"

        ]

      },

      "items": [

        {

          "name": "attr/id"

        },

        {

          "name": "attr/lastannounced"

        },
[... 525 lines cut ...]
        },

        {

          "name": "state/lastupdated"

        },

        {

          "name": "state/temperature",

          "description": "The current temperature in °C &times; 100.",

          "default": 0

        }

      ]

    }

  ]

}`

Smanar, 2022-04-14:
If I remember you want to remove the useless state/power from the ZHAConsumption sensor ?

BabaIsYou, 2022-04-14:
in fact you suggested to remove it as I prefered to have correctly reporting. Perhaps in modifying pre-generated DDF file.

Smanar, 2022-04-15:
I think yes it's possible, but better to keep that for you ^^, deconz rules are more "1 endpoint > 1 sensor"
Go in the ZHAconsumption, select the state/power and in parse fill values to have something like

"at": "0x050B"
"cl": "0x0B04"
"ep": 1
"eval": "if (Attr.val != -32768) { Item.val = Attr.val / 10; }"
"fn": "zcl"

BTW, I m looking your PR, you have "typo" issue, there is a blank line beetween all lines.

BabaIsYou, 2022-04-15:
Many thanks. I’ll try it.
For the PR, I saw it afterward … will correct it.
It Gold correct for DDF stage or is it préférable to have it Bronze ? I read à lot of discussions about this but it’s still not clear.

Mimiix, 2022-04-15:
I've adjusted the post so the bot wont close it. As it was really important for a user to be re-opened.

BabaIsYou, 2022-04-15:
> Many thanks. I’ll try it. 
Tried it and the value is still null in the API 

{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "1f674c52d452fd184f6adb1bd5f07ff2",
  "lastannounced": null,
  "lastseen": "2022-04-15T11:57Z",
  "manufacturername": "Sinope",
  "modelid": "TH1124ZB",
  "name": "Conso Therm Salon",
  "state": {
    "consumption": 326155,
    "lastupdated": "2022-04-15T11:57:38.859",
    "power": 0
  },
  "swversion": "2007",
  "type": "ZHAConsumption",
  "uniqueid": "50:0b:91:40:00:02:ca:1c-01-0702",
  "id": "16",
  "deviceclass": "sensors"
}

Smanar, 2022-04-15:
Ha yes, I haven't read the DDF. You probably miss the binding
Try to add a bind for the cluster 0x0B04, and configure at least 1 report like this one 

```
       {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
```

ATM the only binding you have is for the on/off state on the cluster 0x0006

> Many thanks. I’ll try it. Just for "parse", not read ?

It's for that I m speaking about binding/report (You can already have set them on a previous procedure or using the GUI).
When a device make a report, the value is read with the "parse", if the report not working, deconz make a poll and the value is read with the "read".
So better to configure both, but on the "normal way" it need to be read by the "parse"

But for me your DDF need to work. It's perhaps because it s not possible using another cluster than the one the ressource is created for. If it still noit work I will ask to others devs.

BabaIsYou, 2022-04-17:
> Ha yes, I haven't read the DDF. You probably miss the binding

Seems my mistake. I directly modified the DFF using text editor and restarted deConx, assuming that it would charge the new file but it doesn't. May be I wasn't in the right file ... Weird for so !

I used the GUI then, assured that I was in Gold DDF state and generated a new .json from GUI (overwriting the previous one) and modified it directly using GUI and ... it works, wihtout binding !?

I had to modify some scales after that, voltage was no more un V but un mV and I did not have to divide power by 10 as you proposed.

Here is an extract of debug view
`08:16:39:489 50:0b:91:40:00:02:ca:1c-01-0702/state/consumption expression: Item.val = Attr.val --> 345585
08:16:41:490 50:0b:91:40:00:02:ca:1c-01-0702/state/consumption expression: Item.val = Attr.val --> 345586
08:16:43:086 50:0b:91:40:00:02:ca:1c-01-0b04/state/voltage expression: if (Attr.val != 65535) { Item.val = Attr.val / 10 ; } --> 238.9
08:16:43:130 50:0b:91:40:00:02:ca:1c-01-0b04/state/current expression: if (Attr.val != 65535) { Item.val = Attr.val; } --> 4021
08:16:43:491 50:0b:91:40:00:02:ca:1c-01-0702/state/consumption expression: Item.val = Attr.val --> 345587
08:16:47:102 00:0d:6f:00:15:65:fd:b0-01/state/on expression: Item.val = Attr.val --> true
08:16:52:184 50:0b:91:40:00:02:ca:1c-01-0b04/state/voltage expression: if (Attr.val != 65535) { Item.val = Attr.val / 10 ; } --> 238.9
08:16:52:239 50:0b:91:40:00:02:ca:1c-01-0b04/state/current expression: if (Attr.val != 65535) { Item.val = Attr.val; } --> 4101
08:16:52:626 50:0b:91:40:00:02:ca:1c-01-0702/state/consumption expression: Item.val = Attr.val --> 345588
08:16:54:515 50:0b:91:40:00:02:ca:1c-01-0702/state/consumption expression: Item.val = Attr.val --> 345589
08:16:56:510 50:0b:91:40:00:02:ca:1c-01-0702/state/consumption expression: Item.val = Attr.val --> 345590
08:16:58:514 50:0b:91:40:00:02:ca:1c-01-0702/state/consumption expression: Item.val = Attr.val --> 345591
08:17:01:091 50:0b:91:40:00:02:ca:1c-01-0b04/state/voltage expression: if (Attr.val != 65535) { Item.val = Attr.val / 10 ; } --> 238.9
08:17:01:145 50:0b:91:40:00:02:ca:1c-01-0b04/state/current expression: if (Attr.val != 65535) { Item.val = Attr.val; } --> 4101`

and the actual DDF for that thermostat if it can be used by someone else

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Sinope",
  "modelid": "TH1124ZB",
  "product": "TH1124ZB",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/th1124zb.json",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
[... 246 lines cut ...]
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "description": "The current temperature in °C &times; 100.",
          "default": 0
        }
      ]
    }
  ]
}`

Now that it's correct, many thanks for that,  I have to fix another issue but in Domoticz_deCONZ plugin because it seems I lost informations on a device but it will be an another forum ;-)

Smanar, 2022-04-17:
>it works, wihtout binding !?

Perhaps the binding was already done previously (or by the legacy code), we will see on usage.

BabaIsYou, 2022-04-19:
You're right, I don,t know how the bindings could have gone regarding to my first post to help @MattL0 on issue #5848.

Then I get back bindings into DDF that looks like this 
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Sinope",
  "modelid": "TH1124ZB",
  "product": "TH1124ZB",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/th1124zb.json",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/mode",
          "description": "Mode of the device."
        },
        {
          "name": "config/offset",
[... 250 lines cut ...]
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
`

BabaIsYou, 2022-04-20:
Resulting into 3 PR #5968 , #5962  and #5904 
Still looking for a way to address Outdoor temprature on display specific attribute.

BabaIsYou, 2022-04-22:
Would be nice indeed, much more than Backlight ;-)

Smanar, 2022-04-22:
I m on other thing ATM (Tuya TRV, and reverse command for covering)

Are you skilled enought to make/test change with mimic the "clickmode" code ?
Else I can at least make the C++ code ?
I don't remember if you are able to compile the code ?

BabaIsYou, 2022-04-22:
Hmmm would like to … but never coded in C++ …

And never compiled before on Raspberry.

May I’ll give a try but not before a couple of days …

Smanar, 2022-04-23:
>Hmmm would like to … but never coded in C++ …

Don't worry, will be just copy/paste, I can make the code and give you the procedure.

>And never compiled before on Raspberry.

I will give you the command, no danger for your production network, you just need a full/real OS

>May I’ll give a try but not before a couple of days …

Sure, I will try to give you the workaround this week end.

MattL0, 2022-04-23:
I didn’t read the full topic yet but i will receive the TH1124TB tomorrow. I could compile the code and test it then.

Smanar, 2022-04-23:
Have started the code here https://github.com/Smanar/deconz-rest-plugin/tree/sinope_test

But I have a problem https://github.com/Smanar/deconz-rest-plugin/commit/a247c02be51dd87c81bc6fbd4e7690c443b7adea

From the xml the attribute is on the cluster ff01, but this device haven't this cluster ???

BabaIsYou, 2022-04-23:
Ok will give a try … I can also have a virtuel Mint machine … but my Conbee II, deConz and all my stuff is on Raspberry

BabaIsYou, 2022-04-23:
Hum … weird because using deConz GUI and entering a value in Outdoor temp on display on this cluster I had it displayed

Smanar, 2022-04-24:
>Hum … weird because using deConz GUI and entering a value in Outdoor temp on display on this cluster I had it displayed

So it's a good new, but on the capture, I don't see the sinope cluster ? How you do that ?
I m bad reading the xml file ?

BabaIsYou, 2022-04-24:
Don’t have access to my interface before several hours from now but you can find the attribute line 4342 of this version of general.xml https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5904/files

Did i miss the cluster line in it ?

Smanar, 2022-04-24:
Ok so I presume the capture is trunkated, and the missing cluster is the last one.

So to test change, you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

The code is based on the last deconz version.

```
sudo apt install deconz-dev
git clone --branch sinope_test https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

It don't impact your zigbee database, but can use a lot the CPU (at least the first time, compilation will be faster after).
To rollback fastly you can make a backup of the file /usr/share/deCONZ/plugins/libde_rest_plugin.so in another folder

And you need too add the json file https://github.com/Smanar/deconz-rest-plugin/blob/sinope_test/devices/generic/items/config_temp_to_display_item.json

You will be able to see it using the editor and can add it to your DDF.

The request to chnage the temperature is {"temp_to_display":1234} to /config endpoint.

MattL0, 2022-04-24:
Thanks!!

I have compiled and put the json file in place. I am not sure how to play with a ddf. 

This is what i got in homeseer. Is there any wway to poll the thermostat more frequently for temperature? I can do it mannually by reading the attribut in deconz-gui. 

[image]
[image]
[image]

MattL0, 2022-04-24:
The stats made a couple a friend pretty fast 
[image]

MattL0, 2022-04-25:
The powerusage is somewhat off.  I don’t see any logic in it. 
[image]

BabaIsYou, 2022-04-25:
> Ok so I presume the capture is trunkated, and the missing cluster is the last one.

Hi,

Here is a screenshot of the cluster
[image]

@MattL0 : Are you sure you need a new DDF for that thermostat ? If so you could find one here : #5962 

The aim of this test is to gain access via the rest API to the outdoor temperature display, Thus you do not need to change anything to DFF I guess.

BabaIsYou, 2022-04-25:
> The current is somewhat off. I don’t see any logic in it.

That could be normal that current is 0 when not heating, But what do you mean by "off" ?

Smanar, 2022-04-25:
Ha yes, you need to add the DDF to use last/future featurezs.
A description here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
You need to use the @BabaIsYou file.

To add the "temp_to_display" feature.
- edit DDF
- go on the ZHAThermostat device
- drag and drop the config/temp_to_display from the right panel to the left

If I m right all will be automaticallt filled (not tested on my side)

- Save file
- Hot reload

Now the setting will be visible on the device json.

BabaIsYou, 2022-04-25:
Well, got it compiled (add to do an _sudo apt-get install qt5-default_ to have the _qmake && make -j2_ working.
The first thing I saw was that the current value is weird and doesn't represent the cl:0x0b04 att:0x0508 value as it was before (before what betweenbefore beta version or before compilation I don't know for the moment).
Got the  [](https://github.com/Smanar/deconz-rest-plugin/blob/sinope_test/devices/generic/items/config_temp_to_display_item.json) and updated the DDF file after getting temp_dto_display from right to left in ZHAThermostat
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Sinope",
  "modelid": "TH1124ZB",
  "product": "TH1124ZB",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/th1124zb.json",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/mode",
          "description": "Mode of the device."
        },
        {
          "name": "config/offset",
[... 266 lines cut ...]
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}`
But I'm not at home to see if it works when changing the value. May be after job ...

[50 more replies]

closed 2023-04-13 by manup