dresden-elektronik/deconz-rest-plugin #2106: Gateway not found with Hostname

mawebi, 2019-11-21:
Hi,

in my environment the Phoscon-GW is running in a Docker container and is exposed with an Traefik reverse proxy using a given hostname.

While accessing the /pwa/login.html page, only the internal IP gets listed. But I can't access the application by the internal IP (and I dont want to open the Docker with host mode).

If I change the IP address via the JavaScript console everything worked.
So I patched the login.html as follows:

```
@@ -71,7 +71,8 @@
 var extScan = EXT_SCAN_IDLE;
 var subnets = []; // available subnets
 var subnetsToCheck = [];
-var ipv4 = [];
+var url = new URL(location.origin); // location.hostname did not work??
+var ipv4 = [ url.hostname ];
 var toCheck = [];
 var toCheckTotal = 0;
 var needCheckSubnets = false;

```

Now I can access the gateway. The shown fix is just a dirty hack. Please provide a solid solution.
(Picture of the problem attached)
[image]

b2un0, 2019-12-03:
@manup can you please fix this with the given patch above?

i also have this problem since i wrapped all my docker services with Traefik.

closed 2020-04-08 by stale[bot]

f11h, 2020-05-15:
In mid 2020 Ihave the same issue like the opener of this issue. Is there any plan to implement/fix the autodiscovery for docker users?

boekabart, 2020-09-11:
Same here; it's kind of bad that it seems almost impossible to access Phoscon from 'the outside world' behind a reverse proxy.

boekabart, 2020-09-11:
@manup also, it's not nice that the issue got closed for inactivity when everybody was just waiting for you to apply the provided patch. What's the plan, should we just file the issue again?

Mimiix, 2020-09-11:
@boekabart Please file a issue again. I'll make sure he gets it.
Make sure to add all information that is needed in the template.

manup, 2020-09-12:
Hi, sorry for the late reply. We have now changed the login page to select the host name in preference over the IPv4 address if both are available for the same bridge id.

Will be part of the upcomming release.

boekabart, 2020-12-03:
> Hi, sorry for the late reply. We have now changed the login page to select the host name in preference over the IPv4 address if both are available for the same bridge id.
> 
> Will be part of the upcomming release.

Is this already released? And is any configuration on deConz necessary - or should setting up the reverse proxy suffice? ATM I'm using 2.06.00 in docker; and have not configured anything special except nginx. PWA tries to get the config from the IP (via https, which won't work) instead of the hostname - even though at first it DOES actually once retrieve it via the hostname... but that seems to be ignored.

simonszu, 2021-08-16:
+1 for me. Running 2.12.05 via Docker and reversing with Traefik. Local access via the host IP:port works fine, but not via Traefik. 
When accessed via traefik the device (a conbee II) is shown as a raspi, like in the original post. The IP is shown as the host's LAN IP since the container is running in host mode. Nothing else is really working, and the JS console shows this: 

```
device logged
bf9a7401-5101-438b-98e1-839c8d4a71d7:101 scanner worker init
bf9a7401-5101-438b-98e1-839c8d4a71d7:332 add ip 192.168.10.4
bf9a7401-5101-438b-98e1-839c8d4a71d7:332 add ip 192.168.10.4:8081
Object
192.168.10.4:8081/api/:1 Failed to load resource: net::ERR_TIMED_OUT
192.168.10.4:8081/api/:1 Failed to load resource: net::ERR_TIMED_OUT
192.168.10.4:8081/api/:1 Failed to load resource: net::ERR_TIMED_OUT
192.168.10.4:8081/api/:1 Failed to load resource: net::ERR_TIMED_OUT
192.168.10.4:8081/api/:1 Failed to load resource: net::ERR_TIMED_OUT
```

Although this IP is reachable from the browser i am currently trying to access, it is not really working. It will be even more funnier when i am trying to access the traefik/pwa from remote.