dresden-elektronik/deconz-rest-plugin #2107: Strange issue with IKEA Tradfri Dimmer Switch CB10

dasFelix, 2019-11-21:
I am running deCONZ rest-API v2.5.71 (Beetlejuice) with conbee II on a raspberry PI 4+.

I tried to add a brand-new never added before IKEA on/off switch (aka small dimmer switch):
[image]

I tried to add it in phoscon web-gui by "adding Switch" => "other". Nothing happened, connection failure - seems not to work. After a few tries I had a look into deconz-gui. There I found the added switch. I removed that node and tried again several times - but always with the same result. 
Then I tried a 2nd brand-new, never added before switch. But also the same result again. They never shows up in phoscon, but always in deconz-gui.

But the **really strange thing** is the following... 
In the overview of phoscon web-ui (main page), where you see all groups and lights/plugs of selected group the added IKEA dimmer switch seems to switch ON/off all of the lights. So the bulbs  go to grey state if you switch OFF and according to press ON they become a yellow state. This is just "virtually" and the lights are going back to their real state one after other after some seconds. It does NOT really switch ON or OFF the lights / plugs. 

I tried also to restart "deconz-gui" and also to reboot the PI 4+. But the behaviour was the same. Just to note: I never added this switch (or any other of them) before nor did I connected a switch to "all of my devices". 

Like this it´s not useable with this switch of course. I like to provide further information if I know which or what could help. Any help on this is really appreciated. Thank you in advance.

olemr, 2019-11-21:
Have you tried using the old WEB App?
I had to assign lights to the Open/Close switch in there:
[image]

dasFelix, 2019-11-22:
Interesting. I didn´t even know this exists. But the switch does not show up in the "old" version. I can´t bind it there either. I deleted it in deconz-guy before try to bind. But no success :(

marcelkordek, 2019-11-22:
I'm using v2.5.71 (Docker) on my Synology. I have multiple Ikea ON/OFF switches and they work great. After first Pairing attempt they appear in the GUI and Web App (new and old).

dasFelix, 2019-11-22:
That's nice it's working for you. I know from other posts that normally this on / off button may be working. But for me it's not working at all :-/
Not the pairing process is a problem. This always works but the device seems not to be recognized correctly.
The phoscon web-ui fails after 3 minutes by time out. But always an object / node apears in deconz-gui. All the attributes makes no real sense to me:
[image]
[image]
As you see "factory defaults" to device fails also. 

Meanwile I tried another dozen times to get it working. I also tried going back to 2.5.69 with no success. 

I tried to pair other devices (2 Ikea tradfri LED color bulbs, 2 Ikea range extender) which works without any issue. 
I also removed and paired again a xiaomi aqara wall switch. This also works...

Only this tiny on/off switch refuses to work for me. I don't know what may cause this. Maybe the added a new firmware. Since I have no tradfri - gateway I cannot check this.

dasFelix, 2019-11-22:
Little update
Probably I just didn't wait long enough. But I can read some node information. Seems to be logic to me. 
[image]
Maybe a newer version? Date seems to be 4th of October 2019 ... hmm

Software Infos from OTAU Cluster info:
[image]

robidog, 2019-11-26:
NB: Zigbee standard specifies datecode in  ISO8601 format, i.e. YYYYMMDD. Therefore it's 10th of April, 2019.

thowi, 2019-12-15:
I had to try ~10 times to pair the on/off switch. Eventually, it worked.

IDmedia, 2020-01-04:
I have the same issue with the "TRADFRI open/close remote" which I guess is basically the same as this. The only difference is that there's a Sun/Three lines instead of I/O.

The scan and web UI does not show any new switches when trying to pair it, but I can see it using VNC (I'm using the HASSIO docker container) and I can see its events in Home-Assistant. It's just not listed in the Phoscon UI or old UI.

ebaauw, 2020-01-04:
The open/close remote is *very* different from the on/off switch.  It sends _Window Covering_ commands instead of _On/Off_ and _Level Control_ commands.

IDmedia, 2020-01-04:
> The open/close remote is _very_ different from the on/off switch. It sends _Window Covering_ commands instead of _On/Off_ and _Level Control_ commands.

Thanks for the info. Is there any way of getting it to show up in the Phoscon UI?
It seems to work well apart from not showing up...

ebaauw, 2020-01-04:
Phoscon is not open source; dresden elektronik needs to change it themselves.

closed 2020-05-10 by stale[bot]

mastervol, 2020-05-24:
I have the same problem as described in the the opening post.

ConBee II on Raspberry 2
Version: 2.05.75/8.3.2020
Firmware: 26580700

I cannot add a Ikea Tradfri On/Off Switch, not as IKEA and not as OTHER.
It is not recognized in the phoscon app.

Thanks for comments.

BenAndDevelopment, 2020-06-03:
I was able to add them, but not they right way I guess. I added them as on off switch but the UI didn't show it. Then I just closed the "add new" layer and there it was. But then I couldn't add it to any room as switch. So I switched to the old interface where I was able to add my blinds to it. Now they mostly work but it's kinda strange.

slurpey, 2020-06-08:
Same here. Are there any other ways, it drives me nuts. I'm using in hassio the "deconz" link, then in gateway I try to add it. Nothing happens.
-is there a way to reset that on off switch to factory settings?
-is there any other action than pushing 4 times the pair button?
-people are talking about the old interface in this thread but where is that link?
The crazy part is that this was already paired and I deleted it because I wanted to change its name. Regrets....

drbytes, 2020-12-05:
Here it gets added, I see it in the Switches tab but not in the overview where I would assign it lamps. So that blows, but I understand, Ikea is a super small company, very difficult to support every vendor out there.
Running latest everything on a conbeeII

drbytes, 2020-12-05:
Here it gets added but in PWA I can't assign it lights.

slurpey, 2020-12-06:
Just in case... If they show in phoscon, in HA, go to configuration, integration, the 3 dots next to phoscon and reload.

They do not typically show up under entities. They can only be reached by services.  (Use device as trigger)
`alias: ikea switch via phoscon
description: ''
trigger:
  - device_id: e7810661d9794914800560293e2d8c06
    domain: deconz
    platform: device
    type: remote_button_short_press
    subtype: turn_on
condition: []
action:
  - service: light.toggle
    data: {}
    entity_id: light.color_temperature_light_44
mode: single
`
And on the pairing, I find it handy to make a light turn on, off, brighter, etc at the same time phoscon is trying. Most of the time the interface doesn't find anything but once I close and check, it's paired...
It's a bit wizardry...

drbytes, 2020-12-06:
What would be really nice is that they would appear on the left side on the main PWA page --like all the others, so I can assign lights to them.