dresden-elektronik/deconz-rest-plugin #2108: Add Support for Illumra Dual Rocker Switch

mdolnik, 2019-11-22:
[image]
[Product Page](https://illumra.com/products/wireless-light-switches/dual-rocker-zigbee-switch/)

I'm aware that this switch uses the PTM 215Z module which has already been somewhat covered in issues [#1579](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1579) and [#1754](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1754) but those seem to be for a different type of switch.

I am able to pair this switch in Phoscon by using any of the 4 buttons and then:
1. Hold the button down for 10 seconds and let go.
1. Hold the button down for 1 second and let go.
1. Hold the button down for 10 seconds and let go.
1. Press the button between 1-10 times (to run through the channels) until a connection is made.
1. Press any other button to complete pairing.

(The instructions on [pairing-friends-of-hue-switch](https://phoscon.de/en/support#pairing-friends-of-hue-switch) do not work for me)

The intended way to pair this switch (directly to a zigbee receiver) is to do the steps above for either of the left buttons, then do the same for either of the right buttons [as shown here, but for a single switch](https://www.youtube.com/watch?v=V43Jcw_uUPE)

The issue is only two of the buttons seem to respond (only the top-left and bottom-right) I cannot get the bottom left or top right to trigger any events.

When paired in Phoscon, the switch shows up as a Philips Tap, and I am able to use the switch editor to bind the Philips Tap's "1" and "4" buttons successfully, but the "2" and "3" button bindings do not seem to get triggered.

I have tried completing the pairing process by pressing the 

I have also not been able to get the button holding functionality working, but I don't know where to start with that, I would like to get all 4 buttons working first.

Any suggestions to get this working?

mdolnik, 2019-11-22:
The switch seems to show up in deConz attached to one of my lights ([sunricher switch/receiver](https://sunricher.en.alibaba.com/product/62359944204-811954196/Zigbee_3_0_2_way_no_neutral_Rotary_dimmer.html)) as `Green Power GP Combo Basic`
[image]

And there is also this, which I don't know if it's related or not, I don't know what this is...
[image]

mdolnik, 2020-02-09:
I managed to make some (hacky) changes in order to get this switch to work.

Admittedly I don't know my way around this code, but I've figured that the `DeRestPluginPrivate::gpDataIndication()` is used to determine the type of new switch being added, so we need to identify the `PTM215ZE` switch as something different / more than the `Phillips Tap` switch.

Right now the code identifies this switch as a model ID `ZGPSWITCH` and manufacturer `Philips` and this happens in `DeRestPluginPrivate::gpDataIndication()` around line 1256 since the `options.byte` resolves to `0x81`:
```
if (gpdDeviceId == deCONZ::GpDeviceIdOnOffSwitch && options.byte == 0x81)
{
    sensorNode.setModelId("ZGPSWITCH");
    sensorNode.setManufacturer("Philips");
    sensorNode.setSwVersion("1.0");
}
```

I don't know what should be added here or what model the switch should ultimately be identified as, but I decided to identify this switch as a `FOHSWITCH` switch by changing this to:
```
if (gpdDeviceId == deCONZ::GpDeviceIdOnOffSwitch && options.byte == 0x81 && ind.payload().size() == 27)
{
    sensorNode.setModelId("FOHSWITCH");
    sensorNode.setManufacturer("PhilipsFoH");
    sensorNode.setSwVersion("1.0");
}
elseif (gpdDeviceId == deCONZ::GpDeviceIdOnOffSwitch && options.byte == 0x81)
{
    sensorNode.setModelId("ZGPSWITCH");

...
```
So that we add an extra identifier (`ind.payload().size() == 27` not sure how reliable this is) and set this switch to use the `FOHSWITCH` model / manufacturer.

This change will help change the `Illumra PTM215ZE` switch to be recognised as a `Friends of Hue Switch`, but the button mapping is way off.

In order to fix (hack) the button mapping, we need to make changes in `DeRestPluginPrivate::gpProcessButtonEvent()`. We need to change the button map from:
```
const quint32 buttonMap[] = {
    0x10, S_BUTTON_1,
    0x11, S_BUTTON_2,
    0x12, S_BUTTON_4,
    0x13, S_BUTTON_3,
    0x14, S_BUTTON_1,
    0x15, S_BUTTON_2,
    0x16, S_BUTTON_4,
    0x17, S_BUTTON_3,
    0x62, S_BUTTON_6,
    0x63, S_BUTTON_6,
    0x64, S_BUTTON_5,
    0x65, S_BUTTON_5,
    0x68, S_BUTTON_7,
    0xe0, S_BUTTON_7,
    0
};
```
To:
```
const quint32 buttonMap[] = {
    0x12, S_BUTTON_1,
    0x13, S_BUTTON_1,
    0x14, S_BUTTON_2,
    0x15, S_BUTTON_2,
    0x18, S_BUTTON_3,
    0x19, S_BUTTON_3,
    0x22, S_BUTTON_4,
    0x23, S_BUTTON_4,
    0
};
```

And change:
```
else if (btn == 0x10 || btn == 0x11 || btn == 0x13 || btn == 0x12 || btn == 0x64 || btn == 0x62)
{
    sensor->durationDue = now.addMSecs(500); // enable generation of x001 (hold)
    checkSensorsTimer->start(CHECK_SENSOR_FAST_INTERVAL);
    btn = btnMapped + S_BUTTON_ACTION_INITIAL_PRESS;
}
else if (btn == 0x14 || btn == 0x15 || btn == 0x17 || btn == 0x16 || btn == 0x63 || btn == 0x65)
{
```
To:
```
else if (btn == 0x12 || btn == 0x14 || btn == 0x18 || btn == 0x22)
{
    sensor->durationDue = now.addMSecs(500); // enable generation of x001 (hold)
    checkSensorsTimer->start(CHECK_SENSOR_FAST_INTERVAL);
    btn = btnMapped + S_BUTTON_ACTION_INITIAL_PRESS;
}
else if (btn == 0x13 || btn == 0x15 || btn == 0x19 || btn == 0x23)
{
```

All of the above changes are in the following patch:
[PTM215ZE_hacky.patch.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4178073/PTM215ZE_hacky.patch.txt)

**_These changes will break existing `FOHSWITCH` mapping_**, but will allow the `Illumra PTM215ZE` switch to be recognised as a `Friends of Hue Switch` and all 4 buttons will now work, (including configuring the buttons as `Change Brightness` to allow holding the buttons down to gradually brighten/fade instead of turn on/off)

The next step will be how we go about identifying this switch and how to configure it without breaking other `FOHSWITCH` switch mapping.

mdolnik, 2020-02-13:
Here's a patch that won't break existing FOH switches: [PTM215ZE.patch.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4196613/PTM215ZE.patch.txt)

It is the same as above, but instead of defining the `PTM215ZE` as:
- Model: `FOHSWITCH`
- Manufacturer `PhilipsFoH`
- SwVersion `1.0`

The `PTM215ZE` is defined as:
- Model: `FOHSWITCH`
- Manufacturer `PhilipsFoH`
- SwVersion `PTM215ZE`

I wasn't able to change the model or manufacturer without having the `PTM215ZE` either not show up when linking in Phoscon, or it silently linked in the background and was unusable. I don't know enough about the code to know how to add a brand new model or manufacturer.

Provided that arbitrarily setting the SwVersion is ok, this new patch also defines 2 button maps in `DeRestPluginPrivate::gpProcessButtonEvent()`, one for the existing `FOHSWITCH` and one for a `FOHSWITCH` with SwVersion `PTM215ZE`.
These button maps define the switches':
- Button command
- Mapped button
- Action (`S_BUTTON_ACTION_INITIAL_PRESS` or `S_BUTTON_ACTION_SHORT_RELEASED`)

The remainder of the changes replace the hard-coded button commands with the mapped action as that is all we need to know.

Can anyone that knows the code better than I do have a look at this and see if this is sufficient for a pull request?

marschallphilipp, 2020-02-13:
> Here's a patch that won't break existing FOH switches: [PTM215ZE.patch.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4196613/PTM215ZE.patch.txt)
>
> It is the same as above, but [...]

Great work @mdolnik  !
I would like to test it with my OPUS 55 switch if it works as expected.
Can you assist, which steps are necessary?
- export deconz settings in phoscon
- Download official deconz repository from github
- Adjust all changes you made in diff file (PTM215ZE.patch.txt)
  - [What would be a git command to apply the changes from the file to a local git repository?]
- copy all repository files to Raspberry Pi
  - [I can´t find any folder for the deconz addon, where should it be located when using a Raspi and hass.io 105.2 ?]
- restart deconz addon or hass.io

Appreciate your help!

mdolnik, 2020-02-14:
> Can you assist, which steps are necessary?

The instructions are in the readme on the main [deconz-rest-plugin](https://github.com/dresden-elektronik/deconz-rest-plugin#get-and-compile-the-plugin) page.

I personally don't know how the raspberry pi would work in regards to these steps, I would figure you should run all of these steps on the device itself instead of transferring files over, but regardless the main things that need to be done are:
- Install the dev version of deCONZ and not the main version (not entirely sure this is necessary, but it's what I've been doing)
- Override the `libde_rest_plugin.so` file in `/usr/share/deCONZ/plugins` with the modified/compiled version.

The general steps would be:
1. Backup as much as you can:
    - I personally don't know the full process of what needs backing up as I'm just playing around with deCONZ at the moment, but...
    - Export anything from the UI / Phoscon etc
    - Backup the `/usr/share/deCONZ/` folder
    - Backup the `~/.local/share/dresden-elektronik/` folder
1. Install the dev version:
    - `sudo apt install deconz-dev -y`
1. Install this plugin repository wherever you want:
    - `git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git`
1. Navigate to the git repo you just cloned
    - `cd /path/to/deconz-rest-plugin`
1. Apply the patch I supplied:
   - `git apply PTM215ZE.patch.txt`
   - It's possible that the patch may not apply as it may be based on an outdated commit, if that's the case you can either:
        - Open the patch in a text editor and use it as a guide to make manual changes to the file.
        - Let me know and I can re-roll the patch to the latest commit.
1. Compile the changes and overwrite the deCONZ plugin:
    - `qmake && make -j2 && sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins`
1. Run deCONZ with logging
   - `deCONZ --dbg-info=2`

If you get errors compiling you may need to ask someone that knows more than I do. It took me a while to get it to compile successfully due to differing versions of qmake / make.

I also don't know anything regarding hass.io, I was only using Phoscon to get the `PTM215ZE` to work.

Also note that the `PTM215ZE` does not show up in the deCONZ UI, except as a Green Power node that has attached itself to my light.

mdolnik, 2020-02-27:
@marschallphilipp did you have any luck?

marschallphilipp, 2020-04-14:
@mdolnik Thank you for the detailed description! As soon as my new Rapberry 4 will arrive I will check it out. Don´t want to break anything on my running system.
Although I had a look into the files structure on the home assistant (formerly hass.io) on the pi and couldn´t find the right folders that I will need....

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue.

closed 2020-06-05 by Mimiix

mdolnik, 2020-09-19:
I've opened a [pull request](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3285) for the patch mentioned above

mdolnik, 2020-10-21:
I can confirm the latest master commit 45a7692 works now with the Illumra switch.

Works for both commissioning as type "Other" and is able to control lights with no issues

Thanks for your help.