dresden-elektronik/deconz-rest-plugin #2110: Innr light strip not reacting to signals in .71, works in .69

magistra, 2019-11-22:
I have an Innr light strip (FL 130C) which has bern workimg fine until I updated to version .71 a couple of days ago. It simply does mot respond to any calls but it looks alive in phoscon (not grayed out). Downgrading deconz back to the previous version I ran makes the light strip work again. What can I do to help debug this?

magistra, 2019-11-22:
Works in .70 too, so it seems to have broken in. 71.

manup, 2019-11-23:
That's odd, can you please post a screenshot of the endpoint/cluster list of the light strip?

magistra, 2019-11-26:
Is this what you need?

[image]
[image]

These are taken with the .70 release.

closed 2020-04-02 by stale[bot]