dresden-elektronik/deconz-rest-plugin #2111: Xiaomi Aqara cube shows cube flip event OR rotation events but not both

hvdboorn, 2019-11-22:
Today I connected my aqara cube to my zigbee network using a Conbee II usb.
Via [Deconz supported devices](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices) and various tutorials I expected to obtain cube flip events/double-tap etc. as well as rotation events.
After connecting the cube I did not alter the sensor name and the following showed up in in Phoscon:
[image]

When tracking deconz events in home assistant, I only see cube-flips/double-taps etc. (the digital events) but no analog events (rotation). For example:
`{
    "event_type": "deconz_event",
    "data": {
        "id": "mi_magic_cube",
        "unique_id": "00:15:8d:00:02:9a:af:21",
        "event": 1000
    },
    "origin": "LOCAL",
    "time_fired": "2019-11-22T17:40:43.551809+00:00",
    "context": {
        "id": "a134f7ec8008405dbb6869e508a531cc",
        "parent_id": null,
        "user_id": null
    }
}`

I removed the cube and re-added it and sometimes the events will ONLY reflect rotation (and no cube flips). This seems to be random when I re-add the cube. I deleted and re-added the cube now five times and in three cases the cube only sends digital events and in two cases only analog rotation events. **How can I fix this?**
From this [this issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1365) I found that I shouldn't rename the cube in phoscon, and I haven't yet the problem still exists.

MacJawa, 2019-11-22:
Hi @hvdboorn ,
try to move the cube in all directions (flip and rotate) during pairing. Magic cube is added as 2 devices at the end. One represents "digital" moves - flip, knock, etc, the second one represents "analog" moves - rotation.
These are joined together in Phoscon app:
[image]

But in OpenHab I see Magic Cube as two things
[image]
with several channels each
[image]

hvdboorn, 2019-11-22:
Thank you so much for your answer. I had to re-add the cube twice again, but now it sends events for rotation and cube position.

Thank you.

closed 2019-11-22 by hvdboorn