dresden-elektronik/deconz-rest-plugin #2113: No Start: Rasp3b+ & Hass.io & Conbee II --> /dev/serial/by-id/usb-dresden_elektronik_....

Liese2007, 2019-11-22:
Hi,
I have a very fresh setup of Hass.io on a Rasperry Pi 3B+.
I received the Conbee II Stick , plugged it in -  Rasperry restart.
I nstalled the deCONZ Add-on.

Looked up the serial Hardware interface:
serial:
/dev/ttyAMA0
/dev/ttyACM0
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2132287-if00

I changed the config to:
```
{
  "device": "/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2132287-if00",
  "vnc_password": "",
  "dbg_aps": 1
}
```
So far I did not 
- forward any port for deCONZ
- I did not use INSTALL CODE written on the Conbee packing

> **And JUST NOW I see the USB stick on the Rasperry screen aside my laptop**

I do not really know , what happens here... I will keep this posted and come back soon.....

Liese2007, 2019-11-22:
OK,
I have no clue how to upload a photo here. You can see the actual screen of the Raspy here:
https://community-home-assistant-assets.s3.dualstack.us-west-2.amazonaws.com/original/3X/2/5/25de263a0392f3a4599f7d4c5db3e183886c70af.jpeg
The App still won't start.
I changed the /dev/serial/... to /dev/ttyACM0 but that didn't help either.

Regards,
Jörg

Liese2007, 2019-11-23:
Hi,
Homeassistant log shows:

> 19-11-23 09:52:16 ERROR (SyncWorker_4) [hassio.docker] Can't create container from addon_core_deconz: 404 Client Error: Not Found ("No such image: homeassistant/armhf-addon-deconz:3.8")

Any ideas?

Travis-home, 2019-11-23:
I had same Problem with mine found it was a power supply issue to the pi not giving power to usb upgrade my power supply to raspy pie and got it working https://www.jaycar.com.au/5-1v-2-5a-switchmode-power-supply-for-raspberry-pi/p/MP3536

Liese2007, 2019-11-23:
Thanks a lot,
I finally flashed a new 32GB SD Card with same Hassio Image, installed directly deCONZ and it worked.
Please check the link to HA Community.
My power supply was ordered directly with the Rasperry so I do not expect problems here...
Regsrds
Jörg

closed 2019-11-23 by Liese2007