dresden-elektronik/deconz-rest-plugin #2114: Trying to get the conbee II device running on a synology NAS DS214+

HHasenack, 2019-11-23:
The USB Stick hardware is corecectly recognized by the NAS.  (Check info page of NAS configuration)

When the stick is connected to my windows computer I can use dsCONZ to examine the stick and the Phoscom pages , so the USB stick appears all to be in working order.

However, it appears I cannot get the stick working on my synology NAS, which is where I intend it to use. It seems port 80 is already reserved by all kinds of services on my NAS, or maybe the webserver in the stick does net get up-and running for some other reason.

Does Conbee II require docker to run in the NAS? - Docker is incompatible with my NAS,.

Really stuck here. I can get it running in windows, but I want it on my NAS.

tannyl, 2019-11-23:
The stick it self does not have any webserver built into it. The ConBee II is be it self useless. unless you use it with some software that knows how to use it.

So you need to run the deCONZ software on the Synology Diskstation to be able to use it for anything.

I would have suggested using the Docker image of deCONZ, but the DS214+ does not support the Docker package.

So unless you are ready for some fancy CLI'ing (and you know your way really well around Linux) on the Diskstation, you won't be able to run deCONZ on it.

I would suggest getting a Raspberry Pi and using the ConBee II on that instead.

HHasenack, 2019-11-25:
Hallo Tanny

Thanks for the clarifying answer!

Regards,

Hans Hasenack | Hoofd ontwikkeling

[S&G-new-logo---classic---png]

www.sg-assetmanagement.nl<http://www.sg-assetmanagement.nl>

Bijsterhuizen 1160b | 6546 AS  Nijmegen, the Netherlands
T +31 (0) 24 – 645 0006

Arsur, 2020-03-13:
I dont have the conbee 2 but with version 1 i used this path yesterday
I point you in a brief direction, on how this can be done, maybe i have time to write a more detailed tutorial later)

1. Docker with macvlan virtual network interface (use the first part)
be careful with the network device you are using (eth0/eth1)
https://gist.github.com/xirixiz/ecad37bac9a07c2a1204ab4f9a17db3c#file-pihole-macvlan-synology-docker-txt-L24
2. Free Port 80 on your Nas
https://stackoverflow.com/a/55561347
3. Run
`sudo insmod /lib/modules/usbserial.ko`
`sudo insmod /lib/modules/ftdi_sio.ko`

4. Shh again into the nas and run this with docker-compose up -d
```  
version: "3.5"

services:
  deconz:
    image: marthoc/deconz:latest
    container_name: deconz
    restart: always
    networks:
      macvlan:
        ipv4_address: 192.168.1.211
    ports:
      - 80:80
      - 443:443
    volumes:
      - ./deconz/config:/root/.local/share/dresden-elektronik/deCONZ
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
    environment:
      - DECONZ_WEB_PORT=80
      - DECONZ_WS_PORT=443
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
      - TZ=Europe/Berlin
networks:
  macvlan:
    external: true
```
4. Enjoy

schumi2004, 2020-05-07:
@Arsur 
But how will this work if his NAS does not support docker?

Arsur, 2020-05-08:
I have exacly this device. It does support it.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix