dresden-elektronik/deconz-rest-plugin #2115: Getting unexpected error while aquiring api key

cheak1974, 2019-11-23:
Hi,

I pressed the Unlock Gateway Button and sent the following request:

curl http://127.0.0.1:8080/api/ -X PUT -d '{"devicetype": "test"}' | python -m json.tool

and getting the response:

[
    {
        "error": {
            "address": "/",
            "description": "unauthorized user",
            "type": 1
        }
    }
]

...but I expected to get an API Key back. Is there something wrong with my request?
Would it not be easier to generate an API Key in the WebApp which could be copied by the user?

regards

Chris

cheak1974, 2019-11-23:
Ah, OK I changed the command to POST instead of PUT and it worked instantly. 
I close the issue, but maybe for new users aquiring the API Key via Web App would still be easier to understand. Maybe you could consider to change this in future version.

regards 
Chris

closed 2019-11-23 by cheak1974

ebaauw, 2019-11-23:
> Maybe you could consider to change this in future version.

You're creating a new resource (the whitelist entry), so POST is semantically correct.