dresden-elektronik/deconz-rest-plugin #2116: Can't reach my Second Raspbee  / Phoscon App

bubffm, 2019-11-23:
I just bought a second Raspberry with Raspee Module from Dresden to be placed in the cellar three storeys down to run a couple of temperature and Water sensors form Aquara.  This is then linked to my main setup in the Appartment thru a powerline adapter - which apart for from the Raspbee all works fine.

However, I cannot access the Phoscon App to setup my Raspbee. When I use the Appl ink via the dresden webpage, i get a "Bad Gateway 502" Error.  If I try the local IP with Port 80 (which I assume is the standard port for Raspbee - at least on my other Raspbee it is), then I get a "Cant reach the server" Error in Safari.

Does anyone have any Idea? What is the easiest way to find out which Port the Raspbee might be using if it is not 80?  On 8080 I have the config-UI-X interface of my< homebridge running, whicvh all works fine.

Thanks for any clues

Smanar, 2019-11-23:
With " sudo netstat -tulpn | grep "deCONZ"" ?

bubffm, 2019-11-23:
Thanks, I'll remember that command.  The way I just fixed it is to reflash the firmware of the raspbee. After that it was immediately recognized on standard port.

thanks.

closed 2019-11-23 by bubffm