dresden-elektronik/deconz-rest-plugin #2120: Taster nicht erkannt bzw. Komisches Verhalten

DocGame69, 2019-11-24:
Ich bekomme den Aqara Opple 3fach Switch (WXCJKG13LM) [Aqara Opple Switch](https://de.aliexpress.com/item/4000310767697.html?spm=a2g0o.productlist.0.0.714f476b8p0Quf&algo_pvid=b25e5d3d-3d74-4958-9f67-8867f527eabe&algo_expid=b25e5d3d-3d74-4958-9f67-8867f527eabe-0&btsid=5f5ad3db-0d2f-4acc-a9b3-19c4484fdea6&ws_ab_test=searchweb0_0,searchweb201602_8,searchweb201603_52) nicht in Phoscon angelernt und angezeigt. Allerdings schaltet die Osram Steckdose, an der ich angelernt habe bei jedem Tastendruck auf den Schalten ein und aus. Der Taster wird weder in Phoscon noch in der Deconz oder im Iobroker-Deconz Adapter angezeigt. Verbindung muss er ja haben, da die Steckdose schaltet. 
Genau das gleiche Verhalten habe ich auch bei dem neuen Ikea Dimmer Schalter [Ikea Dimmer Schalter](https://www.ikea.com/de/de/p/tradfri-kabelloser-dimmer-weiss-70408595/).
Werden die Geräte noch eingepflegt? Ich brauche diese für einige Projekte.
Installiert ist die 2.05.70. Die Version 2.05.71, von der hier geschrieben wird finde ich leider noch nicht als .exe zum downloaden.
Gruß DocGame

timbru31, 2019-11-26:
You have way more chances of an answer if you use English here :)

JackPoint, 2019-12-10:
Duplicate of #2137

closed 2020-04-16 by stale[bot]