dresden-elektronik/deconz-rest-plugin #2121: Smartthings SmartPlug support

smartypantsuk, 2019-11-24:
I have just purchased a 2019 Smartthings Smart Plug which supports power monitoring.
Deconz treats it as an on/off plug and works fine in that respect, but I'd love to get power monitoring working. These plugs are a reasonable price without breaking the bank so would be brilliant if the energy side worked.

I have a feeling that these plugs could be manufacturered by Innr, but I aren't certain.
They are almost identical to the Innr plug with size, shape and power switch location, but there are some tiny differences regarding depth and bevel at the front lip. The build quality is identical to the Innr device.

I have attached the important stuff, If i've missed anything please advise.

Thank you
[image]
[image]
[image]
[image]

avotronics101, 2019-11-27:
I'd like this working too. I've just purchased 6 of these smartthings smart plugs to use with Homeseer via deconz. The only alternative which support energy monitoring is the Aeotec or Fibaro smart plugs but they are twice the price, around £60 each which is far too much money. The Samsung plugs support energy monitoring and cost 50% less, approx. £30. We just need to get it needed. Have you found a way yet?

smartypantsuk, 2019-11-30:
> I'd like this working too. I've just purchased 6 of these smartthings smart plugs to use with Homeseer via deconz. The only alternative which support energy monitoring is the Aeotec or Fibaro smart [...]

Unfortunately I'm no programmer. We need someone else involved?

SwoopX, 2019-12-06:
Let me see if I can help you with that. I'd definitively require your feedback regarding correct values for power and consumption...

smartypantsuk, 2019-12-06:
> Let me see if I can help you with that. I'd definitively require your feedback regarding correct values for power and consumption...

That would be great and very appreciated. 
I can help you as long as I understand the question 😆

smartypantsuk, 2019-12-06:
Let me know what you need.

Thanks

SwoopX, 2019-12-07:
No worries. Right now, I just need a little bit of time. Keep an eye on this issue and I'll ping you once ready.

Hope you're not running on docker? It would help if you could compile the plugin and do some testing once the changes are ready. Right now, I'd say I only need some feedback if the values are plausible. If not, the divisor for calculating the consumption needs to be amended.

smartypantsuk, 2019-12-07:
No I'm on Windows but I can get my Linux box out of the cupboard if needed

SwoopX, 2019-12-08:
Ok, should be good to go now. You may try the version from my repo [https://github.com/SwoopX/deconz-rest-plugin](https://github.com/SwoopX/deconz-rest-plugin).

Before you do that, please provide me with screenshots for the reporting configuration from deconz GUI. In the GUI, please go to Simple Metering and double click on attribute 0x0000 (Current Summation Delivered). Then, read config. Please do the same for Electrical Measurement and attribute 0x050B.

Good luck ;)

smartypantsuk, 2019-12-08:
I tried that unfortunately it seems to give me a NOT_FOUND message when I click on read config. I hope this doesn't mean we're out of luck? I did unlock network prior to clicking the button. I can try again tomorrow when I'm not so tired.

SwoopX, 2019-12-08:
Just have the device paired and plugged in should suffice. I wonder that it says not found, doesn't really make sense...

However, nothing serious, I was just after the default values. We can give it pretty much what we want to have in the end.

smartypantsuk, 2019-12-09:
I tried again today, device was paired, plugged in and switched on but no load.
I still got the NOT_FOUND error on read config.

[image]
[image]

SwoopX, 2019-12-09:
> You did mean 0x050B and not 0x0508?

0x050B is correct.

> I tried again today, device was paired, plugged in and switched on but no load.
I still got the NOT_FOUND error on read config.

Thanks for double checking. Currently no idea why this is but let's put that aside for the time being.

> And... pardon my ignorance, I assume that I need to compile the version in your repo for windows? Is there some guidance on that?

All good. I don't know how to compile on Windows to be honest as my stuff is running on a RPi. For Linux, it should work out with the prerequisites mentioned [here](https://phoscon.de/en/conbee2/install). Once the repo is added to apt, the dev-pachakge must be installed via 
`sudo apt-get install deconz-dev`
Lastly, the instructions for compiling the plugin are mentioned in the repo readme.

Hope, that helps.

smartypantsuk, 2019-12-09:
Thanks, I'm looking at compiling it tonight.

smartypantsuk, 2019-12-11:
SwoopX,

I have compiled the plugin and I 'think' I am running it in deCONZ but is there anyway to tell I'm actually running your plugin version, just in case I messed up?

Also, is there any way to tell if the plugin changes you made are working without connecting to deCONZ via a thrid-partry home controller? I normally use Homeseer but that's on my other windows system. I can install a controller if need be.

SwoopX, 2019-12-11:
> Also, is there any way to tell if the plugin changes you made are working without connecting to deCONZ via a thrid-partry home controller? I normally use Homeseer but that's on my other windows system. I can install a controller if need be.

Sure it is. you could use e.g. Rested plugin for Firefox (or any other REST-Plugin will do). However, to connect, you must have a valid API key. To get one, login to Phoscon via browser. Then, open a console and open deconz' DB, located in the home directory of the user deconz is running under (requires sqlite3 to be installed)

`sqlite3  .local/share/dresden-elektronik/deCONZ/zll.db`
`select * from auth;`

First values are the API key. Copy it and call the following URL (use either your local IP or 127.0.0.1 when calling it directly on the device):
`http://192.168.XXX.XXX/api/<APIKEY>/sensors`

Look for 2 sensors haveing the model identifier as in your screenshot above. The should be there and show values while a load is connected. Make sure to refresh the API query several times to see the changes.

>  but is there anyway to tell I'm actually running your plugin version

Not too sure to be honest. Didn't care much about that since Ive got a dedicated test system and gateway.

wvuyk, 2019-12-12:
@smartypantsuk 
You can also find the API key JowiHue is using on the properties of the gateway device, the JowiHue tab. You can use that as well.

smartypantsuk, 2019-12-12:
> @smartypantsuk
> You can also find the API key JowiHue is using on the properties of the gateway device, the JowiHue tab. You can use that as well.

Great, thanks Wim

smartypantsuk, 2019-12-12:
> Sure it is. you could use e.g. Rested plugin for Firefox (or any other REST-Plugin will do). However, to connect, you must have a valid API key. To get one, login to Phoscon via browser. Then, open [...]

Thanks SwoopX

wvuyk, 2019-12-12:
Check the Bridge user field, that is the API key for the plugin you need. 
Should have mentioned that in the first post :-)

smartypantsuk, 2019-12-12:
> Check the Bridge user field, that is the API key for the plugin you need.
> Should have mentioned that in the first post :-)

I actually thought it might be that but even I tried it I got an unauthorised user error. I'll try again later.

smartypantsuk, 2019-12-13:
> Check the Bridge user field, that is the API key for the plugin you need.
> Should have mentioned that in the first post :-)

I couldn't get that API key to work, whereas the method in SwoopX instructions worked, don't know why, everything in Jowihue works well.

smartypantsuk, 2019-12-13:
> > Look for 2 sensors haveing the model identifier as in your screenshot above. The should be there and show values while a load is connected. Make sure to refresh the API query several times to see the changes.

So I tried this, but the Samsung Smart Plug shows up under 'Lights' and not sensors. It also is listed only once, as below:

[image]

SwoopX, 2019-12-14:
Thanks, that's not really the expected behaviour... I'd like to see the debug output from the join process. Bould you please delete the device from Phoscon and then rejoin as a light? Before doing so, stop the deconz GUI service with the first command and enable debug output to be written to a file with the 2nd command

```
sudo systemctl stop deconz-gui
deCONZ --http-port=80 --ws-port=8888 --dbg-error=1 --dbg-info=1 > sam_join
```
Then, please post the file here.

smartypantsuk, 2019-12-15:
Okay, think I got this right...
-----------------------------------------------------------------

19:26:37:066 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
19:26:37:078 CTRL. 3.22.019:26:37:693 ZCLDB init file /home/dan/.local/share/dresden-elektronik/deCONZ/zcldb.txt
19:26:38:105 parent process bash
19:26:38:105 gw run mode: normal
19:26:38:105 GW sd-card image version file does not exist: /home/dan/.local/share/dresden-elektronik/deCONZ/gw-version
19:26:38:107 DB sqlite version 3.22.0
19:26:38:109 DB PRAGMA page_count: 30
19:26:38:109 DB PRAGMA page_size: 4096
19:26:38:109 DB PRAGMA freelist_count: 0
19:26:38:109 DB file size 122880 bytes, free pages 0
19:26:38:109 DB PRAGMA user_version: 6
19:26:38:109 DB cleanup
19:26:38:110 DB create temporary views
19:26:38:122 don't close database yet, keep open for 900 seconds
19:26:38:123 started websocket server at port 8888
19:26:38:129 found node plugin: libde_rest_plugin.so - REST API Plugin
19:26:38:133 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
19:26:38:157 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
19:26:38:312 Device firmware version 0x26490700
19:26:38:318 unlocked max nodes: 200
19:26:38:469 Device protocol version: 0x010B
19:26:38:515 new node - ext: 0x00212effff04fc3b, nwk: 0x0000
19:26:38:627 don't close database yet, keep open for 900 seconds
19:26:38:628 LightNode 1: Color temperature light 1 added
19:26:38:640 don't close database yet, keep open for 900 seconds
19:26:38:641 LightNode 2: Color temperature light 2 added
19:26:38:659 don't close database yet, keep open for 900 seconds
19:26:38:659 LightNode 3: Color temperature light 3 added
19:26:38:670 don't close database yet, keep open for 900 seconds
19:26:38:671 LightNode 4: Dimmable light 4 added
19:26:38:699 SensorNode 2 set node 0x001788010670ad57
19:26:38:712 don't close database yet, keep open for 900 seconds
19:26:38:712 LightNode 5: Color temperature light 5 added
19:26:38:723 don't close database yet, keep open for 900 seconds
19:26:38:724 LightNode 6: Color temperature light 6 added
19:26:38:762 CTRL [d0:cf:5e:ff:fe:ff:dc:9d] not viable in zll.db, skip load cache (101)
19:26:38:763 CTRL [00:17:88:01:06:35:b4:d3] not viable in zll.db, skip load cache (101)
19:26:38:764 CTRL [00:0d:6f:ff:fe:45:a0:bb] not viable in zll.db, skip load cache (101)
19:26:38:765 CTRL [00:17:88:01:06:37:f8:3f] not viable in zll.db, skip load cache (101)
19:26:38:766 CTRL [00:17:88:01:04:f2:72:7d] not viable in zll.db, skip load cache (101)
19:26:38:766 CTRL [00:15:8d:00:03:f1:00:3b] not viable in zll.db, skip load cache (101)
19:26:38:766 CTRL [00:15:8d:00:03:f0:fd:50] not viable in zll.db, skip load cache (101)
19:26:38:979 Current channel 25
19:26:38:997 CTRL got nwk update id 0
19:26:38:003 CTRL ANT_CTRL 0x03
19:26:39:042 Device protocol version: 0x010B
19:26:39:078 UPNP socket not bound, state: 0
19:26:39:120 Current channel 25
19:26:39:139 CTRL got nwk update id 0
19:26:39:145 CTRL ANT_CTRL 0x03
19:26:39:206 ZCL attribute report 0x0017880104F2727D for cluster 0x0001, ep 0x02
19:26:39:215 no button handler for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 0x21
19:26:39:215 ZCL attribute report 0x001788010670AD57 for cluster 0x0001, ep 0x02
19:26:43:057 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
19:26:43:057 GW firmware version: 0x26490700
19:26:43:057 GW firmware version is up to date: 0x26490700
19:26:43:752 Announced to internet http://dresden-light.appspot.com/discover
19:26:46:299 new node - ext: 0x00158d0003f0fd50, nwk: 0x49E3
19:26:51:062 don't close database yet, keep open for 900 seconds
19:26:55:941 saved node state in 3 ms
19:26:56:016 sync() in 74 ms
19:27:01:338 New websocket 10.0.2.15:35844 (state: 3) 
19:27:02:235 Websocket disconnected 10.0.2.15:35844 (state: 0) 
19:27:04:620 New websocket 10.0.2.15:35848 (state: 3) 
19:27:09:917 send permit join, duration: 59
19:27:23:118 DeviceAnnce of LightNode: 0x001788010472d5b2 Permit Join: 46
19:27:23:119 device announce 0x001788010472D5B2 (0xCBE7) mac capabilities 0x8E
[... 9 lines cut ...]
19:27:39:596 device announce 0xCCCCCCFFFE6E2F7D (0x8D12) mac capabilities 0x8E
19:27:39:760 DB UPDATE device_descriptors SET data = x'01408e6811525200002c520000', timestamp = 1576438059 WHERE device_id = (SELECT id FROM devices WHERE mac = 'cc:cc:cc:ff:fe:6e:2f:7d') AND endpoint = 0 AND type = 2
19:27:39:760 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'01408e6811525200002c520000', 1576438059 FROM devices WHERE mac = 'cc:cc:cc:ff:fe:6e:2f:7d'
19:27:39:771 don't close database yet, keep open for 900 seconds
19:27:40:058 don't close database yet, keep open for 900 seconds
19:27:40:059 don't close database yet, keep open for 900 seconds
19:27:40:059 LightNode 8: Smart plug 8 added
19:27:40:060 DB UPDATE device_descriptors SET data = x'01040151000108000003000400050006000207040b82fc0303000a001900', timestamp = 1576438060 WHERE device_id = (SELECT id FROM devices WHERE mac = 'cc:cc:cc:ff:fe:6e:2f:7d') AND endpoint = 1 AND type = 4
19:27:40:061 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'01040151000108000003000400050006000207040b82fc0303000a001900', 1576438060 FROM devices WHERE mac = 'cc:cc:cc:ff:fe:6e:2f:7d'
19:27:40:076 don't close database yet, keep open for 900 seconds
[... cut ...]
19:27:49:938 saved node state in 5 ms
19:27:49:960 sync() in 21 ms
19:28:09:827 Search sensors done
19:28:09:928 don't close database yet, keep open for 900 seconds
19:28:12:071 0x001788010472D5B2 error APSDE-DATA.confirm: 0xE9 on task
19:28:22:670 Websocket disconnected 10.0.2.15:35848 (state: 0) 
19:28:25:693 ZCL attribute report 0x001788010635B4D3 for cluster 0x0001, ep 0x02
19:28:26:857 saved node state in 3 ms
19:28:26:879 sync() in 21 ms
19:28:26:880 REST API plugin shutting down
19:28:26:886 sync() in 5 ms

djashjones, 2019-12-18:
I've just got mine, watching with interest :o)

djashjones, 2019-12-18:
Just got mine. Doing a good deal on amazon at the 1 = £30, 3 = £75, 5 = £120

[image]

Shows up in Home Assistant. Can switch it on/off through HA but the if I use the plugs button, the state is not updated in HA unless I do a 'read' in Deconz. No energy monitoring.

smartypantsuk, 2019-12-18:
Good deal and good plus just need monitoring.

SwoopX, 2019-12-18:
@smartypantsuk thanks for sharing, very strange. Could you please try it again but this time searching for sensors? Maybe it would also help if you put any load on it while pairing.

smartypantsuk, 2019-12-18:
Will do.

SwoopX, 2019-12-18:
Great. Meanwhile, I'dd double check how one of my innr plugs behaves.

smartypantsuk, 2019-12-18:
> @smartypantsuk thanks for sharing, very strange. Could you please try it again but this time searching for sensors? Maybe it would also help if you put any load on it while pairing.

It's difficult to put a load on while pairing because, I think in order to pair it again you have to factory reset it, which turns off the load when you do it.

SwoopX, 2019-12-18:
I see. Then maybe give it another go with the senosor search. I checked the innr plug but it pretty much goes the same as the Samsung does. Eventually, the sensors drop just in too slowly and another option could be to increase search duration for light in Phoscon under Gateway -> Advanced (at the bottom) to 3 mins.

Btw, you can also shortcut to check if the sensors have been created by having a look inside the deconz DB (as you'd do for the API key). While connected with SQLite3, use the query
`select * from sensors;`

smartypantsuk, 2019-12-18:
> I see. Then maybe give it another go with the sensor search. I checked the innr plug but it pretty much goes the same as the Samsung does. Eventually, the sensors drop just in too slowly and another option could be to increase search duration for light in Phoscon under Gateway -> Advanced (at the bottom) to 3 mins.
> 
> Btw, you can also shortcut to check if the sensors have been created by having a look inside the deconz DB (as you'd do for the API key). While connected with SQLite3, use the query
> `select * from sensors;`

Nevermind, I worked out that if I reset the plug first, then turned it on/off I could power a load while it was in pair mode. Here is the output while searching for sensors, but I don't think anything has changed.

20:40:07:034 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
20:40:07:046 CTRL. 3.22.020:40:07:770 ZCLDB init file /home/dan/.local/share/dresden-elektronik/deCONZ/zcldb.txt
20:40:08:294 parent process bash
20:40:08:294 gw run mode: normal
20:40:08:294 GW sd-card image version file does not exist: /home/dan/.local/share/dresden-elektronik/deCONZ/gw-version
20:40:08:296 DB sqlite version 3.22.0
20:40:08:298 DB PRAGMA page_count: 30
20:40:08:298 DB PRAGMA page_size: 4096
20:40:08:298 DB PRAGMA freelist_count: 0
20:40:08:298 DB file size 122880 bytes, free pages 0
20:40:08:299 DB PRAGMA user_version: 6
20:40:08:299 DB cleanup
20:40:08:300 DB create temporary views
20:40:08:314 don't close database yet, keep open for 900 seconds
20:40:08:316 started websocket server at port 8888
20:40:08:323 found node plugin: libde_rest_plugin.so - REST API Plugin
20:40:08:328 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
20:40:08:356 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
20:40:08:535 Device firmware version 0x26490700
20:40:08:541 unlocked max nodes: 200
20:40:08:657 Device protocol version: 0x010B
20:40:08:722 new node - ext: 0x00212effff04fc3b, nwk: 0x0000
20:40:08:876 don't close database yet, keep open for 900 seconds
20:40:08:877 LightNode 1: Color temperature light 1 added
20:40:08:891 don't close database yet, keep open for 900 seconds
20:40:08:892 LightNode 2: Color temperature light 2 added
20:40:08:915 don't close database yet, keep open for 900 seconds
20:40:08:916 LightNode 3: Color temperature light 3 added
20:40:08:929 don't close database yet, keep open for 900 seconds
20:40:08:930 LightNode 4: Dimmable light 4 added
20:40:08:959 SensorNode 2 set node 0x001788010670ad57
20:40:08:973 don't close database yet, keep open for 900 seconds
20:40:08:974 LightNode 5: Color temperature light 5 added
20:40:08:988 don't close database yet, keep open for 900 seconds
20:40:08:989 LightNode 6: Color temperature light 6 added
20:40:09:027 CTRL [d0:cf:5e:ff:fe:ff:dc:9d] not viable in zll.db, skip load cache (101)
20:40:09:028 CTRL [00:17:88:01:06:35:b4:d3] not viable in zll.db, skip load cache (101)
20:40:09:029 CTRL [00:0d:6f:ff:fe:45:a0:bb] not viable in zll.db, skip load cache (101)
20:40:09:031 CTRL [00:17:88:01:06:37:f8:3f] not viable in zll.db, skip load cache (101)
20:40:09:031 CTRL [00:17:88:01:04:f2:72:7d] not viable in zll.db, skip load cache (101)
20:40:09:271 UPNP socket not bound, state: 0
20:40:09:318 Current channel 25
20:40:09:336 CTRL got nwk update id 0
20:40:09:344 CTRL ANT_CTRL 0x03
20:40:09:398 Device protocol version: 0x010B
20:40:09:484 Current channel 25
20:40:09:502 CTRL got nwk update id 0
20:40:09:508 CTRL ANT_CTRL 0x03
20:40:09:555 no button handler for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 0x21
20:40:09:555 ZCL attribute report 0x001788010670AD57 for cluster 0x0001, ep 0x02
20:40:09:561 APS-DATA.indication from unknown node 0x87AF
20:40:09:561 ZCL attribute report 0x0000000000000000 for cluster 0x0402, ep 0x01
20:40:09:569 APS-DATA.indication from unknown node 0x87AF
20:40:09:569 ZCL attribute report 0x0000000000000000 for cluster 0x0405, ep 0x01
20:40:11:750 new node - ext: 0xccccccfffe6e2f7d, nwk: 0x8D12
20:40:13:404 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
20:40:13:404 GW firmware version: 0x26490700
20:40:13:404 GW firmware version is up to date: 0x26490700
20:40:14:138 Announced to internet http://dresden-light.appspot.com/discover
20:40:15:413 new node - ext: 0x00158d0003f0fd50, nwk: 0x49E3
20:40:20:238 don't close database yet, keep open for 900 seconds
20:40:25:238 saved node state in 4 ms
20:40:25:266 sync() in 26 ms
[... 47 lines cut ...]
20:42:15:008 0x001788010472f097 found group 0x0004
20:42:15:758 verified group capacity: 22 and group count: 3 of LightNode 0x0017880104aa7f0d
20:42:15:758 0x0017880104aa7f0d found group 0xFFF0
20:42:15:758 0x0017880104aa7f0d found group 0x0002
20:42:15:758 0x0017880104aa7f0d found group 0x0004
20:42:16:224 saved node state in 4 ms
20:42:16:238 sync() in 13 ms
20:42:16:538 verified group capacity: 23 and group count: 2 of LightNode 0x0017880100df1065
20:42:16:539 0x0017880100df1065 found group 0xFFF0
20:42:16:539 0x0017880100df1065 found group 0x0003
20:42:17:271 verified group capacity: 22 and group count: 3 of LightNode 0x001788010472d5b2
20:42:17:271 0x001788010472d5b2 found group 0xFFF0
20:42:17:271 0x001788010472d5b2 found group 0x0002
20:42:17:271 0x001788010472d5b2 found group 0x0003
20:42:18:036 verified group capacity: 22 and group count: 3 of LightNode 0x0017880104c1a306
20:42:18:036 0x0017880104c1a306 found group 0xFFF0
20:42:18:036 0x0017880104c1a306 found group 0x0001
20:42:18:036 0x0017880104c1a306 found group 0x0002
20:42:27:261 Websocket disconnected 10.0.2.15:38492 (state: 0) 
20:42:30:289 New websocket 10.0.2.15:38534 (state: 3) 
20:42:45:274 Websocket disconnected 10.0.2.15:38534 (state: 0) 
20:42:48:214 saved node state in 4 ms
20:42:48:243 sync() in 29 ms
20:42:48:244 REST API plugin shutting down
20:42:48:263 sync() in 5 ms

smartypantsuk, 2019-12-18:
No sensors have been created. I'll try the light search for longer.

2|Dimmer Switch|ZHASwitch|RWL021|Philips|00:17:88:01:06:70:ad:57-02-fc00|6.1.1.28573|{"buttonevent":null,"lastupdated":null}|{"battery":100,"group":"6416","on":true,"reachable":true}|{"d":2096,"ep":2,"in":[0,1,64512],"out":[6,8,5],"p":49246}|normal|1
1|Daylight|Daylight|PHDL00|Philips|00:21:2e:ff:ff:04:fc:3b-01|1.0|{"dark":true,"daylight":false,"lastupdated":"2019-12-18T20:53:33","status":230,"sunrise":"2019-12-18T08:03:03","sunset":"2019-12-18T15:53:24"}|{"configured":true,"lat":"51.507351","long":"-0.127758","on":true,"sunriseoffset":30,"sunsetoffset":-30}||normal|1
3|Dimmer Switch (2)|ZHASwitch|RWL021|Philips|00:17:88:01:06:35:b4:d3-02-fc00|5.45.1.17846|{"buttonevent":null,"lastupdated":null}|{"group":"10483","on":true,"reachable":true}|{"d":2096,"ep":2,"in":[0,1,64512],"out":[6,8,5],"p":49246}|normal|1
sqlite>

SwoopX, 2019-12-18:
U were a bit too fast on this one according to the timestamps (~40 secs) ;)

smartypantsuk, 2019-12-18:
The sensor search opens for 3 mins. The light search for 1 min. Is there some difference between light and sensor search in terms of what it looks for?

SwoopX, 2019-12-18:
Not that I'm aware of but I'd assume it would skip the search for any light node. The log should show something like this upon success `SensorNode 3: Power 3 added`

smartypantsuk, 2019-12-18:
So this time I've set the light search for 3 minutes, I also placed a larger load on the smart plug just in case the load was too small before.

21:24:49:405 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
21:24:49:420 CTRL. 3.22.021:24:50:678 ZCLDB init file /home/dan/.local/share/dresden-elektronik/deCONZ/zcldb.txt
21:24:51:091 parent process bash
21:24:51:092 gw run mode: normal
21:24:51:092 GW sd-card image version file does not exist: /home/dan/.local/share/dresden-elektronik/deCONZ/gw-version
21:24:51:093 DB sqlite version 3.22.0
21:24:51:095 DB PRAGMA page_count: 30
21:24:51:095 DB PRAGMA page_size: 4096
21:24:51:095 DB PRAGMA freelist_count: 0
21:24:51:095 DB file size 122880 bytes, free pages 0
21:24:51:095 DB PRAGMA user_version: 6
21:24:51:095 DB cleanup
21:24:51:096 DB create temporary views
21:24:51:110 don't close database yet, keep open for 900 seconds
21:24:51:112 started websocket server at port 8888
21:24:51:117 found node plugin: libde_rest_plugin.so - REST API Plugin
21:24:51:121 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
21:24:51:143 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
21:24:51:292 Device firmware version 0x26490700
21:24:51:299 unlocked max nodes: 200
21:24:51:416 Device protocol version: 0x010B
21:24:51:467 new node - ext: 0x00212effff04fc3b, nwk: 0x0000
21:24:51:587 don't close database yet, keep open for 900 seconds
21:24:51:588 LightNode 1: Color temperature light 1 added
21:24:51:600 don't close database yet, keep open for 900 seconds
21:24:51:601 LightNode 2: Color temperature light 2 added
21:24:51:622 don't close database yet, keep open for 900 seconds
21:24:51:623 LightNode 3: Color temperature light 3 added
21:24:51:635 don't close database yet, keep open for 900 seconds
21:24:51:635 LightNode 4: Dimmable light 4 added
21:24:51:646 SensorNode 3 set node 0x001788010635b4d3
21:24:51:661 SensorNode 2 set node 0x001788010670ad57
21:24:51:673 don't close database yet, keep open for 900 seconds
21:24:51:673 LightNode 5: Color temperature light 5 added
21:24:51:684 don't close database yet, keep open for 900 seconds
21:24:51:684 LightNode 6: Color temperature light 6 added
21:24:51:714 CTRL [d0:cf:5e:ff:fe:ff:dc:9d] not viable in zll.db, skip load cache (101)
21:24:51:716 CTRL [00:0d:6f:ff:fe:45:a0:bb] not viable in zll.db, skip load cache (101)
21:24:51:717 CTRL [00:17:88:01:06:37:f8:3f] not viable in zll.db, skip load cache (101)
21:24:51:718 CTRL [00:17:88:01:04:f2:72:7d] not viable in zll.db, skip load cache (101)
21:24:51:936 Current channel 25
21:24:51:953 CTRL got nwk update id 0
21:24:51:959 CTRL ANT_CTRL 0x03
21:24:51:996 Device protocol version: 0x010B
21:24:52:064 UPNP socket not bound, state: 0
21:24:52:073 Current channel 25
21:24:52:096 CTRL got nwk update id 0
21:24:52:102 CTRL ANT_CTRL 0x03
21:24:54:888 new node - ext: 0x00158d0003f0fd50, nwk: 0x49E3
21:24:56:067 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
21:24:56:067 GW firmware version: 0x26490700
21:24:56:067 GW firmware version is up to date: 0x26490700
21:24:56:814 Announced to internet http://dresden-light.appspot.com/discover
21:25:04:964 saved node state in 9 ms
21:25:04:990 sync() in 25 ms
21:25:11:624 created username: 3F430DA686, devicetype: Phoscon#B833x471
21:25:16:419 New websocket 127.0.0.1:42258 (state: 3) 
21:25:16:883 don't close database yet, keep open for 900 seconds
21:25:17:000 Websocket disconnected 127.0.0.1:42258 (state: 0) 
21:25:21:119 New websocket 127.0.0.1:42288 (state: 3) 
21:25:27:497 0x00178801048405BC error APSDE-DATA.confirm: 0xE9 on task
21:25:28:237 ZCL attribute report 0x00158D0003F0FD50 for cluster 0x0405, ep 0x01
21:25:36:819 ZCL attribute report 0x001788010637F83F for cluster 0x0001, ep 0x02
21:25:39:790 0x00178801048405BC error APSDE-DATA.confirm: 0xE9 on task
21:25:41:209 ZCL attribute report 0x0017880104F2727D for cluster 0x0001, ep 0x02
21:25:48:958 Current channel 25
21:25:48:964 CTRL got nwk update id 0
21:25:48:971 Device TTL 2944 s flags: 0x7
[... 78 lines cut ...]
21:28:48:969 Device TTL 2764 s flags: 0x7
21:28:56:093 GW firmware version: 0x26490700
21:28:56:094 GW firmware version is up to date: 0x26490700
21:29:06:932 send permit join, duration: 59
21:29:48:959 Current channel 25
21:29:48:964 CTRL got nwk update id 0
21:29:48:972 Device TTL 2704 s flags: 0x7
21:30:01:049 new node - ext: 0x00158d0003f1003b, nwk: 0x87AF
21:30:06:871 Search sensors done
21:30:10:958 saved node state in 6 ms
21:30:10:972 sync() in 13 ms
21:30:10:986 don't close database yet, keep open for 900 seconds
21:30:11:234 0xCCCCCCFFFE6E2F7D onOff 0 --> 1
21:30:38:104 ZCL attribute report 0x001788010637F83F for cluster 0x0001, ep 0x02
21:30:42:487 ZCL attribute report 0x0017880104F2727D for cluster 0x0001, ep 0x02
21:30:48:391 0x00178801048405BC error APSDE-DATA.confirm: 0xE9 on task
21:30:48:959 Current channel 25
21:30:48:968 CTRL got nwk update id 0
21:30:48:975 Device TTL 2644 s flags: 0x7
21:30:54:797 0x0017880104C1A306 error APSDE-DATA.confirm: 0xE9 on task
21:30:55:367 Websocket disconnected 127.0.0.1:42322 (state: 0) 
21:31:03:526 saved node state in 4 ms
21:31:03:568 sync() in 41 ms
21:31:03:569 REST API plugin shutting down
21:31:03:581 sync() in 3 ms

smartypantsuk, 2019-12-18:
I've just deleted and recompiled your plugin just to make sure I was running the right one. Same outcome.

SwoopX, 2019-12-22:
Sorry, I cannot really wrap my head around that. You may try manually reading the simple descriptors once again in deconz GUI (next to the bullet giving you all the cluster IDs). Also, not sure if you tried that, please restart deconz.

@ebaauw would you have any other ideas here? Might be difficult otherwise without a sniff.

smartypantsuk, 2019-12-25:
> Sorry, I cannot really wrap my head around that. You may try manually reading the simple descriptors once again in deconz GUI (next to the bullet giving you all the cluster IDs). Also, not sure if [...]

Ok I'll give another go over holidays

SwoopX, 2019-12-28:
Just received the EU version of the plug today. Of course, its different than the UK version :( Corious to see if I can get than one to work.

smartypantsuk, 2019-12-28:
> Just received the EU version of the plug today. Of course, its different than the UK version :( Corious to see if I can get than one to work.

Bloody typical 👎 
How is it different, do you mean just physically? or software?

SwoopX, 2019-12-29:
Well, I'd say in any regard. Different manufacturer, different MAC address, different clusters...
[image]

djashjones, 2019-12-29:
Well I just brought a Heiman UK plug. I hope that works. Why is it so hard to buy a zigbee plug with power monitoring that's supported in the UK?

smartypantsuk, 2019-12-29:
> Well I just brought a Heiman UK plug. I hope that works. Why is it so hard to buy a zigbee plug with power monitoring that's supported in the UK?

This^
Maybe we're just undeserved 😂.
I'm not buying zwave plugs as much as I would prefer to. Too expensive.

smartypantsuk, 2019-12-29:
> Well, I'd say in any regard. Different manufacturer, different MAC address, different clusters...
> [image]
> Your version reminds me of the innr SP 120 EU version. mine on the innr SP 240 UK version.

I had the Innr UK plug SP222 but it never supported power monitoring.

djashjones, 2019-12-29:
Can't even blame Brexit, lol.

Yes, I have 4 fibaro plugs. great devices, lots of config but £55 each!!!!. The TKB Home TZ69E has energy monitoring and the cheapest is around £35, less options and huge.

SwoopX, 2019-12-29:
Guys, what whould I say. adding support for the EU plug and everything works as it should.

Can you guys try putting a load on the plug (e.g. charge a mobile) and have a look at deconz GUI, electrical measurement, cluster 050B and see, if you get any values? That should at least give us some insights if anything is measured at all.

djashjones, 2019-12-29:
I put a 60w light bulb on mine and the values changed when I did a read

smartypantsuk, 2019-12-29:
> Guys, what should I say. adding support for the EU plug and everything works as it should.
> 
> Can you guys try putting a load on the plug (e.g. charge a mobile) and have a look at deconz GUI, electrical measurement, cluster 050B and see, if you get any values? That should at least give us some insights if anything is measured at all.
> 
> However, my Samsung plug is kinda useless to me. Simple metering is missing and upon swithcing, I feel the fuse is popping right out with it. Keep sticking with the inner SP120. Also, less than half the price here.

I did put a few hundred watt on the plug on my last attempt. Im gonna try again tomorrow and report back.

SwoopX, 2019-12-29:
@djashjones thanks for the feedback. Asking once again not only to verify that measurement works with your plug but also as the EU version dosn't support attribute reporting but only polling (which is kinda annoying). But that would also explain @smartypantsuk feedback [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2121#issuecomment-563298724).

djashjones, 2019-12-30:
Sorry, I'm not being much use and I'm running it on stock as I'm using Deconz in HASS.
[image]

smartypantsuk, 2019-12-30:
SwoopX, something has changed.... I am getting readings of sorts.
Stayed tuned!

SwoopX, 2019-12-30:
Fingers crossed. Anyhow, I've issued a pull request to get the changes in.

Most important is that you somehow get the sensors in. What still might be worth a try is once the plug is joined, put a load on it and search for sensors once again. Worked for the EU version before I added the support.

smartypantsuk, 2019-12-30:
> Guys, what should I say. adding support for the EU plug and everything works as it should.
>
> Can you guys try putting a load on the plug (e.g. charge a mobile) and have a look at deconz GUI, [...]

Tonight I tried again, but for whatever reason deCONZ was not playing nice with the USB stick.
So I removed and reinstalled deCONZ, recompiled the plugin, and deCONZ starting working again.
The smart plug was still installed from previous, so I put a load of approx 1000W (1kW) on the plug using a small heater. 

I still cannot get the READ_CONFIG button to work, it just errors like before.
However, both the 0x0000 (Current Summation Delivered) and 0x050B (Electrical Measurement and attribute) are giving readings now, and both alter as one would expect, i.e. the summation value increases steadily and the electrical measurement value reads reasonably accurately respective of the load applied. While I haven't checked the accuracy with my Fluke equipment, I did try a cheap plug in wattage meter inline with the smartplug to give me something to compare to.

1000w (1kW) rated heater:
Plug in Meter read = 1068w
Smartplug read = 1074w

However, what I could not get working was the measurements via the Rested plugin for Firefox.
The Smartplug still appeared under 'Lights' not sensors and there was no additional reading from before. The only thing I could try is removing and re-adding the smartplug once more??

SwoopX, 2019-12-30:
> Most important is that you somehow get the sensors in. What still might be worth a try is once the plug is joined, put a load on it and search for sensors once again. Worked for the EU version before I added the support.

Already played that card?

smartypantsuk, 2019-12-30:
> > Most important is that you somehow get the sensors in. What still might be worth a try is once the plug is joined, put a load on it and search for sensors once again. Worked for the EU version before I added the support.
> 
> Already played that card?

Last time I tried this I had a dehumidifier running as a load, which puts around 150w-200w on the plug, so that should have worked okay. But obviously something has changed so I'll remove it and rejoin again in a few minutes, just need to feed the whinging baby 👎 

So is it enough to remove the plug via Phoscon, that's removing it directly from deCONZ right?

SwoopX, 2019-12-30:
jop

SwoopX, 2019-12-30:
Just to be clear. First join as light and afterwards search for new sensors.

[185 more replies]

closed 2020-06-06 by Mimiix