dresden-elektronik/deconz-rest-plugin #2122: No refresh firmware version

jasimancas, 2019-11-24:
I've updated the firmware of 2 Ikea controllers and an Osram plug, but I don't see the new versions on the Phoscon website.

[image]

And in the web:
[image]

Same with controller:
[image]

And the web:
[image]

Any idea?

Smanar, 2019-11-29:
I think the API doesn't manage update return (and idk if if there is any). So to update the version in the API, try to re-include your device, without deleting it.
Or easier (harder with battery device), make phoscon in permit join, go on deconz, on the updated device, on the cluster 0000, and make a read attribute.

hellcry37, 2019-12-08:
I have the same issue

ebaauw, 2019-12-08:
> I think the API doesn't manage update return (and idk if if there is any).

Indeed.  The update is done by the (closed source) STD OTAU plugin, unknown to the REST API plugin.  The issue applies to all devices.

> make phoscon in permit join, go on deconz, on the updated device, on the cluster 0000, and make a read attribute.

When the device fingerprint has changed by the firmware upgrade (e.g. update from ZLL to ZigBee 3; new clusters), you also need to re-read the _Node descriptor_ and _Simple Descriptor(s)_ from the GUI.

closed 2020-04-14 by stale[bot]