dresden-elektronik/deconz-rest-plugin #2124: Philips Hue-Tap and Senic FOH Switch not visible after pairing

yablacky, 2019-11-24:
Pairing a Hue Tap or a Senic FOH Switch (which are both in the compatibility list of deCONZ)
can be paired using the phoscon app (on pairing the lights are all flashing).
But after pairing, both switch are only visible in the device list.
They do not appear in the list of Mainpage. All other Switch (even a Hue dimmer) appear there
and can be assiged to lights etc.

I was using gateway version 2.05.69 / 6.9.2019 and firmware version 26330500

Calling

`deCONZ --dbg-info=1 --dbg-error=1 --dbg-tlink=1 --upnp=0`

generates this output (for the time when pairing the Tap):

```
23:14:31:318 GW firmware version: 0x26330500
23:14:31:318 GW firmware version is up to date: 0x26330500
23:14:47:165 send permit join, duration: 59
23:14:51:251 GP srcId: 0x0044C061 cmd: 0x12 frameCounter: 859
23:15:05:123 GP srcId: 0x0044C061 cmd: 0xE0 frameCounter: 0
23:15:05:123 send GP pairing
23:15:05:123 SensorNode 9: Tap reactivated
23:15:05:372 FP indication 0x0104 / 0x0003 (0x00212EFFFF029CD2 / 0x0000)
23:15:05:372                       ...     (0x0000000000000000 / 0x0000)
23:15:11:319 GP srcId: 0x0044C061 cmd: 0x22 frameCounter: 861
23:15:20:159 Search sensors done
23:15:20:166 send permit join, duration: 0
23:15:25:206 don't close database yet, keep open for 900 seconds
23:15:25:250 Current channel 25
23:15:25:260 CTRL got nwk update id 5
```

while pairing the Senic FOH the output is:

```
23:41:43:535 GP srcId: 0x0171682A cmd: 0xE0 frameCounter: 0
23:41:43:535 send GP pairing
23:41:43:536 don't close database yet, keep open for 900 seconds
23:41:43:536 SensorNode 10: Hue Tap 10 added
23:41:43:836 FP indication 0x0104 / 0x0003 (0x00212EFFFF029CD2 / 0x0000)
23:41:43:836                       ...     (0x0000000000000000 / 0x0000)
23:41:50:280 GP srcId: 0x0171682A cmd: 0x68 frameCounter: 114
23:41:58:679 Search sensors done
23:41:59:166 send permit join, duration: 0
23:42:03:231 don't close database yet, keep open for 900 seconds
23:42:09:463 verified group capacity: 5 and group count: 1 of LightNode 0x00124b001ada7a88
23:42:09:463 0x00124b001ada7a88 found group 0xFFF0
23:42:16:146 Websocket disconnected 192.168.178.100:29698 (state: 0)
23:42:17:319 New websocket 192.168.178.100:29764 (state: 3)
23:42:24:988 verified group capacity: 13 and group count: 3 of LightNode 0x00124b001ad72693
23:42:24:988 0x00124b001ad72693 found group 0xFFF0
23:42:24:989 0x00124b001ad72693 found group 0x0002
23:42:24:989 0x00124b001ad72693 found group 0x1B01
23:42:25:208 Current channel 25
23:42:25:218 CTRL got nwk update id 5
23:42:25:230 Device TTL 2300 s flags: 0x7
23:42:26:540 no button handler for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 0x21
23:42:26:540 ZCL attribute report 0x0017880104E52514 for cluster 0x0001, ep 0x02
23:42:29:503 verified group capacity: 10 and group count: 3 of LightNode 0x00124b001d450bf3
```

I would expect that these switches appear in deCONZ gui (albeit this is not the main
target) and they should appear in phoscon on the mainpage in order to use them.

yablacky, 2019-11-25:
Looks similar: #2038 (there, the switch seems to be in the list on Mainpage; it does not in my case).

yablacky, 2019-11-25:
I found the reason. The bug was me.
Hue Tab and Senic FOH switch must be linked using the phoscon "switch editor": 
Mainlist -> select a group -> switch editor -> in the presented list, the switch appear, can be selected and then work as expected.

So it is no longer a technical issue and I close this.
Its more a documentation issue: it could help if it would be written somewhere that these switch do not appear in the mainlist (at least I didn't found that information).

closed 2019-11-25 by yablacky

manup, 2019-11-25:
There is a proper integration wizard for these in the works. Currently it's a bit fiddely to integrate.
In the meantime we have this little guide how to pair the Friends of Hue switches with a reference to the switch editor and its documentation.

https://phoscon.de/en/support#pairing-friends-of-hue-switch

yablacky, 2019-11-27:
Yes, thank you. I found this after opening the issue (and I was searching days before opening ...).
I'm looking forward the wizard! :-)