dresden-elektronik/deconz-rest-plugin #2126: Headless setup question raspian lite

thedannymullen, 2019-11-25:
I am having to start over on my pi setup.  The last time I setup the pi.  I used ssh -x then launched deconz from my Ubuntu GUI and configured the serial port for Conbee.

It is giving a qt error this time so I think I am missing some type of x server files.  Does anyone know the minimum required packages for me to run on raspian lite.  

So far what I did was:
1. Install deconz
2. Turn x11 forwarding on in ssh config.
3. Ssh from Ubuntu into raspian using ssh -x
4. Run deConz from command line.

Am I missing a step?

Smanar, 2019-11-27:
What is your QT error ?

thedannymullen, 2019-11-28:
> What is your QT error ?

Sorry forgot that in initial post.  

qt.qpa.screen QXCbcConnection: Could not connect to display
Could not connect to any x display

If I export the display I get: 

qt.qpa.screen QXCbcConnection: Could not connect to display 192.168.1.235
Could not connect to any x display

Where the .235 is my remote machine.  

**I have it working now. Not sure what happened the other day!!**

Here is what I did:

On Ubuntu machine typed $xhost+
Then $ssh -XY user@target

On Target I typed deCONZ at the prompt and it ran.  It was as simple as last time I did this.  

The last time I did not have to type xhost+ that I remember, but maybe that was the step I skipped.  I am going to close this issue. I need to type up some note and past in the comment for closing so anyone else can see what I did.

Flurkmark, 2019-11-29:
Qt refuses to work over x forwarding to one of my machines, xterm works fine. Qt forwards from the same machine to another just fine. Gave up and uses regular remote X display with deconz since they are on the same network. Works fine. 

xhost not needed with ssh. Needed with remote x.

thedannymullen, 2019-12-02:
Deconz headless setup.

  Follow the raspian instructions here: https://www.phoscon.de/en/raspbee/install#raspbian

Next step with a machine running a full xwindows environment, I used ubuntu 18.04 desktop.  Open a terminal type ssh -X user@deconzip.  Once logged in execute /usr/bin/deCONZ and the deconz gui should just load.

closed 2019-12-02 by thedannymullen