dresden-elektronik/deconz-rest-plugin #2128: Ghost events with Aqara vibration Sensor

Loader23, 2019-11-25:
Hi,

I having a Aqara Vibration Sensor for over 2 weeks now and everything was fine for about 1 1/2 weeks.
Recently the sensor triggers even when there is no reason for it. Can you see something in the logs or is the sensor simply broken?

    19:25:49:361 0x00158D00039D694B: 0x0101/0x0505: vibration strength: 2
    19:25:49:362 no button handler for: lumi.vibration.aq1 ep: 0x01 cl: 0x0101 cmd: 0x0A pl[0]: 0x05
    19:25:49:362 ZCL attribute report 0x00158D00039D694B for cluster 0x0101, ep 0x01
    19:25:55:844 ZCL attribute report 0x7CB03EAA00B26CE7 for cluster 0x0006, ep 0x03
    19:26:27:609 Current channel 25
    19:26:27:617 Device TTL 6419 s flags: 0x7
    19:26:58:147 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0000, ep 0x01
    19:27:27:609 Current channel 25
    19:27:27:617 Device TTL 6359 s flags: 0x7
    19:27:33:618 GW firmware version: 0x264a0700
    19:27:33:618 GW firmware version is up to date: 0x264a0700
    19:28:27:608 Current channel 25
    19:28:27:616 Device TTL 6299 s flags: 0x7
    19:29:12:032 ZCL attribute report 0x7CB03EAA0A003CDF for cluster 0x0006, ep 0x03
    19:29:12:036 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
    19:29:12:036 ZCL attribute report 0x00158D00036D5AD1 for cluster 0x0405, ep 0x01
    19:29:12:075 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
    19:29:12:075 ZCL attribute report 0x00158D00036D5AD1 for cluster 0x0403, ep 0x01
    19:29:27:608 Current channel 25
    19:29:27:616 Device TTL 6239 s flags: 0x7
    19:30:27:610 Current channel 25
    19:30:27:618 Device TTL 6179 s flags: 0x7
    19:30:48:885 0x00158D00039D694B: 0x0101/0x0505: vibration strength: 2
    19:30:48:887 no button handler for: lumi.vibration.aq1 ep: 0x01 cl: 0x0101 cmd: 0x0A pl[0]: 0x05
    19:30:48:887 ZCL attribute report 0x00158D00039D694B for cluster 0x0101, ep 0x01
    19:30:49:940 0x00158D00039D694B: 0x0101/0x0055: event: 1
    19:30:49:942 button 1007 Shake
    19:30:49:942 ZCL attribute report 0x00158D00039D694B for cluster 0x0101, ep 0x01
    19:30:50:489 0x000D6FFFFE464DC2 force poll (2)
    19:30:50:493 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0006, ep 0x01
    19:30:50:572 0x000D6FFFFE464DC2 force poll (2)
    19:30:50:594 Set sensor check interval to 100 milliseconds
    19:30:51:573 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0300, ep 0x01
    19:30:52:093 delay sending request 90 dt 1 ms to 0x000D6FFFFE464DC2, cluster 0x0300
    19:30:52:134 0x000D6FFFFE464DC2 force poll (2)
    19:30:52:214 0x000D6FFFFE464DC2 force poll (2)
    19:30:53:217 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0300, ep 0x01
    19:30:54:092 0x000D6FFFFE464DC2 force poll (2)
    19:30:54:171 0x000D6FFFFE464DC2 force poll (2)
    19:30:55:175 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0300, ep 0x01
    19:30:58:024 delay sending request 138 dt 0 ms to 0x000D6FFFFE464DC2, cluster 0x0300
    19:30:58:024 delay sending request 140 dt 0 ms to 0x000D6FFFFE464DC2, cluster 0x0006
    19:30:58:086 0x000D6FFFFE464DC2 force poll (2)
    19:30:58:094 delay sending request 140 dt 1 ms to 0x000D6FFFFE464DC2, cluster 0x0006
    19:30:58:171 0x000D6FFFFE464DC2 force poll (2)
    19:30:58:175 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0300, ep 0x01
    19:30:58:257 0x000D6FFFFE464DC2 force poll (2)
    19:30:58:261 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0006, ep 0x01
    19:31:27:607 Current channel 25
    19:31:27:615 Device TTL 6119 s flags: 0x7
    19:31:28:938 Announced to internet https://phoscon.de/discover
    19:31:33:618 GW firmware version: 0x264a0700
    19:31:33:618 GW firmware version is up to date: 0x264a0700
    19:31:54:994 sensor 8 (lumi.vibration.aq1): disable vibration
    19:31:55:193 Set sensor check interval to 1000 milliseconds
    19:32:27:607 Current channel 25
    19:32:27:615 Device TTL 6058 s flags: 0x7
    19:33:27:610 Current channel 25
    19:33:27:618 Device TTL 5998 s flags: 0x7
    19:34:04:938 ZCL attribute report 0x7CB03EAA0A003CDF for cluster 0x0006, ep 0x03
    19:34:22:733 no button map for: lumi.weather ep: 0x01 cl: 0x0000 cmd: 0x0A pl[0]: 001
    19:34:22:733 ZCL attribute report 0x00158D00036D5AD1 for cluster 0x0000, ep 0x01
    19:34:22:733 0x00158D00036D5AD1 extract Xiaomi special attribute 0xFF01
    19:34:22:733 01 battery 2995 (0x0BB3)
    19:34:22:733 04 unknown 13224 (0x33A8)
    19:34:22:733 05 RSSI dB (?) 13 (0x000D)
    19:34:22:734 06 LQI (?) 4294967296 (0x0100000000)
    19:34:22:734 64 temperature 2871
    19:34:22:734 65 humidity 2997
    19:34:22:734 66 pressure 1008
    19:34:22:734 0a unknown 13463 (0x3497)
    19:34:27:608 Current channel 25
    19:34:27:618 Device TTL 5938 s flags: 0x7
    19:35:27:610 Current channel 25
    19:35:27:618 Device TTL 5878 s flags: 0x7
    19:35:33:617 GW firmware version: 0x264a0700
    19:35:33:618 GW firmware version is up to date: 0x264a0700
    19:35:48:981 0x00158D00039D694B: 0x0101/0x0505: vibration strength: 3
    19:35:48:983 no button handler for: lumi.vibration.aq1 ep: 0x01 cl: 0x0101 cmd: 0x0A pl[0]: 0x05
    19:35:48:983 ZCL attribute report 0x00158D00039D694B for cluster 0x0101, ep 0x01
    19:35:49:549 ZCL attribute report 0x7CB03EAA00B26CE7 for cluster 0x0006, ep 0x03
    19:35:50:036 0x00158D00039D694B: 0x0101/0x0055: event: 1
    19:35:50:039 button 1007 Shake
    19:35:50:039 ZCL attribute report 0x00158D00039D694B for cluster 0x0101, ep 0x01
    19:35:50:094 Set sensor check interval to 100 milliseconds
    19:35:50:593 0x000D6FFFFE464DC2 force poll (2)
    19:35:50:597 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0006, ep 0x01
    19:35:50:678 0x000D6FFFFE464DC2 force poll (2)
    19:35:51:680 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0300, ep 0x01
    19:35:52:093 delay sending request 184 dt 1 ms to 0x000D6FFFFE464DC2, cluster 0x0300
    19:35:52:129 0x000D6FFFFE464DC2 force poll (2)
    19:35:52:207 0x000D6FFFFE464DC2 force poll (2)
    19:35:53:211 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0300, ep 0x01
    19:35:54:089 0x000D6FFFFE464DC2 force poll (2)
    19:35:54:170 0x000D6FFFFE464DC2 force poll (2)
    19:35:55:170 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0300, ep 0x01
    19:35:58:025 delay sending request 233 dt 0 ms to 0x000D6FFFFE464DC2, cluster 0x0300
    19:35:58:025 delay sending request 235 dt 0 ms to 0x000D6FFFFE464DC2, cluster 0x0006
    19:35:58:088 0x000D6FFFFE464DC2 force poll (2)
    19:35:58:094 delay sending request 235 dt 1 ms to 0x000D6FFFFE464DC2, cluster 0x0006
    19:35:58:175 0x000D6FFFFE464DC2 force poll (2)
    19:35:58:179 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0300, ep 0x01
    19:35:58:259 0x000D6FFFFE464DC2 force poll (2)
    19:35:58:263 ZCL attribute report 0x000D6FFFFE464DC2 for cluster 0x0006, ep 0x01

I have removed the sensor, reset and paired it again.
In the first line of the log you can see "vibration strength: 2" but it did not trigger there. But it triggered at 19:30:48 and 19:35:50.

Iam using Docker and have also asked in the repository there. Not really sure where it fits better, but I might reach more people this way. 
Any help would be appreciated.

ebaauw, 2019-11-25:
Did you check/replace the battery?

Loader23, 2019-11-25:
The device is 2 weeks old and the battery says 95%. It was 100% first.

ebaauw, 2019-11-25:
Better replace it, to be sure.  The sensor only reports battery voltage, which deCONZ translates to a percentage.  This is the best we can do, but fundamentally flawed.  The voltage remains high for a long time and then drops when the battery is almost empty.  There’s multiple reports of Xiaomi sensors behaving erroneously when the battery is low.

zigistar, 2019-11-25:
I had the same issue with an Xiaomi vibration sensor after I changed vibration strength to 1. I changed vibration strength to 3, this solved the issue for me.

Loader23, 2019-11-26:
Do you mean sensitivity? I have set that to 21.

zigistar, 2019-11-26:
yes, I meant sensitivity.

Loader23, 2019-11-26:
OK, any other Ideas? I have changed the battery, took less than 5 minutes and it triggered...

closed 2020-04-02 by stale[bot]