dresden-elektronik/deconz-rest-plugin #2129: Latest firmware but still on 2.05.65??

sinowine, 2019-11-26:
Hello,

I have a question about my Conbee II stick. I am on Firmware 264A0700, which should be the latest firmware if i look on this page: http://deconz.dresden-elektronik.de/deconz-firmware/
My gateway version is still 2.05.65, but but if I take a look on the changelog for deconz on https://github.com/dresden-elektronik/deconz-rest-plugin/releases, they show me the version 2.05.71.

Can you help me or is everything correct?

Smanar, 2019-11-26:
For me everything is correct.
Nothing magic, there is not a new firmware for all deconz version, so you can have the last firmware but an old deconz version.

sinowine, 2019-11-26:
And how can I get new deconz version? On the gateway no update is showing

peer69, 2019-11-26:
You will have to update on OS-level. If you run it on ubuntu/raspbian the following should update deconz if you set up the repo earlier:

´sudo apt update && sudo apt upgrade´

closed 2019-11-27 by sinowine