dresden-elektronik/deconz-rest-plugin #2130: Deconz creates "dummy" Gateway on non-existing IP

bubffm, 2019-11-26:
I am having two RaspBees in my setup.  But suddenly Phoscon is showing a third Gateway on an IP that does not even exist in my Network  (192.168.1.112)

Due to this, my Homebridge no longer starts as homebridge-hue expects credentials for this IP.

How can I get rid of it or restrict deconz to look for specific IP's only.

Thanks
[image]

closed 2019-11-26 by bubffm

DerOetzi, 2020-03-16:
I have the same here. @bubffm what is your solution?

bubffm, 2020-04-09:
sorry not around here that often.  I have just specified the IP's of my gateways in the home bridge-hue plugin, so it ignores all other gateways.  

No problems ever since

ebaauw, 2020-04-09:
Your Internet provider is sharing “your” public IPv4 address, causing the discovery to show your (Internet) neighbour’s gateway.  For Homebridge Hue, set `nupnp` to false to only do discovery using (local) UPnP, or explicitly list your gateway(s) under `hosts`.

DerOetzi, 2020-04-09:
> Your Internet provider is sharing “your” public IPv4 address, causing the discovery to show your (Internet) neighbour’s gateway. For Homebridge Hue, set `nupnp` to false to only do discovery using (local) UPnP, or explicitly list your gateway(s) under `hosts`.

But if I got you right this although means my neighbours can discover my gateway as well, without the possibility to forbide this. I think this is a huge security issue, pronouncing my gateway to the world outside.

How does discovery work exactly so this is possible behind the router I'm using is deconz opening any ports to outside? I thought deconz is a really private solution, that was the argument for deconz for me. I'm really shocked about this.

ebaauw, 2020-04-09:
There’s no open ports, only outgoing traffic.  deCONZ registers with the appspot server, who stores the registration under your public IP address.  When discovery queries the appspot server, it returns the gateways that registered from the same public IP address.  The meethue portal uses exactly the same mechanism.  I think there a hidden setting to disable the registration, try `ph -p get /config` to find it.

Note that the gateway is only visible to the “neighbours” with whom you share the public IPv4 address.