dresden-elektronik/deconz-rest-plugin #2132: unable to join Kit Pro mation sensor from Konke

adonno, 2019-11-26:
Hello I found these sensors, but I am unable to pair them in deconz  
http://www.ikonke.us/pro/kitpro/index.php#kitpro 
Have you ever seen those? will there be support for these? 

[image]

Smanar, 2019-11-26:
IDK for this sensors, but from my memory konke don't work on all channel (15, 20, 25)

adonno, 2019-11-26:
are these the chennels they work on ?

Smanar, 2019-11-27:
Yep

gijsje, 2019-12-30:
I can see this sensor being added to my VNC view of the network but if i try to add this it does not show in the phoscom app and the ofcourse it will not be added to Home Assistant.
Is there a way to solve this?

adonno, 2019-12-30:
so I tried on all available channels and i was unable to join them

closed 2020-05-06 by stale[bot]

amirthraj, 2020-06-03:
Is there any solution for this issue?